aqualinkd/AqualinkD #1: Need to do anything with Alexa skill development?

csherida, 2018-01-13:
is:issue is:open More of a question: how do you expose it as an Alexa skill that the Echo Plus will pick up?  Does it support Zigbee?  Thanks.

sfeakes, 2018-01-13:
No Zigbee support.
Alexa will depend on how you want to implement it, but ultimately you need to expose either the API end points or a MQTT broker to AWS Lambda.

If you google "Alexa to mqtt" or "Alexa tunneling" there is a lot of information on how to do that. There are other options, but those are probably the easiest. Once that's inlace, you can use the Alexa developer console to configure the endpoints.

If you are using a smart hub, I'd personally connect Aqualinkd to the smart hub on your home network then let the hub communicate to Alexa / AWS Lambda

closed 2018-01-13 by sfeakes

csherida, 2018-01-18:
You are awesome sir.  My problem is that my smart hub is Nexia which only supports Zwave which i don’t think works with what you developed. I was thinking of getting the Echo Plus with smart hub built to connect with aqualinkd.  Otherwise, i was going to wrap it like you suggested.

Thanks,
Chris