aqualinkd/AqualinkD #2: Hardware connection

johnluber, 2018-01-27:
I very excited to see your work on this.  It is exactly what I've been looking for for a while now.  Can you give some more detail on how to hook this to the controller?  I don't have an inside controller so I'd be attaching to the control panel.  

Thanks!

sfeakes, 2018-01-27:
Your control panel will have 4 wires for connections to control keypads. (usually a small screw-in terminal block). These are labeled 1,2,3,4 or Red,Blk,Yel,Gre.  2 wires are for power, and 2 are for data.  I don't want to tell you which wires are which, as this may depend on year / model / version. But use your manual to find the 2 data lines, (D+ and D-).
Use page 20 of the below link as an example.
https://www.zodiacpoolsystems.com/~/media/Zodiac/Global/Downloads/0748-91071/6594.ashx

Your USB2RS485 converter will wither have 2 or 3 connections. (if 2 ignore the next part). One is power and the other two are A and B.  Find the two labeled A and B.

Then simply connect the USB2RS485 A to D+ on the Control Panel and B to D- on the control panel. That should get the connection working.  If not, just flip the wires over.

If you've got a windows PC, then you can connect the USB2RS adapter to that and use the below jandy simulator software to test it.  I found it painful to get working & use, but it's a good test of the connection.
http://www.jandy.com/~/media/zodiac/global/downloads/simulators/aqualink_rs_simulator.exe

csherida, 2018-01-27:
Same here.  I have the box outside and working with an electrician to get an out let to it so I can plug in the raspberry pi. Happy to post video and share once I get it up.

https://www.zodiacpoolsystems.com/~/media/Zodiac/Global/Downloads/0748-91071/6594.ashx<https://www.zodiacpoolsystems.com/%7E/media/Zodiac/Global/Downloads/0748-91071/6594.ashx>

http://www.jandy.com/~/media/zodiac/global/downloads/simulators/aqualink_rs_simulator.exe<http://www.jandy.com/%7E/media/zodiac/global/downloads/simulators/aqualink_rs_simulator.exe>

johnluber, 2018-01-28:
Thank you so much both of you.  This helps me tremendously.  So, now I need to start setting this up.  I’ll post back how it goes.

Thanks again!

anovaski, 2018-02-03:
@sfeakes Thank you for your work on this. As others have stated, this was something I've been looking for. I too will be building an automation system with this and post how it goes. Just a quick question on your hook up - I have a RS8 control panel inside the house already wired to the 485 4-pin. Do I continue to leave the RS8 wired up and simply add the A/B cables from the USB2RS485? Or do I disconnect the RS8 and use the Pi solely for control?

sfeakes, 2018-02-03:
@anovaski you can do either.  I designed this to act as another control panel on the RS485 buss, so it will work with the RS8 connected or without it connected.  I have an RS6 panel connected as well as the PI (aqualinkd) on the RS buss (also SWG and Heater.).  It’s very hard to tell from the documentation, but I believe Aqualinkd took the ID of a “one touch” control panel, so if you have that hooked up to the RS buss then you may have a conflict.  The aqualinkd config allowed you to change the ID on the RS buss, so you can change the ID if there is a conflict.

johnluber, 2018-02-11:
Hi

Finally had a dry and warm enough day to try this.  Unfortunately, it doesn’t seem to work.  Any ideas how to troubleshoot?

Here’s what I’ve done:

1. Found the specs for the controller and wired accordingly replacing my iaqualink 2.0 with the pi.  I had an extra 4 pin connector so that was handy.  For reference I wired the A pin to pin 2 and the B pin to pin 3.
2.  All wiring connections were done with everything powered down.  Powered everything up.
3.  aqualinkd web interface comes up and connects.
4.  /var/log/messages indicates:

Feb 11 14:04:17 raspberrypi aqualinkd: Aqualink Daemon v0.8a
Feb 11 14:04:17 raspberrypi aqualinkd: Config log_level         = 7
Feb 11 14:04:17 raspberrypi aqualinkd: Config socket_port       = 80
Feb 11 14:04:17 raspberrypi aqualinkd: Config serial_port       = /dev/ttyUSB0
Feb 11 14:04:17 raspberrypi aqualinkd: Config web_directory     = /var/www/aqualinkd/
Feb 11 14:04:17 raspberrypi aqualinkd: Config device_id         = 0x0a
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_server       = localhost:1883
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_dz_sub_topic = domoticz/out
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_dz_pub_topic = domoticz/in
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_aq_topic     = aqualinkd
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_user         = (null)
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_passwd       = (null)
Feb 11 14:04:17 raspberrypi aqualinkd: Config mqtt_ID           = aqualinkd_b827ebbbc0
Feb 11 14:04:17 raspberrypi aqualinkd: Config idx water temp    = 13
Feb 11 14:04:17 raspberrypi aqualinkd: Config idx pool temp     = 14
Feb 11 14:04:17 raspberrypi aqualinkd: Config idx spa temp      = 15
Feb 11 14:04:17 raspberrypi aqualinkd: Config deamonize         = YES
Feb 11 14:04:17 raspberrypi aqualinkd: Config log_file          = (null)
Feb 11 14:04:17 raspberrypi aqualinkd: Config light_pgm_mode    = 0.00
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Filter_Pump   = label Filter Pump     | dzidx 37
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Spa_Mode      = label Spa Mode        | dzidx 38
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_1         = label Cleaner         | dzidx 39
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_2         = label Waterfall       | dzidx 40
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_3         = label Spa Blower      | dzidx 41
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_4         = label Pool Light      | dzidx 42
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_5         = label Spa Light       | dzidx 43
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_6         = label NONE            | dzidx 0
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Aux_7         = label NONE            | dzidx 0
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Pool_Heater   = label Pool Heater     | dzidx 44
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Spa_Heater    = label Spa Heater      | dzidx 56
Feb 11 14:04:17 raspberrypi aqualinkd: Config BTN Solar_Heater  = label Solar Heater    | dzidx 0
Feb 11 14:04:17 raspberrypi aqualinkd: Starting web server on port 80
Feb 11 14:04:17 raspberrypi aqualinkd: Starting MQTT client to localhost:1883
Feb 11 14:04:17 raspberrypi aqualinkd: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
5.  Web interface indicates it’s that it is not connected (I assume to the RS8)  I do see that the red dot in the top left turns green once everything is powered up and has loaded.
6.  Swapped the A / B pins; no joy
7.  Turned on debug level logging but no apparent difference.

When I ordered the RS485 adapter the one linked on your site was unavailable.  So, I ordered this one instead which appears to be identical.  https://www.amazon.com/gp/product/B01N7N62B8/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B01N7N62B8/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1>  

Thanks.

John

sfeakes, 2018-02-12:
That RS485 adapter looks perfect.  Do the lights on it flash when aqualinkd is running? (that means at least it's trying to send / receive info over the adapter).

Can you run aqualinkd in debug mode and write the output to a file, then upload a copy. That way I should be able to help you.

In the /etc/aqualinkd.conf make sure you set debug and log to file, as below
```
log_level=DEBUG
log_file=/tmp/aqualinkd.log
```

Make sure aqualinkd is not running as a service.
```
sudo systemctl stop aqualinkd
```

Then run aqualinkd manually and stop it demonizing.
```
sudo /usr/local/bin -d -c /etc/aqualinkd.conf
```

Let that run for a few minutes, hit it from the web UI if you want, then post/upload /tmp/aqualinkd.log to this thread.

For reference, as long as you only hit the data lines on the RS485 you shouldn't need to worry about it being powered down, the 2nd and 3rd pins are usually the correct ones to tap and 1 and 4 are usually power. (But that's just a general rule, not a given).

The first thing to check (assuming the USB connection & RS pins are accurate) would be the device_id.  the RS8 will only probe particular ID's.   0x0a is default, next you could try 0x09.

Also, if you have a windows laptop, you could try Jandy's emulator, just plug the RS458 into that laptop and download the software from a few posts above.

johnluber, 2018-02-12:
Thanks for getting back to me with your suggestions.  I appreciate it and understand that this something that you you spend your spare time on so that makes it even more impressive.  Unfortunately, it’s raining here again all week so it’ll be a bit before I get to test again, but I’m hopeful to get this working still.

I’ll let you know what I find.

Thanks again.

johnluber, 2018-02-17:
Hi

I’ve had some time today to work with this some more and am making progress.  In general, non-aux devices seem to work well.  Aux devices, however, seems a bit inconsistent.  For a bit Aux2 was causing a segfault in the program but that seems to have stopped.  The lights appear to be most inconsistent; I think this may have to do with my colors being different than yours.  My light system is a Jandy model.

So far, none of the temps are working in either the web nor the home kit app.Good news is this is now installed in my control panel so I can work with it at anytime now regardless of the weather.

I’ve attached some sample logs.  Notable is that the logs explicitly state that they’re ignoring all of the temp sensors.  Also, so I don’t forget…  A nice feature enhancement would be to capture the Salt and AquaPure statuses to display through to mosquito.

Thanks for any thoughts you might have on how to progress it further.

John Luber
john@luberfamily.net

sfeakes, 2018-02-17:
Great to hear you are making progress.  (The segfault is worrying on aux, that simply sends a blind command to the RS485 buss). The light modes are all hard coded to my setup (as in the instructions, that needs some work to make it generic, your logs will help). But the on / off should be working, only the pulsing to set modes will be an issue.  I don't see any logs, try to upload them again so I can take a look and help fix some of your issues.

csherida, 2018-02-17:
Awesome job John. Sounds like I have a similar setup and will try it next weekend.

My control panel is outside. Did any of you guys run power to it for the raspberry pi?

Thanks,
Chris

johnluber, 2018-02-18:
There’s prob a better way to attach logs but I’m on an iPad right now...  So, hopefully this works.

Thanks again for your help.

John

Feb-17-18 16:45:03 PM Notice: Aqualink Daemon v0.8a
Feb-17-18 16:45:03 PM Notice: Config log_level         = 7
Feb-17-18 16:45:03 PM Notice: Config socket_port       = 80
Feb-17-18 16:45:03 PM Notice: Config serial_port       = /dev/ttyUSB0
Feb-17-18 16:45:03 PM Notice: Config web_directory     = /var/www/aqualinkd/
Feb-17-18 16:45:03 PM Notice: Config device_id         = 0x09
Feb-17-18 16:45:03 PM Notice: Config mqtt_server       = localhost:1883
Feb-17-18 16:45:03 PM Notice: Config mqtt_dz_sub_topic = domoticz/out
Feb-17-18 16:45:03 PM Notice: Config mqtt_dz_pub_topic = domoticz/in
Feb-17-18 16:45:03 PM Notice: Config mqtt_aq_topic     = aqualinkd
Feb-17-18 16:45:03 PM Notice: Config mqtt_user         = (null)
Feb-17-18 16:45:03 PM Notice: Config mqtt_passwd       = (null)
Feb-17-18 16:45:03 PM Notice: Config mqtt_ID           = aqualinkd_b827ebbbc0
Feb-17-18 16:45:03 PM Notice: Config idx water temp    = 13
Feb-17-18 16:45:03 PM Notice: Config idx pool temp     = 14
Feb-17-18 16:45:03 PM Notice: Config idx spa temp      = 15
Feb-17-18 16:45:03 PM Notice: Config deamonize         = NO
Feb-17-18 16:45:03 PM Notice: Config log_file          = /tmp/aqualinkd.log
Feb-17-18 16:45:03 PM Notice: Config light_pgm_mode    = 0.00
Feb-17-18 16:45:03 PM Notice: Config BTN Filter_Pump   = label Filter Pump     | dzidx 37
Feb-17-18 16:45:03 PM Notice: Config BTN Spa_Mode      = label Spa Mode        | dzidx 38
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_1         = label Cleaner         | dzidx 39
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_2         = label Blower          | dzidx 40
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_3         = label Light           | dzidx 41
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_4         = label Pool Light      | dzidx 42
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_5         = label Spa Light       | dzidx 43
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_6         = label NONE            | dzidx 0
Feb-17-18 16:45:03 PM Notice: Config BTN Aux_7         = label NONE            | dzidx 0
Feb-17-18 16:45:03 PM Notice: Config BTN Pool_Heater   = label Pool Heater     | dzidx 44
Feb-17-18 16:45:03 PM Notice: Config BTN Spa_Heater    = label Spa Heater      | dzidx 56
Feb-17-18 16:45:03 PM Notice: Config BTN Solar_Heater  = label Solar Heater    | dzidx 0
Feb-17-18 16:45:03 PM Notice: Starting web server on port 80
Feb-17-18 16:45:03 PM Notice: Starting MQTT client to localhost:1883
Feb-17-18 16:45:03 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Feb-17-18 16:45:03 PM Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827ebbbc0
Feb-17-18 16:45:03 PM Info:   MQTT: Connection acknowledged
Feb-17-18 16:45:03 PM Info:   MQTT: Subscribing to 'aqualinkd/#'
Feb-17-18 16:45:03 PM Info:   MQTT: Subscribing to 'domoticz/out'
Feb-17-18 16:45:03 PM Info:   MQTT: Subscription(s) acknowledged
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Filter_Pump 11.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Filter_Pump
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Spa_Mode 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Spa_Mode
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Aux_1 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_1
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Aux_2 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_2
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Aux_3 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_3
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Spa_Heater 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Spa_Heater
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Aux_4 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_4
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Aux_5 1.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_5
Feb-17-18 16:45:03 PM Debug:  MQTT: topic aqualinkd/Pool_Heater 0.00
Feb-17-18 16:45:03 PM Debug:  MQTT: ignoring topic aqualinkd/Pool_Heater
Feb-17-18 16:45:06 PM Debug:  RS Received PROBE length 7.
Feb-17-18 16:45:06 PM Debug:  ++ Websocket joined
Feb-17-18 16:45:06 PM Debug:  ++ Websocket joined
Feb-17-18 16:45:06 PM Info:   WS: Message - Key 'command' Value 'GET_AUX_LABELS' | Key2 '(null)' Value2 '(null)'
Feb-17-18 16:45:06 PM Info:   WS: Message - Key 'command' Value 'GET_AUX_LABELS' | Key2 '(null)' Value2 '(null)'
Feb-17-18 16:45:07 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:45:07 PM Info:   MQTT: Published id=1: aqualinkd/Filter_Pump 1
[... 608 lines cut ...]
Feb-17-18 16:49:23 PM Debug:  Ignoring 'Salt 3000 PPM'
Feb-17-18 16:49:24 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:49:28 PM Debug:  RS Message :- '02/17/18 Sat'
Feb-17-18 16:49:28 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:49:29 PM Debug:  RS Message :- '11:47 AM'
Feb-17-18 16:49:29 PM Debug:  time not checked, will check in 3401 seconds
Feb-17-18 16:49:29 PM Debug:  RS time is accurate '11:47 AM 02/17/18 Sat'
Feb-17-18 16:49:30 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:49:32 PM Debug:  RS Message :- 'Air Temp 50ßF'
Feb-17-18 16:49:32 PM Debug:  Ignoring 'Air Temp 50ßF'
Feb-17-18 16:49:32 PM Debug:  RS Received STATUS length 12.
[... cut ...]
Feb-17-18 16:49:35 PM Debug:  Ignoring 'AquaPure 10%'
Feb-17-18 16:49:36 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:49:38 PM Debug:  RS Message :- 'Salt 3000 PPM'
Feb-17-18 16:49:38 PM Debug:  Ignoring 'Salt 3000 PPM'
Feb-17-18 16:49:38 PM Debug:  RS Received STATUS length 12.
Feb-17-18 16:49:38 PM Notice: Stopping!
Feb-17-18 16:49:38 PM Debug:  -- Websocket left
Feb-17-18 16:49:38 PM Debug:  -- Websocket left
Feb-17-18 16:49:38 PM Warning:MQTT Connection closed
Feb-17-18 16:49:38 PM Notice: Exit!

johnluber, 2018-02-18:
Hi Chris

I did run AC to mine to power the pi.  I cheated a bit and will prob do a better job of it on a warmer drier day.  I had an old outdoor extension cord that needed an end replaced so I just ran that into the panel to power the pi for now.

John

sfeakes, 2018-02-18:
@johnluber Thanks for the log.  Interesting how your panel differs from mine.  I uploaded a version than should fix the temp and a lot of other missing information from the programming mode.
Getting your salt generator % and PPM will be easy, but it will take me some time to implement, let's get you up and running first before adding functionality.
Please use the latest version, give it a try, and report back with logs.

johnluber, 2018-02-18:
Hi Shaun

Running this new code now and it is providing temps!  So, good stuff!

The only thing I see right now is that it is not reading set temps.

Here’s the output from a status pull:

{"type": "status","status":"Ready","version":"B0029223 REV T.202/18/18 Sun","date":"02/18/18 Sun","time":"9:14 AM","pool_htr_set_pnt":"-999","spa_htr_set_pnt":"-999","frz_protect_set_pnt":"-999","air_temp":"47","pool_temp":" ","spa_temp":"54","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "flash","Spa_Mode": "on","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off"}}
I noticed too this morning that you’re doing some stuff with Unifi cameras.  I’ll have to take a look at that sometime.  I’ve got 6 of those along with a bunch of other Ubiquiti gear; their stuff is fantastic!

John Luber
john@luberfamily.net

johnluber, 2018-02-18:
> john@luberfamily.net <mailto:john@luberfamily.net>

Hi Shaun

Don’t mean to be a pest…

I can replicate the crash now at will.  It happens when I create a scene in Homekit that attempts to turn on the spa and it’s accessories.  

In my case, I have it set to turn off the cleaner, turn on the pump, lights, blower, heater, and set spa mode on.

What I see is that normally when the blower goes on / off the values are 1.00 or 0.00 but with the scene it is 10.00

Feb-18-18 14:39:03 PM Debug:  MQTT: topic aqualinkd/Aux_2/set 1.00
Feb-18-18 14:39:03 PM Info:   MQTT: MATCH Aux_2 to topic aqualinkd/Aux_2/set
Feb-18-18 14:39:03 PM Info:   MQTT: received 'ON' for 'Aux_2', turning 'ON'
Feb-18-18 14:39:03 PM Debug:  Sent '0x0a' to controller
Feb-18-18 14:39:03 PM Debug:  00 10 02 00 01 00 0a 1d 10 03 00
Feb-18-18 14:39:03 PM Debug:  RS Message :- 'Air Blower ON'
Feb-18-18 14:39:03 PM Debug:  Ignoring 'Air Blower ON'
Feb-18-18 14:39:03 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:03 PM Info:   MQTT: Published id=35: aqualinkd/Aux_2 1
Feb-18-18 14:39:03 PM Info:   MQTT: Published id=36: domoticz/in {"idx":40,"nvalue":1,"svalue":""}
Feb-18-18 14:39:03 PM Info:   MQTT: Message publishing acknowledged (msg_id: 35)
Feb-18-18 14:39:03 PM Info:   MQTT: received (msg_id: 35), looks like my own message, ignoring
Feb-18-18 14:39:03 PM Debug:  MQTT: topic aqualinkd/Aux_2 1.00
Feb-18-18 14:39:03 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_2
Feb-18-18 14:39:03 PM Info:   MQTT: Message publishing acknowledged (msg_id: 36)
Feb-18-18 14:39:05 PM Debug:  RS Message :- 'Will turn OFF after 30 Minutes'
Feb-18-18 14:39:05 PM Debug:  Ignoring 'Will turn OFF after 30 Minutes'
Feb-18-18 14:39:05 PM Debug:  RS Message :- '9:37 AM'
Feb-18-18 14:39:05 PM Debug:  time not checked, will check in 3280 seconds
Feb-18-18 14:39:05 PM Debug:  RS time is accurate '9:37 AM 02/18/18 Sun'
Feb-18-18 14:39:05 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:07 PM Debug:  RS Message :- 'Air Temp 49�F'
Feb-18-18 14:39:07 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:09 PM Debug:  RS Message :- 'Pool Temp  53�F'
Feb-18-18 14:39:09 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:10 PM Debug:  RS Message :- 'AquaPure 55%'
Feb-18-18 14:39:10 PM Debug:  Ignoring 'AquaPure 55%'
Feb-18-18 14:39:11 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:12 PM Debug:  MQTT: topic aqualinkd/Aux_2/set 0.00
Feb-18-18 14:39:12 PM Info:   MQTT: MATCH Aux_2 to topic aqualinkd/Aux_2/set
Feb-18-18 14:39:12 PM Info:   MQTT: received 'OFF' for 'Aux_2', turning 'OFF'
Feb-18-18 14:39:12 PM Debug:  Sent '0x0a' to controller
Feb-18-18 14:39:12 PM Debug:  00 10 02 00 01 00 0a 1d 10 03 00
Feb-18-18 14:39:12 PM Debug:  RS Message :- 'Air Blower OFF'
Feb-18-18 14:39:12 PM Debug:  Ignoring 'Air Blower OFF'
Feb-18-18 14:39:12 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:12 PM Info:   MQTT: Published id=37: aqualinkd/Aux_2 0
Feb-18-18 14:39:12 PM Info:   MQTT: Published id=38: domoticz/in {"idx":40,"nvalue":0,"svalue":""}
Feb-18-18 14:39:12 PM Info:   MQTT: Message publishing acknowledged (msg_id: 37)
Feb-18-18 14:39:12 PM Info:   MQTT: received (msg_id: 37), looks like my own message, ignoring
Feb-18-18 14:39:12 PM Debug:  MQTT: topic aqualinkd/Aux_2 0.00
Feb-18-18 14:39:12 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_2
Feb-18-18 14:39:12 PM Info:   MQTT: Message publishing acknowledged (msg_id: 38)
Feb-18-18 14:39:14 PM Debug:  RS Message :- 'Salt 3000 PPM'
Feb-18-18 14:39:14 PM Debug:  Ignoring 'Salt 3000 PPM'
Feb-18-18 14:39:14 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:16 PM Debug:  RS Message :- '02/18/18 Sun'
Feb-18-18 14:39:16 PM Debug:  RS Received STATUS length 12.
Feb-18-18 14:39:16 PM Debug:  MQTT: topic aqualinkd/Aux_2/set 10.00
Feb-18-18 14:39:16 PM Info:   MQTT: MATCH Aux_2 to topic aqualinkd/Aux_2/set

END of LOG —  Crash

John Luber
john@luberfamily.net

sfeakes, 2018-02-18:
Set Temps are probably the labels again, (your's give different text over RS485 than mine) I'll double check my code, I know the problems in my code.
I can see the issue on buttons, I looks like the Aux and Names are mismatched.  I'll take a look when I get some time, but this may be a little harder to debug.  Just checking that you changed the aqualinkd config to match your setup with the Button Names to AUX labels?

johnluber, 2018-02-18:
Hi Shaun

I had made some label changes in aquiline.conf but they only appeared to change the labels in the web interface .  Since I’m not really wanting to use the web interface I’ve reverted those changes and am running with your default aqualinkd.conf file with the exception of debug mode.

For MQTT, there were some buttons that I did not want to expose to Homekit so I removed them from the json file.  So, I am running with the json file from your site except that I have removed:

>   "Aqualinkd Aux4": {
>     "id": "AqualinkdAux4Button",
>     "name": "Aux4 ",
>     "service": "Switch",
>     "manufacturer": "Feakes Inc",
>     "model": "AqualinkdAuxButton",
>     "topic": {
>       "setOn": "aqualinkd/Aux_4/set",
>       "statusOn": "aqualinkd/Aux_4"
>     },
>     "payload": {
>       "onTrue": 1,
>       "onFalse": 0
>     }
>   },
>   "Aqualinkd Aux5": {
>     "id": "AqualinkdAux5Button",
>     "name": "Aux5",
>     "service": "Switch",
>     "manufacturer": "Feakes Inc",
>     "model": "AqualinkdAuxButton",
>     "topic": {
>       "setOn": "aqualinkd/Aux_5/set",
>       "statusOn": "aqualinkd/Aux_5"
>     },
>     "payload": {
>       "onTrue": 1,
>       "onFalse": 0
>     }
>   },
>   "Aqualinkd Aux6": {
>     "id": "AqualinkdAux6Button",
>     "name": "Aux6",
>     "service": "Switch",
>     "manufacturer": "Feakes Inc",
>     "model": "AqualinkdAuxButton",
>     "topic": {
>       "setOn": "aqualinkd/Aux_6/set",
>       "statusOn": "aqualinkd/Aux_6"
>     },
>     "payload": {
>       "onTrue": 1,
>       "onFalse": 0
>     }
>   },
>   "Aqualinkd Aux7": {
>     "id": "AqualinkdAux7Button",
>     "name": "Aux7",
>     "service": "Switch",
>     "manufacturer": "Feakes Inc",
>     "model": "AqualinkdAuxButton",
>     "topic": {
>       "setOn": "aqualinkd/Aux_7/set",
>       "statusOn": "aqualinkd/Aux_7"
>     },
>     "payload": {
>       "onTrue": 1,
>       "onFalse": 0
>     }
>   },
>   "Aqualinkd PoolHeater": {
>     "id": "AqualinkdPoolHeater",
>     "name": "Pool Heater",
>     "service": "Thermostat",
>     "manufacturer": "Feakes Inc",
>     "model": "AqualinkDThermostat",
>     "topic": {
>       "setTargetTemperature": "aqualinkd/Pool_Heater/setpoint/set",
>       "statusTargetTemperature": "aqualinkd/Pool_Heater/setpoint",
>       "statusCurrentTemperature": "aqualinkd/Temperature/Pool",
>       "setTargetHeatingCoolingState": "aqualinkd/Pool_Heater/set",
>       "statusTargetHeatingCoolingState": "aqualinkd/Pool_Heater"
>     },
>     "payload": {},
>     "config": {}
>   },

If I’ve done something wrong please let me know.  I hope I haven’t done something that has caused you to do work on your part.  I am enjoying working on this with you and appreciate all of your help.  I’m excited to get this working.

FYI, my desire is more for controlling the hot tub and seeing status through Siri than actually programming the pool; just looking to simplify it for my wife.   I can use the web Zodiac interface for programming the pool.

John Luber
john@luberfamily.net

sfeakes, 2018-02-18:
As long as you have the Aux_? set correctly in the MQTT json file to match the buttons on the Homekit, then all should be fine.  All I have to go by is my setup and the really bad documentation from the pool equipment companies.  So this will be a bit of trial and error, I'll upload some code to general more important debug info when I get chance and we should have it sorted.  BTW a few days ago, homekitmqtt made some changes for spa & pool heaters for this project, now you can set HomeKit to heat only (no AC, no Auto), and have the correct set points in HomeKit.  I haven't published a new json file yet, but here is my heater setup, hopefully you can make the appropriate changes.

```
"Aqualinkd SpaHeater": {
    "id": "AqualinkdSpaHeater",
    "name": "Spa Heater",
    "service": "Thermostat",
    "manufacturer": "Feakes Inc",
    "model": "AqualinkDThermostat",
    "topic": {
      "setTargetTemperature": "aqualinkd/Spa_Heater/setpoint/set",
      "statusTargetTemperature": "aqualinkd/Spa_Heater/setpoint",
      "statusCurrentTemperature": "aqualinkd/Temperature/Spa",
      "setTargetHeatingCoolingState": "aqualinkd/Spa_Heater/set",
      "statusTargetHeatingCoolingState": "aqualinkd/Spa_Heater"
    },
    "props": {
      "TargetTemperature": {
        "maxValue": 40,
        "minValue": 4
      },
      "TargetHeatingCoolingState": {
        "validValues": [0, 1]
      }
    },
    "payload": {},
    "config": {}
  }
```

sfeakes, 2018-02-18:
@johnluber can you download the latest code, let it run for 3~4 mins without hitting it from the web or HomeKit (I need to see the programming messages), then run the command sequence that makes it crash.

On the unifivideo code, it's so you can use the motion detection on unifvideo server to post to home automation motion sensors and turn on lights etc.

johnluber, 2018-02-18:
Thanks for debug code.

Here’s the output:

sfeakes, 2018-02-18:
@johnluber   The file didn't post again..:).

johnluber, 2018-02-18:
Sorry about that.  Here it is:

Feb-18-18 19:29:32 PM Notice: Aqualink Daemon v0.8a
Feb-18-18 19:29:32 PM Notice: Config log_level         = 7
Feb-18-18 19:29:32 PM Notice: Config socket_port       = 80
Feb-18-18 19:29:32 PM Notice: Config serial_port       = /dev/ttyUSB0
Feb-18-18 19:29:32 PM Notice: Config web_directory     = /var/www/aqualinkd/
Feb-18-18 19:29:32 PM Notice: Config device_id         = 0x0a
Feb-18-18 19:29:32 PM Notice: Config mqtt_server       = localhost:1883
Feb-18-18 19:29:32 PM Notice: Config mqtt_dz_sub_topic = domoticz/out
Feb-18-18 19:29:32 PM Notice: Config mqtt_dz_pub_topic = domoticz/in
Feb-18-18 19:29:32 PM Notice: Config mqtt_aq_topic     = aqualinkd
Feb-18-18 19:29:32 PM Notice: Config mqtt_user         = (null)
Feb-18-18 19:29:32 PM Notice: Config mqtt_passwd       = (null)
Feb-18-18 19:29:32 PM Notice: Config mqtt_ID           = aqualinkd_b827ebbbc0
Feb-18-18 19:29:32 PM Notice: Config idx water temp    = 13
Feb-18-18 19:29:32 PM Notice: Config idx pool temp     = 14
Feb-18-18 19:29:32 PM Notice: Config idx spa temp      = 15
Feb-18-18 19:29:32 PM Notice: Config deamonize         = YES
Feb-18-18 19:29:32 PM Notice: Config log_file          = /tmp/aqualinkd.log
Feb-18-18 19:29:32 PM Notice: Config light_pgm_mode    = 0.00
Feb-18-18 19:29:32 PM Notice: Config BTN Filter_Pump   = label Filter Pump     | dzidx 37
Feb-18-18 19:29:32 PM Notice: Config BTN Spa_Mode      = label Spa Mode        | dzidx 38
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_1         = label Cleaner         | dzidx 39
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_2         = label Waterfall       | dzidx 40
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_3         = label Spa Blower      | dzidx 41
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_4         = label Pool Light      | dzidx 42
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_5         = label Spa Light       | dzidx 43
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_6         = label NONE            | dzidx 0
Feb-18-18 19:29:32 PM Notice: Config BTN Aux_7         = label NONE            | dzidx 0
Feb-18-18 19:29:32 PM Notice: Config BTN Pool_Heater   = label Pool Heater     | dzidx 44
Feb-18-18 19:29:32 PM Notice: Config BTN Spa_Heater    = label Spa Heater      | dzidx 56
Feb-18-18 19:29:32 PM Notice: Config BTN Solar_Heater  = label Solar Heater    | dzidx 0
Feb-18-18 19:29:32 PM Debug:  process_id of child process 3073 
Feb-18-18 19:29:32 PM Notice: Starting web server on port 80
Feb-18-18 19:29:32 PM Notice: Starting MQTT client to localhost:1883
Feb-18-18 19:29:32 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Feb-18-18 19:29:32 PM Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827ebbbc0
Feb-18-18 19:29:32 PM Info:   MQTT: Connection acknowledged
Feb-18-18 19:29:32 PM Info:   MQTT: Subscribing to 'aqualinkd/#'
Feb-18-18 19:29:32 PM Info:   MQTT: Subscribing to 'domoticz/out'
Feb-18-18 19:29:32 PM Info:   MQTT: Subscription(s) acknowledged
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Filter_Pump 11.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Filter_Pump
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Spa_Mode 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Spa_Mode
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Aux_1 11.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_1
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Aux_2 11.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_2
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Aux_3 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_3
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Spa_Heater 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Spa_Heater
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Aux_4 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_4
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Aux_5 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Aux_5
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Pool_Heater 1.00
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Pool_Heater
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Temperature/Air 17.22
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Temperature/Air
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Temperature/Pool 12.78
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Temperature/Pool
Feb-18-18 19:29:32 PM Debug:  MQTT: topic aqualinkd/Temperature/Spa 12.79
Feb-18-18 19:29:32 PM Debug:  MQTT: ignoring topic aqualinkd/Temperature/Spa
Feb-18-18 19:29:37 PM Debug:  RS Received PROBE length 7.
Feb-18-18 19:29:37 PM Debug:  RS Received STATUS length 12.
Feb-18-18 19:29:37 PM Info:   MQTT: Published id=1: aqualinkd/Filter_Pump 0
Feb-18-18 19:29:37 PM Info:   MQTT: Published id=2: domoticz/in {"idx":37,"nvalue":0,"svalue":""}
[... 782 lines cut ...]
Feb-18-18 19:30:04 PM Debug:  Kicking thread
Feb-18-18 19:30:04 PM Debug:  Programming mode: found message '(null)' in 'All OFF      ~*'
Feb-18-18 19:30:04 PM Debug:  Find item in Menu: loop 17 of 25 looking for 'SET TIME' received message 'All OFF      ~*'
Feb-18-18 19:30:04 PM Debug:  Sent '0x18' to controller
Feb-18-18 19:30:04 PM Debug:  Programming mode: loop 1 of 1 looking for '(null)' received message 'All OFF      ~*'
Feb-18-18 19:30:04 PM Debug:  00 10 02 00 01 00 18 2b 10 03 00 
[... cut ...]
Feb-18-18 19:30:04 PM Debug:  Kicking thread
Feb-18-18 19:30:04 PM Debug:  Programming mode: found message '(null)' in 'Review       ~*'
Feb-18-18 19:30:04 PM Debug:  Find item in Menu: loop 19 of 25 looking for 'SET TIME' received message 'Review       ~*'
Feb-18-18 19:30:04 PM Debug:  Sent '0x18' to controller
Feb-18-18 19:30:04 PM Debug:  Programming mode: loop 1 of 1 looking for '(null)' received message 'Review       ~*'
Feb-18-18 19:30:04 PM Debug:  00 10 02 00 01 00 18 2b 10 03 00 
Feb-1

sfeakes, 2018-02-19:
@johnluber Just made a few changes, don't think I've fixed everything but it should be closer. Also I should be storing your SWG info (not posting it to HomeKit just yet).
Can you do the same as last time, get he latest version, run for a ~5 mins then post the Debug output.

johnluber, 2018-02-19:
Hi Shaun

Here’s the latest outputs.  I see the freeze protect temp is being read now.

{"type": "status","status":"Ready","version":"B0029223 REV T.202/19/18 Mon","date":"02/19/18 Mon","time":"1:49 AM","pool_htr_set_pnt":"-999","spa_htr_set_pnt":"-999","frz_protect_set_pnt":"35","air_temp":"49","pool_temp":" ","spa_temp":"56","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "on","Spa_Mode": "on","Aux_1": "off","Aux_2": "off","Aux_3": "on","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off"}}

It did crash more or less the same as before when setting the scene.  The log entries are bit more this time however.

Feb-19-18 01:46:17 AM Notice: Aqualink Daemon v0.8a
Feb-19-18 01:46:17 AM Notice: Config log_level         = 7
Feb-19-18 01:46:17 AM Notice: Config socket_port       = 80
Feb-19-18 01:46:17 AM Notice: Config serial_port       = /dev/ttyUSB0
Feb-19-18 01:46:17 AM Notice: Config web_directory     = /var/www/aqualinkd/
Feb-19-18 01:46:17 AM Notice: Config device_id         = 0x0a
Feb-19-18 01:46:17 AM Notice: Config mqtt_server       = localhost:1883
Feb-19-18 01:46:17 AM Notice: Config mqtt_dz_sub_topic = domoticz/out
Feb-19-18 01:46:17 AM Notice: Config mqtt_dz_pub_topic = domoticz/in
Feb-19-18 01:46:17 AM Notice: Config mqtt_aq_topic     = aqualinkd
Feb-19-18 01:46:17 AM Notice: Config mqtt_user         = (null)
Feb-19-18 01:46:17 AM Notice: Config mqtt_passwd       = (null)
Feb-19-18 01:46:17 AM Notice: Config mqtt_ID           = aqualinkd_b827ebbbc0
Feb-19-18 01:46:17 AM Notice: Config idx water temp    = 13
Feb-19-18 01:46:17 AM Notice: Config idx pool temp     = 14
Feb-19-18 01:46:17 AM Notice: Config idx spa temp      = 15
Feb-19-18 01:46:17 AM Notice: Config deamonize         = YES
Feb-19-18 01:46:17 AM Notice: Config log_file          = /tmp/aqualinkd.log
Feb-19-18 01:46:17 AM Notice: Config light_pgm_mode    = 0.00
Feb-19-18 01:46:17 AM Notice: Config BTN Filter_Pump   = label Filter Pump     | dzidx 37
Feb-19-18 01:46:17 AM Notice: Config BTN Spa_Mode      = label Spa Mode        | dzidx 38
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_1         = label Cleaner         | dzidx 39
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_2         = label Waterfall       | dzidx 40
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_3         = label Spa Blower      | dzidx 41
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_4         = label Pool Light      | dzidx 42
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_5         = label Spa Light       | dzidx 43
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_6         = label NONE            | dzidx 0
Feb-19-18 01:46:17 AM Notice: Config BTN Aux_7         = label NONE            | dzidx 0
Feb-19-18 01:46:17 AM Notice: Config BTN Pool_Heater   = label Pool Heater     | dzidx 44
Feb-19-18 01:46:17 AM Notice: Config BTN Spa_Heater    = label Spa Heater      | dzidx 56
Feb-19-18 01:46:17 AM Notice: Config BTN Solar_Heater  = label Solar Heater    | dzidx 0
Feb-19-18 01:46:17 AM Debug:  process_id of child process 3836 
Feb-19-18 01:46:17 AM Notice: Starting web server on port 80
Feb-19-18 01:46:17 AM Notice: Starting MQTT client to localhost:1883
Feb-19-18 01:46:17 AM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Feb-19-18 01:46:17 AM Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827ebbbc0
Feb-19-18 01:46:17 AM Info:   MQTT: Connection acknowledged
Feb-19-18 01:46:17 AM Info:   MQTT: Subscribing to 'aqualinkd/#'
Feb-19-18 01:46:17 AM Info:   MQTT: Subscribing to 'domoticz/out'
Feb-19-18 01:46:17 AM Info:   MQTT: Subscription(s) acknowledged
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Filter_Pump 11.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Filter_Pump
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Spa_Mode 11.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Spa_Mode
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Aux_1 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Aux_1
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Aux_2 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Aux_2
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Aux_3 11.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Aux_3
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Spa_Heater 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Spa_Heater
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Aux_4 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Aux_4
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Aux_5 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Aux_5
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Pool_Heater 1.00
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Pool_Heater
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Temperature/Air 9.44
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Temperature/Air
Feb-19-18 01:46:17 AM Debug:  MQTT: topic aqualinkd/Temperature/Pool 13.89
Feb-19-18 01:46:17 AM Debug:  MQTT: ignoring topic aqualinkd/Temperature/Pool
[... 760 lines cut ...]
Feb-19-18 01:50:44 AM Debug:  RS Message :- 'Air Temp 49ﬂF'
Feb-19-18 01:50:44 AM Debug:  RS Received STATUS length 12.
Feb-19-18 01:50:46 AM Debug:  RS Message :- 'Spa Temp   56ﬂF'
Feb-19-18 01:50:46 AM Debug:  RS Received STATUS length 12.
Feb-19-18 01:50:48 AM Debug:  RS Message :- 'AquaPure 15%'
[... cut ...]
Feb-19-18 01:50:56 AM Debug:  RS Received STATUS length 12.
Feb-19-18 01:50:58 AM Debug:  RS Message :- 'Air Temp 49ﬂF'
Feb-19-18 01:50:58 AM Debug:  RS Received STATUS length 12.
Feb-19-18 01:50:59 AM Debug:  MQTT: topic aqualinkd/Spa_Heater/setpoint/set 37.80
Feb-19-18 01:50:59 AM Info:   MQTT: request to set spa heater setpoint to 37.80c
Feb-19-18 01:50:59 AM Debug:  MQTT: topic aqualinkd/Spa_Heater/set 10.00
Feb-19-18 01:50:59 AM Info:   MQTT: MATCH Spa_Heater to topic aqualinkd/Spa_Heater/set

johnluber, 2018-02-19:
Checked again and the set points are working for the pool and spa.

{"type": "status","status":"Ready","version":"B0029223 REV T.2","date":"","time":"2:04 AM","pool_htr_set_pnt":"37","spa_htr_set_pnt":"101","frz_protect_set_pnt":"35","air_temp":" ","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "on","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off"}}

csherida, 2018-02-19:
Perhaps dumb question but what wires do you use to connect the RS485 to the controller?  I assume some kind of shielded copper wire.

Thanks,
Chris

johnluber, 2018-02-19:
Hi Chris

Looking at what is in my panel it is essentially old style telephone wire.   2 pair solid 22g - red, green, yellow, and black.  

In my case I wired mine in with a single pair stranded shielded 22g wire - white and black.

John

johnluber, 2018-02-20:
> Thanks John - I had spare low voltage landscape wires that seem to do the trick. I connected the RS485 adapter and loaded up the drivers for my Windows 10 surface and set the COM port to COM2, 3, 4, etc but the AquaLink windows simulator doesn’t seem to find the control panel.
>
> I connected A/B to the 2nd and 3rd pins. After I installed the driver, I started to get flashy lights on the RA485 board but the software couldn’t find the control panel. I will connect up the Raspberry Pi tomorrow but was hoping to see something on my Windows machine.
>
> Has anyone else tried the Windows simulator?

Hi Chris

Unfortunately, I don’t have any Windows machines to test from.  All of my computers are Mac or RPi so that has made this a little more challenging.

I am working on what I think is  pretty cool script to automate turning on the spa.  I’m still working on it but the basic idea is to set all the elements of the spa and to do it intelligently.  So, for example, turn on the filter, set spa mode, and turn on the spa heater.  Then check spa temp until it reaches the set point and then turn on the spa blower and lights if it is dark outside.

That’s the idea anyway.  It seems to be working in theory; now just need to test thoroughly.

John

csherida, 2018-02-21:
That’s awesome John!

I was able to get past my issues. After a reboot and messing with the ports, Ianded on COM port 7 which seemed to work after a couple of times. I’m awaiting a mini HDMI converter for the Raspberry Pi and will setup aqualinkd from there.

csherida, 2018-02-23:
Shaun - very impressive!  I downloaded your code and ran the install.sh and fired up aqualinkd. It worked great even though it wasn’t connected to the pool. I got the web interface with no data (as I’d expect). 

However, after rebooting my RaspPi, I tried to fire up aqualinkd again and I get the message saying “Can not start web server on port 80.”  Is this correct or should I expect something different?

sfeakes, 2018-02-24:
If you ran the install, it will setup the pi so it automatically starts at boot.  So the cannot connect to port 80 is probably because you are trying to start it twice. (Unless you installed another web server). Have you tried hitting the web interface without ruining aqualinkd manually?

csherida, 2018-02-24:
Hey Shaun - you are correct!  Thanks!  I’ll be plugging in today to panel. I found a 5v Molex to USB connector that I will try for power as well. Will let you know how it goes.

johnluber, 2018-02-24:
Hey Chris

Sounds like you’re making some good progress.  I’ve been continuing to work on my scripts and am at a point where I am finally testing them with Homekit.

I’m curious about the 5v Molex you found.  Can you share a link for it?

Thanks.

John

csherida, 2018-02-24:
Unfortunately, I was not initially successful. The web interface showed it was not connected to the equipment and while there was a solid light on the RS485, nothing was flashing (like it did when connected to my PC).

I will run it in debug mode tomorrow and grab the logs. Was there another configuration step I may have missed?  I think I have the Aqualinkd RS 8. We just got it installed.

csherida, 2018-02-24:
Thanks John. I haven’t tried it out yet (want to at least get the web interface working first) but here’s the link for the Molex to USB cable: Coolerguys 4 pin Molex to Dual USB 5V Power Connecter https://www.amazon.com/dp/B006JCHN08/ref=cm_sw_r_cp_apip_NngDwgVe2d4Yu

johnluber, 2018-02-25:
Hi Chris

Thanks for sharing the link.  It wasn’t quite what I was looking for but it triggered me to do some searching…

What I want to do is eliminate the need for 110v to power the RPi.  So, searching I found this https://www.ebay.com/itm/DC-DC-Buck-Step-Down-Converter-6V-24V-12V-to-5V-3A-USB-Charger-Power-Supply-DIY/262756388778?_trkparms=aid%3D222007%26algo%3DSIM.MBE%26ao%3D1%26asc%3D20160908110712%26meid%3Db8eda4ca74a64405b07ed05b5033e449%26pid%3D100677%26rk%3D12%26rkt%3D30%26sd%3D252925160260%26itm%3D262756388778&_trksid=p2385738.c100677.m4598 <https://www.ebay.com/itm/DC-DC-Buck-Step-Down-Converter-6V-24V-12V-to-5V-3A-USB-Charger-Power-Supply-DIY/262756388778?_trkparms=aid=222007&algo=SIM.MBE&ao=1&asc=20160908110712&meid=b8eda4ca74a64405b07ed05b5033e449&pid=100677&rk=12&rkt=30&sd=252925160260&itm=262756388778&_trksid=p2385738.c100677.m4598>  If it works I should be able to grab the 10v off the red connector from the RS8 and step it down to 5v 3a to power the RPi.  Given the price of this item I’m not banking on it too much but conceptually it should do the job.

BTW, for the first time ever I picked up my phone and said, “Hey Siri, turn on the hot tub” and it’s arming up right now! 

Thanks Shaun for your code!  By itself it’s fantastic!  The fact that I can extend it with scripting makes it AWESOME!

John Luber
john@luberfamily.net

csherida, 2018-02-25:
I completely agree with you John - nice work Shaun!  Hopefully I'm not far behind you.

I enabled serial connections on the pi and switched the device_id to 0x09 and the lights started flashing on the RS485.  In addition the battery icon in the top/right of the web interface went to a small red line (versus the full green battery icon).

However, it still stated that there was no connection.  I ran it in DEBUG mode and hit the interface again (log file attached).  I'm guessing my next approach will be to try a few different device_ids.   I feel like there's something I'm still missing.

Apologies for being a pest but appreciate any thoughts you may have.

[aqualinkd.2018-02-24.log](https://github.com/sfeakes/aqualinkd/files/1755717/aqualinkd.2018-02-24.log)

sfeakes, 2018-02-25:
@csherida Your problem at the moment looks to be the usb connection to the RS485, there doesn’t seem to be anything happening over that connection.  Are you sure /dev/ttyUSB0 is the correct port for your setup?

johnluber, 2018-02-25:
I meant to reply this morning.  I suspect that’s the issue as well.  Depending on which USB port you plug into you get a different dev.  If you check you probably have a /dev/ttyUSB1 and not 0.  If you move the RS485 to a different port you should get a /dev/ttyUSB0.

John

csherida, 2018-02-25:
Thanks gents.  I will order a new cord just in case but I see the the file ttyUSB0 in /dev (screenshot attached).  I followed your directions Shaun.  I assume you wanted me to stop running as a daemon and capture the log files.  I also attached a screenshot of the web UI.  Appreciate any other thoughts.
[image]
[image]
[aqualinkd.2018-02-25a.log](https://github.com/sfeakes/aqualinkd/files/1756632/aqualinkd.2018-02-25a.log)

sfeakes, 2018-02-25:
There is no reason to not run it as a daemon, just make sure debug is on as well as log to file.

csherida, 2018-02-26:
Thanks Shaun.  Unfortunately, the same result.  I followed your steps on waiting before hitting the web interface but the same results..  I tested again with my Windows emulator to confirm nothing happened with the cable/card and it worked fine.  I will order a new USB cable anyways because I have an adapter in there.

I also turned on the Serial switch in the Pi configuration to see if that would make a difference but no luck.

Looking at the log file, nothing stands out to me.

I will try again tomorrow perhaps trying different device_ids.  Thanks again gents.
[aqualinkd.2018-02-25c.log](https://github.com/sfeakes/aqualinkd/files/1757108/aqualinkd.2018-02-25c.log)

csherida, 2018-03-03:
Once again, apologies for the nag but same luck as today.  Went back through the install steps and believe everything is in the same location.  No errors posted to the daemon logs (like the ttylUSB0 missing like before).  Is there something in the attached log files I don't see?  Should I just skip trying to connect to the web interface?
[aqualinkd.2018-03-03a.log](https://github.com/sfeakes/aqualinkd/files/1777572/aqualinkd.2018-03-03a.log)

johnluber, 2018-03-03:
Hey Chris

If you enter this in your browser http://aqualinkd.ip.address?command=status <http://aqualinkd.ip.address/?command=status> replacing aqualinkd.ip.address with your rpi address does it return any data?

John Luber
john@luberfamily.net

sfeakes, 2018-03-03:
@csherida  There is a problem with seeing anything over the serial interface.  I’ll need to add some debug statements in the initializing function to get any information.  That will take some time.  I can see in the log it’s been sitting there for over 10mins and not seen anything it understands, what’s strange is it should have reset the connection within that time if it saw nothing at all, and since it didn’t, I’m assuming it may be seeing data, but it either doesn’t understand it, or it’s labeled to a different device on the rs485 buss.

csherida, 2018-03-03:
Thanks Shaun. I got the wrong cable but I think I think it’s working fine as it connects to the PC okay. Here’s the JSON from the request: {"type": "status","status":"Ready","version":"","date":"","time":"","pool_htr_set_pnt":"-999","spa_htr_set_pnt":"-999","frz_protect_set_pnt":"-999","air_temp":" ","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "off","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off"}}

csherida, 2018-03-03:
Shaun - should I have the Serial option disabled for the pi?  I have it enabled.

csherida, 2018-03-04:
Gents - I tried with a different Raspberry Pi (3) and same result :(.  I confirmed ttylUSB0 is there and see flashing lights on the RS485 board.  I attached the logs and the status.

Shaun - I am happy to debug.  Are there good visual debuggers out there for this or is it just throwing in DEBUG statements?
[aqualinkd.2018-03-04.status.log](https://github.com/sfeakes/aqualinkd/files/1778700/aqualinkd.2018-03-04.status.log)
[aqualinkd.2018-03-04.log](https://github.com/sfeakes/aqualinkd/files/1778701/aqualinkd.2018-03-04.log)

sfeakes, 2018-03-05:
@csherida I've added some extra information to be able to debug the serial connection.  Please download the latest release and make sure you have the following in your config file
```log_level=DEBUG_SERIAL```

Then run it for a few minutes and post the output.

It still won't work, but I may be able to tell what's going on from the log.

csherida, 2018-03-06:
Thanks Shaun.  I gave it a go tonight with the same result.  I tried changing the device_id as well as turning on/off the Serial option for the Pi but no luck.  I don't see something in the logs but hoping you do.  Thanks again!
[aqualinkd.2018-03-05.log](https://github.com/sfeakes/aqualinkd/files/1783619/aqualinkd.2018-03-05.log)

sfeakes, 2018-03-07:
@csherida you either didn't load/use the latest code, or you missed the log_level=DEBUG_SERIAL in the config file.

I've uploaded a new version with a few more debug statements.
From your logs you have
```
Mar-06-18 00:05:51 AM Notice: Aqualink Daemon v0.8a
Mar-06-18 00:05:51 AM Notice: Config log_level         = 7
```

You should have
```
Aqualink Daemon v0.9a
Config log_level         = 8
```

It may be easier to load the new code and run it manually.  Once downloaded
```
sudo systemctl stop aqualinkd
sudo <path to aqualinkd>\aqualinkd -d -c <path to cfg>\aqualinkd.conf
```
That should output messages to console AND log file. You should see a lot more information.

sfeakes, 2018-03-07:
I missed this

> Shaun - I am happy to debug. Are there good visual debuggers out there for this or is it just throwing in DEBUG statements?

csherida, 2018-03-07:
Apologies (once again) - I got the latest code but missed the DEBUG_SERIAL direction.  :(  I made that fix earlier tonight am getting messages like the ones below which makes me thing the device_id is off:
othing read on serial
Mar-06-18 20:36:09 PM Debug:  Serial read 7 bytes
Mar-06-18 20:36:09 PM Debug:  Received Packet, but not for us, ID is for 0x78
Mar-06-18 20:36:09 PM Debug:  Serial read 9 bytes
Mar-06-18 20:36:09 PM Debug:  Received Packet, but not for us, ID is for 0x00
Mar-06-18 20:36:09 PM Debug:  Serial read 7 bytes
Mar-06-18 20:36:09 PM Debug:  Received Packet, but not for us, ID is for 0x81
Mar-06-18 20:36:09 PM Debug:  Nothing read on serial
Mar-06-18 20:36:09 PM Debug:  Nothing read on serial
Mar-06-18 20:36:09 PM Debug:  Nothing read on serial
Mar-06-18 20:36:09 PM Debug:  Nothing read on serial
Mar-06-18 20:36:09 PM Debug:  Serial read 7 bytes
Mar-06-18 20:36:09 PM Debug:  Received Packet, but not for us, ID is for 0x33

sfeakes, 2018-03-07:
ok this is good, it's reading the information on the RS485 buss, but for some reason it's not registered correctly, so the controller is not sending messages to the PI directly.  Can you run if for a minute and upload the complete log?  Also what other devices (pool equipment) is connected to the pool controller?  ie keypads / SWG / Heater etc?

csherida, 2018-03-07:
Thanks again Shaun.  I did the make install and confirmed version 0.9a and the logs are attached.  As far as equipment goes, this is on the second RS485 terminal bar as the link for the remote PDA is on the first one.  I'll attach a picture on my next post.
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1787759/aqualinkd.log)

csherida, 2018-03-07:
[image]
The first terminal bar goes to the PDA I believe and we have a spa, heater, etc.

johnluber, 2018-03-07:
Hey Chris

Do you use your PDA to program (ie; set aquapure, temp setpoints, run schedules, etc)?  Do you use the IOS/Android or web app at all?

Here was my situation.

I had the PDA as my primary control point.  Meaning all programming was initiated from it.  I also had the IOS app but it could only toggle devices on and off.

Not all RS units are created equal.  There’s a daughter card on the back of the panel that determines what you can do.

For example, in the PDA as the control point world there is one daughter card.  If you can control from the app or web then there’s  a different daughter card.

When I first tried to use aqualinkd it didn’t work and it was because of the fact that I had the PDA daughter card.  My pool guy happened to be out and we were talking about it.  While he was there he swapped out my daughter card to an RS4 ( that’s the smaller one; there’s also a 6 and 8 ).  Once swapped aqualinkd came right and worked.

So, this might be your situation also.

John

csherida, 2018-03-08:
Interesting point John. I do control all settings from the PDA remote but I don’t have connectivity to the iOS app.  However, because the same terminal port works through the Windows app, it makes me think the box is okay and it’s something in the configuration. 

My remote also says the model is PDA-PS4 which leads me to believe it’s the same card in the box. 

I am tempted to disconnect the first port and plug the aqualinkd into that to see if that changes things but want to do it when it’s daylight.

johnluber, 2018-03-08:
Hi Chris

From what you’re saying I think your setup is similar to or the same as what I had before my pool guy changed my daughter card.   Unfortunately, I don’t know the model card that I had though the next time I see Adam I can ask.  I know I have the RS4 card now though.  This is the card that can interface to the All Button which is the controller type that Shaun describes on his Github site.

It makes sense to me that the Window app written by Jandy/Zodiac would work with all of the various board types made.  However, and I could be wrong on this, I believe that Shaun’s app is written specifically to his environment and possibly nothing more.  It sounds like he’s gone to great lengths through trial and error to reverse engineer the protocols to his setup.  So, it makes sense that it would not work with all of the various boards made by Jandy/Zodiac.

Shaun, I don’t mean to put words in your mouth.   I very well could be wrong in my interpretation.  But, it seems likely that you haven’t had a great variety of configurations to test /debug with.  Most of us only own one pool and I suspect the user community for an app such as this is somewhat limited as well.

Jandy seems to go to great lengths to make all of this sufficiently confusing to homeowners so that the authorized installers maintain their business.  From what I think I understand the main cabinet is referred to as RS and then the boards inside can be PS or RS and control different numbers of relays.  On top of that there is the confusion around what rev chips are required for various functionality.  They really seem to be very anti DIY.

John

[31 more replies]

closed 2018-07-15 by sfeakes