aqualinkd/AqualinkD #3: One Touch

johnluber, 2018-03-07:
Is it possible to toggle a One Touch button using aqualinkd?

sfeakes, 2018-03-07:
Not quite sure what you mean.  You can do everything with this that you can with the one touch control panel.

sfeakes, 2018-03-09:
John,

The only multiple keypress options that aqualinkd supports are fixed ones. 
setting the time & date
retrieving & setting spa heater temp
retrieving & setting pool heater temp
retrieving & setting freeze set point
Light mode (at present very specific, I will make this more generic soon)

Custom Macros are designed to be created by HomeKit / Domoticz / other automation hub.  So if a series of things you've put into HomeKit is causing a crash, that's a bug I need to fix.

Can you list the series of events you are trying to create in HomeKit, and set the logging to debug and post the output?

One thing comes to mind that may cause this is if you are trying to use one of the internal macros with a series of other key presses.  The internal macros will cause the controller to go into programming mode and there for can't accept other key-presses.  I think I've only set the buffer to remember one key press. But it shouldn't crash, it should just ignore subsequent requests.  So if you are setting a temprature within your macro, you could try to put a pause for maybe 10 seconds after it (before the next request), see if that fixes it.

johnluber, 2018-03-10:
Hi Shaun

I've run the aqualinkd in debug mode and created an scene in Homekit.  The automation is set to turn on the filter pump, lights, blower, heater and set spa mode on.

As soon as I activate the scene the aqualinkd segfaults.  Log file attached.  

John

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1798880/aqualinkd.log)

johnluber, 2018-03-10:
> Thanks. It’s a bug, but me fixing the bug probably won’t help you. It seems home kit is setting the spa mode to 10. Spa mode only takes a 1 or 0 (on or off). So I’ll fix the core dump to ignore anything other that 1 or 0, but you should look at why HomeKit is sending a 10 rather than a 1.
>
> I’ll Duplicate what you are doing with my setup over the weekend and see what I get, if it’s a 10 then I’ll probably be able to find out why and look for a solution.

I kind of thought the cause.  Could it be happening in mqtt?

John Luber
john@luberfamily.net

sfeakes, 2018-03-10:
Probably an easy test.  Hit the spa button on the HomeKit app, (not in Marco mode). If it still sends a 10 chances are it’s MQTT or the config in some way.  If it’s a 1 then it’s homekit.

Button press on HomeKit app sends a 1 on my setup, but I have never tried HomeKit macros.

johnluber, 2018-03-10:
OK, did a couple of more tests.

Pressing just Spa Mode like you suggested sent a 0x01.  

Debug:  Sent '0x01' to controller

Then, I modified my scene to remove Spa Mode and left just the filter pump and the blower.  This time I got:

Debug:  MQTT: topic aqualinkd/Filter_Pump/set 10.00
Info:   MQTT: MATCH Filter_Pump to topic aqualinkd/Filter_Pump/set

I thought I seen this behavior testing with it before.  Not sure what to make of it.

John Luber
john@luberfamily.net

sfeakes, 2018-03-10:
```
Debug:  MQTT: topic aqualinkd/Filter_Pump/set 10.00
Info:   MQTT: MATCH Filter_Pump to topic aqualinkd/Filter_Pump/set
```

Above are the lines you are looking for, the Send 0x01 is what Aqualinkd sends to the controller, so not important for this issue.
Hopefully if you just hit the button (no macro) what you should see is
```
Debug:  MQTT: topic aqualinkd/Filter_Pump/set 1
```

I just checked the code, and I do have tests in there for invalid numbers.  So this might be a little harder to decipher.   Can you try sending the MQTT commands directly?  On the machine where you have mosquitto installed, this is how you can send them directly without HomeKit.
```
mosquitto_pub -h localhost -m 1 -t '/aqualinkd/Spa_Mode/set'
```
-m is the message, so 1 or 0
-t it the topic.  looks like your config is /aqualinkd/Spa_Mode/set

johnluber, 2018-03-10:
HeSending it directly as you specified didn’t appear to do anything.  No output and the spa didn’t turn on.

With the -d on I got:

pi@raspberrypi:/var/log/mosquitto $ mosquitto_pub -d -h localhost -m 1 -t '/aqualinkd/Spa_Mode/set'
Client mosqpub/902-raspberrypi sending CONNECT
Client mosqpub/902-raspberrypi received CONNACK
Client mosqpub/902-raspberrypi sending PUBLISH (d0, q0, r0, m1, '/aqualinkd/Spa_Mode/set', ... (1 bytes))
Client mosqpub/902-raspberrypi sending DISCONNECT

sfeakes, 2018-03-10:
Crap.  There is no leading / in the topic.  Sorry, it should be. 

mosquitto_pub -h localhost -m 1 -t 'aqualinkd/Spa_Mode/set'

If that doesn’t work, cut n paste the mqtt topic from the aqualinkd config file.

johnluber, 2018-03-10:
No output on the command line but the aqualinkd.log ha:

Mar-09-18 22:01:09 PM Debug:  MQTT: topic aqualinkd/Spa_Mode/set 1.00
Mar-09-18 22:01:09 PM Info:   MQTT: MATCH Spa_Mode to topic aqualinkd/Spa_Mode/set
Mar-09-18 22:01:09 PM Info:   MQTT: received 'ON' for 'Spa_Mode', turning 'ON'
Mar-09-18 22:01:09 PM Debug:  Sent '0x01' to controller

johnluber, 2018-03-10:
Interesting that even though the command seems successful the spa mode does not aurally activate at the pool.

sfeakes, 2018-03-10:
Try turning on the pool pump.  spa mode just turns the actuators on the valves, won’t see much without the pump also being on.

I’m assuming all this works from using the buttons in HomeKit and web interface.?

johnluber, 2018-03-10:
I was judging whether or not it actually worked or not based on what the iAquaLink program showed.  Right now my pump is being controlled by freeze protect so I can’t toggle it.

But, now the iAquaLink seems to reflect that it is working.

sfeakes, 2018-03-10:
Ok glad that works.  I’ll have a play over the weekend with HomeKit macros and see if I can reproduce the problem.

johnluber, 2018-03-10:
Awesome!  and thanks!   Let me know I you ‘d like me to test anything else.

sfeakes, 2018-03-10:
I've updated the code.  It shouldn't core dump now.  But it will ignore a filter pump mode of 10, so it's not a total fix to your HomeKit problem. (I haven't tried Macro's yet).  But, run the new code turn the pump and spa on AND off from either the web interface or a HomeKit button with no macro.  Then try your macro and see if it works. I've got an inclinin that the MQTT cache has a bad value in it.

I've also added some code to capture your SQG information.  It won't publish the messages yet, but if you could run aqualinkd in debug mode and post the logs, it'll let me know if I've at least got the capture part right.

johnluber, 2018-03-11:
Hi Shaun

Thanks for what you're doing.  I appreciate it very much!

I installed last night and tried it a bit.  It does appear to be ignoring the mode 10 on the filter pump now.  I'll need to test some more and send you the logs from that.  Don't have my IOS devices handy right now.

I am attaching logs of a plain run where you can see the SWG info.

Thanks again,

John
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1800358/aqualinkd.log)

sfeakes, 2018-03-12:
@johnluber I've fixed the HomeKit error and now added code to publish SWG Percent and PPM, but I'm at a loss as to why I'm not capturing the SWG information from your setup. So I haven't uploaded the new code yet as I want to either fix this or add more debug information before you try.  So can you tell me if you are using the binary I upload, or are you compiling aqualinkd on your system?  Answering that may give me some insight.

Thanks
Shaun

sfeakes, 2018-03-12:
@johnluber I uploaded the code anyway. (made a small change).  Please try and report back with log.  Homekit macro & SWG should now be working.
SWG information should be published to the below MQTT topics
```
aqualinkd/SWG/Percent
aqualinkd/SWG/PPM
```

I haven't looked at which HomeKit option within HomeKit2MQTT would be best consume this information yet.  But it should also be available over the json web return.  http://aqualink_IP/?command=status

johnluber, 2018-03-12:
@sfeakes 
Thanks for the update.  I am running your pre-compiled binary, btw.

I'm getting proper SWG data and Salt ppm in the web status:

{"type": "status","status":"Ready","version":"B0029223 REV T.203/12/18 Mon","date":"03/12/18 Mon","time":"6:07 PM","pool_htr_set_pnt":"37","spa_htr_set_pnt":"100","frz_protect_set_pnt":"37","air_temp":"35","pool_temp":"46","spa_temp":" ","swg_percent":"50","swg_ppm":"3200","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "on","Spa_Mode": "off","Aux_1": "on","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off"}}

The macro (turn on Aux2, Aux3, Filter pump, Heater, and Spa Mode) seems to be throwing errors in the log:

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1804742/aqualinkd.log)

John

sfeakes, 2018-03-12:
Thanks.  Glad that the SWG is finally working.  I see it's throwing errors,  I'm sure that's my mistake in trying to quickly solve this rather than coding it correctly.   I'll get an update out to fix this.

sfeakes, 2018-03-12:
@johnluber OK, think I've fixed that, please download and try again.  (Pleased the binary works for you, and that's what you are using).
One thing to note, aqualinkd takes a few minutes to sort itself out when you first start it. This is because it has to set the control panel into programming mode to get additional information. It's best not to try to send commands while it's doing this.  
If you load the web interface you'll get a status messages at the top saying (programming please wait), or if you monitor the log, wait for the "programming mode" messages to stop.  Once that's happened, then try the macros.  (They should work in programming mode, but it's probably best not to try at the moment).

Shaun

johnluber, 2018-03-13:
Hey Shaun

Did some testing with the latest...  And, making progress.  I'm some wonkiness and I think it's related to freeze protect mode - it's snowing here in NC right now.  Seems the Jandy app can overcome freeze protect mode but aqualinkd can't for some reason.

I've attached logs, but I noticed I can turn on most items individually except that spa mode would be ignored.  I also had trouble with aux2 (air bubbles).  Once I turned on spa mode from the Jandy app then aux2 worked.

I tried setting a scene as well.  Some of it worked but not all.  It toggling the scene the app crashed.

I've attached the log for your reference.

John

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1805002/aqualinkd.log)

sfeakes, 2018-03-13:
Thanks for the logs, I can see why spa mode (and other items) won't override freeze-protect as it expects another keypress to confirm the override, (and I have not programmed that into the sequence).  My freeze protect comes on about once in a blue moon since I currently live in Houston. I also have it disabled and use my own freeze protect, as Jandy's did't cycle through all the equiptment as I needed it to on my setup.  But I also don't have the kind-a "deep freeze" that your guys get.
I'll go over the logs in more detail tomorrow and see what I can come up with as a solution.

Thanks
Shaun

johnluber, 2018-03-13:
That’s terrific that you see the problem so easily.  It was much warmer a few weeks back when I did the initial testing so freeze protect wasn’t in play then.  Weird weather here lately...

johnluber, 2018-03-13:
Hey Shaun

Wondering if you’ve seen this program and the accompanying Jandy doc?  It’s basically a simple comandline serial interface to the RS.  I’ve tried it and it does work with my hardware.

Program itself is old and incomplete but it does work.  One thing that might be helpful if you’re not already is there is an ability to to change the response format coming back from the RS.  If you issue a #RSPFMT=1 command then all returns after are numeric only and easier to deal with programatically.

If nothing else, the doc may be useful to you.

John

sfeakes, 2018-03-13:
@johnluber Sounds interesting, can you link the doc?

sfeakes, 2018-03-13:
@johnluber I just added some code to override freeze protect.  I did this completely blindly, just from your logs as I can't get my equipment to go into freeze protect mode easily.  You will need to add an option to the aqualinkd.conf to enable it.
```
override_freeze_protect = yes
```

Also I noticed you have both domoticz and HomeKit enabled on MQTT.  This works fine, but if you are not using domoticz you may want to disable that.  In the config make sure the below are commented out to disable domoticz MQTT
```
#mqtt_dz_pub_topic = domoticz/in
#mqtt_dz_sub_topic = domoticz/out
```

FYI, HomeKit MQTT only needs these two lines. (user and passed are optional)
```
mqtt_address = server_name:1883
mqtt_aq_topic = aqualinkd
#mqtt_user = someusername    
#mqtt_passwd = somepassword
```

johnluber, 2018-03-13:
Hey Shaun

I’ll try this when I get hoe tonight.  

Here’s the doc I was referencing:  http://kb.homeauto.com/redirfile.asp?id=203 <http://kb.homeauto.com/redirfile.asp?id=203>

John

sfeakes, 2018-03-13:
Thanks for the doc.  Yes I had seen that, It was actually page 32/33/34 in that doc where I was finally able to deciphered the binary status packets of each of the devices. But I thought it was specifically for their own adapter which is RS485 to RS232, so I never actually tried sending those commands. I always thought you connect that device to your control panel and connect a serial port on a PC to that device (as you would a modem), then you send commands over RS232.
Are you saying you can send the RS232 commands listed in that document directly to the RS485 buss?

johnluber, 2018-03-13:
Yes, you absolutely can.  I used the aqualinkRS.py app that is on github to do so.  The doc is a little confusing in regards to command structure as the doc implies the use of spaces when, in fact, the spaces don’t work.  So, for example on page 11 the command is #aux1 CR not # aux1 CR.  And, you do have to type the #.

johnluber, 2018-03-13:
Hi Shaun

Did some quick testing with this tonight.  Wasn't able to test freeze protect because today it's warm enough that it is off and so I have to reprogram it to test.

I did run through some scenarios.  Generally, individual buttons work from home kit but scenes are still hit or miss.  I did get it to crash again when turning off a scene.  I've attached logs.

John

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1808983/aqualinkd.log)

sfeakes, 2018-03-14:
I can see the issue, a scene is sending commands that need to be programmed (which takes time over the RS485 interface). Since I'm not buffering requests, subsequent requests either get lost, or override each other.  I control this on the web page interface, but can't on HomeKit.  I've made a few changes that may help, but the best solution is for me to cache requests better than I am.  Can you list out the scenes and commands you are sending within each scene.  That way I can tell how people are using it and program accordingly.
A quick solution would be to put anything that requires programming at the end of each scene, so make "spa setpoint" the last command.

johnluber, 2018-03-14:
Hey Shaun

Thanks for continuing to look at this.  I’m really only after one scene; to turn on the spa.  So, I’m toggling on the pump, spa mode, heater, lights, and bubbles.  The goal is to get to the point where my wife can control the spa with Siri and that’s about it.  Beyond that the only thing I’m really interested in is being able to display the salt ppm in HomeKit.  The other toggles are interesting and I might choose to use them later, but for now this is my focus.

Thanks.

John

sfeakes, 2018-03-14:
@johnluber I've made some changes to support this, while not that large (code wise), it's a big change to how I was communicating to the RS controller, so you may want to run through a lot of tests. For me the scene you listed above and also including a spa heater setpoint within that now all work.  Please re-test when you get chance. The RS control panel can't take the commands as quickly as HomeKit can send them, and that's what was causing the problems. So now I buffer all the commands and send them out slowly to the RS control panel. So you may see a delay in everything being activated. (only a few seconds though). 

BTW. What I did notice though is if the scene is asking the pool controller to do something stupid, it will just be ignored.  IE, if the scene turns on the spa blower before the spa mode, and your pool equipment has the blower locked out if it's not in spa mode, then the blower command will get ignored.  So ordering of the requests that HomeKit sends is (or can be) very important.

johnluber, 2018-03-14:
Thanks Shaun.  I’ll try this ASAP.  I’m not sure that I can set the order that HomeKit sends commands but I’ll look into it.  The only part might an issue is that I might to be in spa mode before the heater comes on, but I’ll have to test that to be sure.

johnluber, 2018-03-14:
OK, did a quick test and it's really good.  I ran the scene and everything turned on as expected.  When turned the scene off the spa heater remained on.  And, for the first time there was a discrepency between the iaqualink app and aqualinkd.  iaqualink show spa mode off but aqualinkd show spa mode on.

Logs attached.

Great progress!

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1813458/aqualinkd.log)

sfeakes, 2018-03-15:
This is great news.  I can see the discrepancy between iAqualink and aqualinkd as well.  When you turned the spa off, your system has a cool down period set.  So aqualinkd will show it as on (in HomeKit) until it's actually turned off completely.  In the aqualinkd web interface the cool down period is shown by a flashing red status light, but home kit only supports on & off not other status.  Obviously iAqualink has the cool down shown as off.   So, it looks as though all is working normally.
Next time, wait a few minutes for the cooldown period to complete and you should see everything correctly. (or pull up the web interface).

johnluber, 2018-03-15:
That makes sense. Ill try to do more testing tomorrow.

sfeakes, 2018-03-15:
I'm closing this out as it seems fixed.

closed 2018-03-15 by sfeakes

johnluber, 2018-03-15:
Hey Shaun

I agree; I believe it is fixed as well.  Great job!  Thank you so much!

One quick question, should the SWG data show up at this point in HomeKit?  I’m a bit confused as to whether this part is done or not.  If so, it’s not.  

Thanks again!

John Luber
john@luberfamily.net

sfeakes, 2018-03-15:
@johnluber  Sorry, I wasn't clear on that.  It is finished, but haven't looked at the best HomeKit / HomeKit2MQTT configuration just yet.

Salt information is published to MQTT topics of
```
aqualinkd/SWG/Percent
aqualinkd/SWG/PPM
``` 

So you should just need to find a HomeKit2MQTT accessory type that allows you to post plane values to it (Like the Air Temp), and configure it to show values posted on the above topics.

I am not sure on the best HomeKit accessory for these values, (and which ones are supported by the particular HomeKit app you are using) you may be best asking on the HomeKit2MQTT as a quick look I didn't see anything that would work all that well.

You could try the temperature one, but my bet is it'll either try to convert the values or say they are out of range.

But below is something to try quickly. (I haven't tested it), just add it to your HomeKit2MQTT son config and be careful with the commars ","   I think the Fahrenheit : true will stop any conversion being done, but you may need to delete that.
```
  "Aqualinkd Salt Percent": {
    "id": "AqualinkdSaltPercentSensor",
    "name": "Salt Percent",
    "service": "TemperatureSensor",
    "manufacturer": "Feakes Inc",
    "model": "AqualinkDSaltPPMSensor",
    "topic": {
      "statusTemperature": "aqualinkd/SWG/Percent"
    },
    "payload": {
      "fahrenheit", "true"
    },
    "config": {}
  },
    "Aqualinkd Salt Parts Per Million": {
    "id": "AqualinkdSaltPPMSensor",
    "name": "Salt PPM",
    "service": "TemperatureSensor",
    "manufacturer": "Feakes Inc",
    "model": "AqualinkDSaltPPMSensor",
    "topic": {
      "statusTemperature": "aqualinkd/SWG/PPM"
    },
    "payload": {
      "fahrenheit", "true"
    },
    "config": {}
  }
```

johnluber, 2018-03-15:
Cool!  Thanks.  I’ll let you know how this works out.

John Luber
john@luberfamily.net

johnluber, 2018-03-16:
> john@luberfamily.net <mailto:john@luberfamily.net>

Hey Shaun

I had a chance work this this morning and I did get the data to display in HomeKit.  I modified what you sent to be:
 

 "Aqualinkd Salt Percent": {
    "id": "AqualinkdSaltPercentSensor",
    "name": "Salt Percent",
    "manufacturer": "Feakes Inc",
    "model": "AqualinkDSaltPPMSensor",
    "services": [
      {
        "name": "Salt Percent",
        "service": "TemperatureSensor",
        "topic": {
          "statusTemperature": "aqualinkd/SWG/Percent"
        },
        "payload": {
          "fahrenheit": "true"
        },
        "config": {},
        "props": {}
      }
    ]
  },
  "Aqualinkd Salt Parts Per Million": {
    "id": "AqualinkdSaltPPMSensor",
    "name": "Salt PPM",
    "manufacturer": "Feakes Inc",
    "model": "AqualinkDSaltPPMSensor",
    "services": [
      {
        "name": "Salt PPM",
        "service": "TemperatureSensor",
        "topic": {
          "statusTemperature": "aqualinkd/SWG/PPM"
        },
        "payload": {
          "fahrenheit": "true"
        },
        "config": {},
        "props": {
                "CurrentTemperature": {
                        "maxValue": 4000,
                        "minValue": 0
                }
          }
        }
    ]
  },

Note there was a small syntax error with what you sent.  There needed to be a : after “fahrenheit” instead of a ,.  The additional props in Salt PPM allow for exceeding a temperature of 212.

The display win HomeKit is not perfect.  Salt % displays as degrees and Salt PPM is limited to the first 3 digits of the data although if you press it you  get the full field.

I guess I’m little confused on homekit2mqtt as some of Service Types I tried reported unsupported in HomeKit.  I thought the whole purpose of homekit2mqtt was to expand what is supported but I guess I am mistaken.  If there was support for all types then BatteryService looks like a good option for Salt %.

Thanks again for all of your help with this.  Open source does really work!  You’ve been awesome!

John Luber
john@luberfamily.net

sfeakes, 2018-03-16:
Thanks for the info, I'll add that.
As for "unsupported" this is where Apple missed the boat, the Home app that comes on the iphone/ipad does not support all the accessories that HomeKit does. That's why you have the unsupported message.  There are plenty of 3rd party Home apps that do support all the accessories, but none of them work as well as Apples Home app IMO.  For the standard aqualinkd stuff, I was particular in using devices that had support in Apple's home app, and that's why originally that the thermostats would have Heat, Cool & Off, but with a recent update I asked MQTT2HomeKit to modify for just Heat & Off.

There may be a better way to do the Salt, I'll have a look when I get time.

johnluber, 2018-03-16:
Gotcha.  Thanks for clarifying that point about supportability.

BTW, I forgot to mention but I’ve scripted a couple of things in bash that may be of interest to you or others as offshoots of this endeavor.

For example,  I’ve got a script that runs from cron once a week that texts me the salt ppm.  If it’s below 2900 then it also sets a reminder for me on the upcoming Saturday to buy and add salt.  Another one watches for Spa_Mode to be on the then waits until the set temp os reached.  Once that happens it triggers the lights, if appropriate, music and bubbles.  It also will retriever the bubbles after the 30 minute timeout.

All in all, I am very pleased with what your script has allowed me to do.

Thanks again!

John Luber
john@luberfamily.net