aqualinkd/AqualinkD #4: Cant get install.sh to run error free on RPi 3

BigDogJonx, 2018-03-13:
First downloaded the code and just dropped it into my RPi3 under the /home/pi/Downloads/aqualinkd.  Then just went to ./release folder and attempted to run the install.sh as shown below, but got three errors.

> pi@raspberrypi:/Downloads/aqualinkd/release $ sudo sh install.sh
install.sh: 7: install.sh: Bad substitution
install.sh: 24: install.sh: [[: not found
install.sh: 29: install.sh: [[: not found
Created symlink from /etc/systemd/system/multi-user.target.wants/aqualinkd.service to /etc/systemd/system/aqualinkd.service.

Tried to do the make, then sudo make install, but that failed and had many more errors. as shown below:

> pi@raspberrypi:/Downloads/aqualinkd $ make
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aqualinkd.c  -o aqualinkd.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c utils.c  -o utils.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c config.c  -o config.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aq_serial.c  -o aq_serial.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c init_buttons.c  -o init_buttons.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c aq_programmer.c  -o aq_programmer.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c net_services.c  -o net_services.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c json_messages.c  -o json_messages.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -c mongoose.c  -o mongoose.o
gcc -Wall  -lpthread -lm -D MG_DISABLE_MD5 -D MG_DISABLE_HTTP_DIGEST_AUTH -D MG_DISABLE_MD5 -D MG_DISABLE_JSON_RPC  -o ./release/aqualinkd aqualinkd.o utils.o config.o aq_serial.o init_buttons.o aq_programmer.o net_services.o json_messages.o mongoose.o  -lpthread -lm
pi@raspberrypi:/Downloads/aqualinkd $ sudo make install
./release/install.sh
make: execvp: ./release/install.sh: Permission denied
Makefile:51: recipe for target 'install' failed
make: *** [install] Error 127

So going back to install.sh, trying to understand what these three errors are of 7, 24, 29, not sure if they line numbers, even if they are that part doesnt make sense looking at the install.sh.

> pi@raspberrypi:/Downloads/aqualinkd/release $ sudo sh install.sh
install.sh: 7: install.sh: Bad substitution
install.sh: 24: install.sh: [[: not found
install.sh: 29: install.sh: [[: not found
Config exists, did not copy new config, you may need to edit existing! /etc/aqualinkd.conf
Defaults exists, did not copy new defaults to /etc/default/aqualinkd
Avahi/mDNS defaults exists, did not copy new defaults to /etc/avahi/services//aqualinkd.service
Starting daemon aqualinkd
Job for aqualinkd.service failed. See 'systemctl status aqualinkd.service' and 'journalctl -xn' for details.
pi@raspberrypi:~/Downloads/aqualinkd/release $ systemctl status aqualinkd.service
● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled)
   Active: failed (Result: exit-code) since Tue 2018-03-13 14:29:35 EDT; 1min 36s ago
  Process: 1431 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=203/EXEC)

BigDogJonx, 2018-03-13:
I did run the install.sh a different way, didnt use sh, just ran it as ./install.sh and didnt get any errors as shown below, however the service fails to stsart as shown at the bottom

> pi@raspberrypi:/Downloads/aqualinkd/release $ sudo ./install.sh
Config exists, did not copy new config, you may need to edit existing! /etc/aqualinkd.conf
Defaults exists, did not copy new defaults to /etc/default/aqualinkd
Avahi/mDNS defaults exists, did not copy new defaults to /etc/avahi/services//aqualinkd.service
Starting daemon aqualinkd
Job for aqualinkd.service failed. See 'systemctl status aqualinkd.service' and 'journalctl -xn' for details.

> pi@raspberrypi:~/Downloads/aqualinkd/release $ systemctl status aqualinkd.service
> ● aqualinkd.service - Aqualink RS daemon
>    Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled)
>    Active: failed (Result: exit-code) since Tue 2018-03-13 14:45:23 EDT; 21s ago
>   Process: 1562 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=203/EXEC)

sfeakes, 2018-03-13:
First error looks like you are trying to run it in the wrong shell. /bin/sh, it's designed for bash not sh. 

Try
```
sudo install.sh      <--- no sh
```

All the other errors look like they are permission based, do you have permission to execute the install script?  Maybe try the below
```
sudo chmod a+x <path>/release/install.sh
```

sfeakes, 2018-03-13:
OK.  I read the next post.  Seems you have it installed now.   As for it not executing, try to run it manually and see what errors there are.
```
sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
```

BigDogJonx, 2018-03-13:
Seems like it thinks its not there, however the aqualinkd is there when I go to /usr/local/bin

> 
> pi@raspberrypi:~/Downloads/aqualinkd/release $ sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
> sudo: /usr/local/bin/aqualinkd: command not found

sfeakes, 2018-03-13:
check the permissions on /usr/local/bin/aqualinkd.  ie ls -al /usr/local/bin/aqualinkd, make sure you can execute it.

sfeakes, 2018-03-13:
Yes, it should be executable, something like the below
```
-rwxr-xr-x 1 root staff 217016 Mar 12 18:17 /usr/local/bin/aqualinkd
```

BigDogJonx, 2018-03-13:
Yeap that did the trick when I added the +x via chmod.  Now its running but waiting for the webpage to show up, right now just showing "controller.html" on my chrome browser page itself, i think you mentioned in the other issue thread on how to wire it up something about waiting 10mins or so.

sfeakes, 2018-03-13:
It shouldn't take 10 mins if all is working.  You should get an display in the browser the same as on the homepage of this site with "programming please wait" as the status, then after a minute that should go and all should be working.   You may have the bad permissions again.  If you don't see that, check syslog for errors, or stop the deamon, run it manually and look at the output for errors.

```
cat /etc/messages | grep aqualinkd
-- or depending on system setup--
cat /etc/syslog | grep aqualinkd
```

To run it manually
```
sudo systemctl stop aqualinkd
sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
```

It you want more information, edit /etc/aqualinkd.conf and enable debug.

BigDogJonx, 2018-03-13:
Ok, so my gut tells me its fails on parts of the install.sh and didnt fully complete it and we may have a partial install.  So on that hunch I started looking at these following sections on your Note section under the install

> Note:-
> The install script is designed for systemd / systemctl to run as a service or daemon. If you are using init / init-d then don't run the install script, install manually and the init-d script to use is in the xtras directory. Manual install for init-d systems
> 
> copy ./release/aqualinkd to /usr/local/bin -------> this is present
> copy ./release/aqualinkd.conf to /etc ----------> this is present
> copy ./release/aqualinkd.service.avahi to /etc/avahi/services/aqualinkd.service ----------> this is present
> copy ./extras/aqualinkd.init.d to /etc/init-d/aqualink -------> not present
> copy recuesivley ./web/* to /var/www/aqualinkd/  ----------> this is present and all files in subdirectories
> sudo update-rc.d aqualinkd defaults

So only see one piece missing which is for the init.d but dont believe thats the issue, running it manually here is what I get:, seems to clearly get connected and grabs my panel version, labels off my aux inputs etc.  Just browser only shows a white page with the words "controller.html"  Here is the output of it run manually:

pi@raspberrypi:/var/www/aqualinkd $ sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
        Aqualink Daemon v0.9f
        Config log_level         = 6
        Config socket_port       = 80
        Config serial_port       = /dev/ttyUSB0
        Config web_directory     = /var/www/aqualinkd/
        Config device_id         = 0x0a
        Config override frz prot = NO
        Config mqtt_server       = (null)
        Config mqtt_dz_sub_topic = (null)
        Config mqtt_dz_pub_topic = (null)
        Config mqtt_aq_topic     = (null)
        Config mqtt_user         = (null)
        Config mqtt_passwd       = (null)
        Config mqtt_ID           = aqualinkd_b827eb508b
        Config idx water temp    = 13
        Config idx pool temp     = 14
        Config idx spa temp      = 15
        Config idx SWG Percent   = 0
        Config idx SWG PPM       = 0
        Config deamonize         = NO
        Config log_file          = (null)
        Config light_pgm_mode    = 0.00
        Config BTN Filter_Pump   = label Filter Pump     | dzidx 37
        Config BTN Spa_Mode      = label Spa Mode        | dzidx 38
        Config BTN Aux_1         = label Cleaner         | dzidx 39
        Config BTN Aux_2         = label Waterfall       | dzidx 40
        Config BTN Aux_3         = label Spa Blower      | dzidx 41
        Config BTN Aux_4         = label Pool Light      | dzidx 42
        Config BTN Aux_5         = label Spa Light       | dzidx 43
        Config BTN Aux_6         = label NONE            | dzidx 0
        Config BTN Aux_7         = label NONE            | dzidx 0
        Config BTN Pool_Heater   = label Pool Heater     | dzidx 44
        Config BTN Spa_Heater    = label Spa Heater      | dzidx 56
        Config BTN Solar_Heater  = label Solar Heater    | dzidx 0
        Starting web server on port 80
        Starting MQTT client to (null)
        Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
        URI request: '/ '
        Control Panel E0260801 REV R
        Getting pool & spa heat setpoints from aqualink
        URI request: '/ '
        Aqualink time is off by -429 seconds...
        RS time is NOT accurate '3:28 PM 03/13/18 TUE', re-setting on controller!
        Could not select REVIEW menu
        Setting time on aqualink
        Could not select SET TIME menu
        Getting freeze protection setpoints
        Could not select REVIEW menu
        URI request: '/ '
        URI request: '/?command=status'
        WEB: Message command='status'
        URI request: '/ '

sfeakes, 2018-03-13:
It looks like the install is complete.  aqualinkd is connecting to your control panel and trying to get the appropriate information.  But seems it cannot quite get everything it needs.  Can you enable debug, put the lot to a file, let it run for a few minutes and post the output?
edit `/etc/aqualinkd.conf`, make sure the below are set
```
log_level=DEBUG
log_file=/var/log/aqualinkd.log
 ```
Then run and post output.

BigDogJonx, 2018-03-13:
So I notice its having trouble setting the time and getting spa temp and pool temp set points.  And freeze protection set points.  Looks like its mimicking essentially the button presses and waiting for messages on the RS8 keypad, so trying to understand where you are storing all this to understand where/how im going to have to tweak this to match my setup.  My only difference is my panel outside was upgraded about 5yrs ago to the iAqualink compatible panel (so I could use it off my phone and what not).  I got interested in your project mainly because I want to consolidate all this using either SmartThings or putting it into Homekit (which is why you saw Homebridge references in some of my outputs in the directory listings).

sfeakes, 2018-03-13:
@BigDogJonx I can see from the logs that it cannot get that information, also that your panel is a "combo" panel with both RS & iAqualink functionality.  the RS protocol is very archaic and only option is to mimic key presses on the control panel. It looks like your menu system is slightly different from the plane RS control panel that this was designed for.  I'm trying to find the documentation on your equipment so I can code around it.
But, everything else should work, (all equipment etc), it's only when it needs to program the control panel it will fail. So getting & setting freeze & heater set points & setting the time. The button functionality and showing the status of the equipment should all work.

BigDogJonx, 2018-03-13:
@sfeakes Yeap, the buttons work for sure.  I just re-did the conf file a bit since Im using all 8 of my buttons/auxs.  So re-labelled those and tested it out with my phone sitting right above the RS8 keypad and noticed if I pressed filter pump, it turned on, showed it on RS8 keypad, could press Floor Cleaner on my RS8 keypad and it showed up on the iPhone web interface.  So most of it seems to be working, Pool temp and air temps work.  Just the pool set temp, spa set temp, freeze protection temps dont.  I dont entirely care for the freeze protection set temp knowing it or changing it, I dont plan on ever changing that.  But the pool spa set temps would be nice to have as that would allow me to adjust the heater temps.

Since I have my keypad, Im assuming we can mimic the same concept if we write down the exact way we get to the pool set temp, and spa set temp, i can see you are using Review button to get data, so that could be done.  Let me know if you want me to help you write down the exact button presses and screen data on my RS8 keypad.

Ultimate goal like I said was to get this into Homekit eventually or SmartThings.  Got too many smart devices and each one has its own app, trying to get into of a more consolidated approach app wise.  So far Homekit has been decent when using homebridge for most of the items I want it to communicate with like my Wemo switches, Garage doors, Alarm, etc.

sfeakes, 2018-03-14:
@BigDogJonx I've made some changes, if you can chance, can you download the latest code, and see if it made a difference.  Please post logs.

Cheers

BigDogJonx, 2018-03-14:
@sfeakes Ill do that shortly, but not much of a github expert, do I just download the whole thing again and run the install again?  Or should I just manually move the files into place?  Only asking as generally been using this for homebridge and ive been using more the npm update concept (again granted not the same thing here).

johnluber, 2018-03-14:
If you’re able to run the pre-compiiled binary then you can just download that and copy it to the right location (for me /usr/local/bin).  You’ll probably need to stop the service, copy the new file to the location, change permissions and then restart the service.

You may also have to grab updated configuration files if that applies to the changes made.

John

BigDogJonx, 2018-03-14:
The temp set points and freeze still showing -999.  Time still is off.  Have not fully analyzed the log myself yet, was watching it live on the screen as it was flashing by.  
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1812497/aqualinkd.log)

sfeakes, 2018-03-14:
@BigDogJonx Thanks for trying.   I can see it's not selecting the menu from the logs.  Do you have a Aqualink RS control panel like this one?
https://www.activepoolsupply.com/Products/Large/21808_0_2017103084425.jpg

If so, can you try the following key presses.  MENU.  ENTER.  FORWARD.    and does the display shot REVIEW ?

Thanks

BigDogJonx, 2018-03-15:
@sfeakes Sorry got sidetracked yesterday and had to take the kids to their practices.  I do have that exact same RS Control Panel down to every detail/button.

And yes when I press in that order MENU, ENTER, FORWARD the display will be showing REVIEW on the screen.

sfeakes, 2018-03-15:
That is really strange.  That's the series of commands aqualinkd sends, but from your logs, it looks like that last FWD press cancels the command sequence. I've made a really quick change, if you wouldn't mind downloading the latest and trying again / post logs.  I don't think it will work, but it's worth a shot.

BigDogJonx, 2018-03-15:
Yea that didnt do much or change the results.  The set temps and freeze protections are still -999f.  I did initially run this with my own modified aqualinkd.conf (since I changed the label names around), but went ahead and used the one off github as well, and that didnt change either just to make sure I didnt do anything on the conf file, both logs attached.
[aqualinkd-modifiedconf.log](https://github.com/sfeakes/aqualinkd/files/1815783/aqualinkd-modifiedconf.log)
[aqualinkd-githubconf.log](https://github.com/sfeakes/aqualinkd/files/1815786/aqualinkd-githubconf.log)

Also attaching a video showing me pressing that sequence that gets us to the Review option.
[RS8.zip](https://github.com/sfeakes/aqualinkd/files/1815807/RS8.zip)

sfeakes, 2018-03-15:
@BigDogJonx I'm running out of ideas at the moment, I even put the code on a Pi3 to mimic your setup, just to see if I had the timings wrong with a faster processor.  But that worked as expected.
I'm wondering if you already have a device at the address aqualinkd is configured for that *might* be messing things up.  I've created a piece of code that will monitor all the ID's on the RS485 buss and record used / unused ID's.  Can you download the latest and run this please.
```
sudo ./release/serial_logger /dev/ttyUSB0
```
Let that run for a minute, then hit `ctrl C`, that will terminate and list out al the ID's it found.
if is states `0x0a` is in use, then that's out problem, you just need to configure aqualinkd to use a different ID.

if you need to make the serial_logger then simply run `make slog`

BigDogJonx, 2018-03-16:
@sfeakes I think we getting somewhere, I noticed you updated the aqualinkd executable in the past 3hrs and grabbed that too while I was getting ready to run the serial_logger and just ran the aqualinkd and to my surprise it captured now the pool set temp and spa set temps.  Freeze temp is still -999f though.  Log of that attached and screenshot that shows we making progress.  It still failing i guess going through menus trying to set time as well.

[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1817667/aqualinkd.log)
[image]

Here is also the list out of IDs it found.  Not seeing 0x0a

> ^C        Stopping!        Serial read 9 bytes
>         Received Packet for ID 0x00 of type Ack
>         ID is in use 0xa2
> 
>         ID's found
>         ID 0x60 is not used
>         ID 0x38 is not used
>         ID 0x32 is in use
>         ID 0xa2 is in use
>         ID 0x08 is in use
>         ID 0x6b is not used
>         ID 0x20 is not used
>         ID 0x21 is not used
>         ID 0x80 is not used
>         ID 0x22 is not used
>         ID 0x78 is in use
>         ID 0x23 is not used
>         ID 0x28 is not used
>         ID 0x81 is not used
>         ID 0x29 is not used
>         ID 0x2a is not used
>         ID 0x2b is not used
>         ID 0x30 is not used
>         ID 0x31 is not used
>         ID 0x33 is not used
>         ID 0x10 is in use

BigDogJonx, 2018-03-16:
This wierd, must have got lucky on that one attempt because now I cant get it to show the 34f and 100f that I actually do have set.  I see in the live debug its sitting on waiting for REVIEW to show up.   Interesting how the one time it got through and got the pool and spa set temps but didnt get the freeze temp.

sfeakes, 2018-03-16:
I had made a few changes to the timings, so maybe that helped.  I made a mistake in the serial loser and did not have allocate enough memory so it missed some ID's.  If you download the latest and try again, hopefully you'll get better results.  Also just as a quick test, can you edit aqualinkd.conf and change the address from 0x0a to 0x0b and see if that helps.
```
device_id=0x0b
```

BigDogJonx, 2018-03-16:
Here is latest log, still no change and didnt get the temps like I did the one time.  
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1819869/aqualinkd.log)

Just to help, I have basically 4 devices on the RS485 ports from the pool equipment side.  Jandy Aquapure EI SWG, Jandy LX400 Pool Heater, Jandy Variable Speed ePump, and the RS8 keypad.  We also have a 4button Spa remote which just controls 4 devices at the Spa side (Air blower, Spa Heat, Spa, and Spillway)

Here is the new list of IDs, seems like 0x09 is not in use, which is what I used in the above log.

>         Ran out of storage, some ID's were not captured, please increase SLOG_MAX and recompile
>         ID's found
>         ID 0x32 is in use
>         ID 0xa2 is in use
>         ID 0x81 is not used
>         ID 0x50 is not used
>         ID 0x38 is not used
>         ID 0x60 is not used
>         ID 0x08 is in use
>         ID 0x53 is not used
>         ID 0x58 is not used
>         ID 0x80 is not used
>         ID 0x68 is not used
>         ID 0x78 is in use
>         ID 0x69 is not used
>         ID 0x6a is not used
>         ID 0x6b is not used
>         ID 0x70 is not used
>         ID 0x71 is not used
>         ID 0x72 is not used
>         ID 0x73 is not used
>         ID 0x88 is not used
>         ID 0x10 is in use
>         ID 0x89 is not used
>         ID 0x09 is not used  <-- can use for Aqualinkd
>         ID 0x0a is not used  <-- can use for Aqualinkd
>         ID 0x0b is not used  <-- can use for Aqualinkd
>         ID 0x40 is not used
>         ID 0x41 is not used
>         ID 0x42 is not used
>         ID 0x43 is not used
>         ID 0x02 is in use
>         ID 0x18 is not used
>         ID 0x20 is not used
>         ID 0x21 is not used
>         ID 0x22 is not used
>         ID 0x23 is not used
>         ID 0x28 is not used
>         ID 0x29 is not used
>         ID 0x2a is not used
>         ID 0x2b is not used
>         ID 0x30 is not used
>         ID 0x10 is in use

sfeakes, 2018-03-16:
Every now and then it looks like there is a bad packet received from the control panel.  So I've now implemented some checks and will ignore them.  I can't check that my side so hard to say if it will fix it or not. But if you get chance can you try the latest on ID of 0x0a, and you may also want to check all your RS485 wiring, (lose wires, bad connections etc).

BigDogJonx, 2018-03-16:
That seemed to help for sure.  I noticed first time I ran it grabbed Freeze Protection but didnt grab the Pool or Spa set temp as shown below.  Then ran it again and this time it grabbed it all.  I noticed in logs it was trying to go through the time set and was having issues setting the time.  

Initial run screenshot:
[image]

Ran it again 2nd time, didnt change anything, just CTRL+C and ran the command again.
[image]

Here is the full log of both runs.
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1820274/aqualinkd.log)

I will also check all my connections, I can recheck at the outside RS8/Aqualink panel.  And will have to check the pump, aquapure, lx400 heater, and then recheck inside RS8 keypad (which is where I piggybacked the USB Serial adapter to).

BigDogJonx, 2018-03-16:
Just another update, just ran it again 3rd time and this time it set the time as well, noticed it grabbed all the pool/spa/freeze protection temps, and even sucessfully set the time.   Third times the charm i guess? hahha.

sfeakes, 2018-03-16:
That's great news.  What's interesting is the packet is smaller than the minimum size, (Well smaller than the size  from any documentation I can find), but the checksum is valid, making me think that this is actually a valid packet.  I'll have to do some more research on this and see what it actually is.

BigDogJonx, 2018-03-16:
Its progress for sure, but what I notice is its still hit or miss at times cause stopping and restarting it gets it at times to show the temps, sometimes it missed the pool/spa temps but gets freeze, sometimes it gets all of them, Ive been playing around with the controller.html to adjust it more to my needs and thats how ive been noticing it (since I keep stopping and restarting aqualinkd).

Im going to make sure all the connections are solid, but not having too much faith in that just because stopping and restarting it fixes it, my pool controller hasnt had any flaky issues with loose RS485 connections.

BigDogJonx, 2018-03-16:
Yeap tried 200 and it didnt even really run, just sat on GET_AUX_LABELS, then tried 100, same thing (when running this in CMD mode so I can see where its hanging up.  

Put it back to 0 and worked (just let it stay in service mode).

sfeakes, 2018-03-17:
100 should wait for 1 second on network requests, maybe try 10 if you get time, and that should wait 0.1 of a second.  I’ll play tomorrow, I was just looking for an easy/quick way to slow things down just to test the theory of polling the RS485 buss too quickly on your setup, before I start writing code.

BigDogJonx, 2018-03-18:
The 10 didnt seem to work either, just sort of sits there just cycling through following messages:

> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller
> Warning:Serial read bad checksum, ignoring
> Warning:Serial read bad checksum, ignoring
> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller
> Warning:Serial read bad checksum, ignoring
> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller
> Warning:Serial read bad checksum, ignoring
> Warning:Serial read bad checksum, ignoring
> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller
> Warning:Serial read bad checksum, ignoring
> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller
> Debug:  RS received packet of type Probe length 7
> Debug:  Send '0x00' to controller

So I tried 5, and seemed to have worked in that it obviously started and all three set temps (pool, spa, freeze protection) came on the first shot.  7 even works too.  But still see the `Warning:Serial read too small, (but good checksum),` messages, not a huge amount, I saw maybe 5-7 as it started up during the Programming.... part.

sfeakes, 2018-03-18:
Ok, that looks like it may be the problem, reading too quickly and not waiting long enough for all the data over the rs485 protocol.  I’ll work on a solution, still strange that I couldn’t duplicate it so still may be missing something.

BigDogJonx, 2018-05-01:
@sfeakes, not sure if you wanted to keep this open or close, but I think you can close it.  Im fairly happy with it working on the iAqualink board.  There are times when it doesnt grab all 3 temps, but restarting it usually gets it.  For the most part it has been grabbing the temps without too much of an issue.  Just occasionally missed one of the 3, but not a deal breaker since it seems to be resolved by restarting it.

Appreciate the work on this.

closed 2018-05-01 by BigDogJonx

bryan-barnard, 2018-08-15:
have Raspberry Pi 3 setup with USBtoRS485 adapter. installed AqualinkD and connected to existing remote controller connection in main control panel.

From the serial logger I see...
Notice: Logging serial information, please wait!
Warning:Serial read bad checksum, ignoring
Warning:Serial read bad checksum, ignoring
Warning:Serial read bad checksum, ignoring
Warning:Serial read bad checksum, ignoring
Notice: ID's found
Notice: ID 0x80 is not used 
Notice: ID 0x58 is not used 
Notice: ID 0x60 is not used 
Notice: ID 0x68 is not used 
Notice: ID 0x33 is not used 
Notice: ID 0x38 is not used 
Notice: ID 0x50 is not used 
Notice: ID 0x81 is not used 
Notice: ID 0x04 is in use