aqualinkd/AqualinkD #5: Enhancement request

johnluber, 2018-03-18:
Hi Shaun

Aqualinkd seems to be working well.  One thing I noticed though is that it holds on to temperature data beyond it being valid.  For example, I heated the spa yesterday and in HomeKit the temp is still reported as 100 even though it has cooled back down to air temp.  

So, question is, when aqualinkd returns a null value for temp can we have it pass some value to HomeKit?  Ideally a ‘-‘ would be return but I don’t know if HomeKit can display that in the temperature data type.  So, maybe ‘0’ or current air temp instead? 

Does this make sense?

Thanks.

John

sfeakes, 2018-03-18:
MQTT holds the last value, so that would be it.  Turn the hot tub on without the heater and that’ll sort it out..:).  I’ll work on a solution, but a - wouldn’t work, I’ll probably have to use the pool water temp if the hot tub is not on.

sfeakes, 2018-04-18:
@johnluber I just committed an update to do this.  Looked at a few options, best seemed to be simply post the pool water temp to spa water temp when in pool mode.  Can't use '-' as iOS doesn't;t support it, using 0 would mean changing the Thermostat range to work from 0 to 110, but Aqualink only supports 40 and above, so would cause other issues.

On a side note, I'm trying to control the SWG AquaPure & AquaRite in a better manor, would you mind running a piece of diagnostics software on your system for 10 or 15 mins and posting the output? 

If so, can you stop aqualinkd, turn off SWG, run diagnostics, press a few buttons, and post output?

Commands are 
```
sudo systemctl stop aqualinkd
(turn off SWG - turns on/off with pump usually)
sudo /.../aqualinkd/release/aquarite_logger /dev/ttyUSB0 > aqua.out
(turn on SWG )
wait for SWG to show "remote controlled" / or "no flow" is not flashing
use control panel to set SWG % to 25
wait a few seconds
use control panel to set SWG % to 50
wait a few seconds
kill diagnostics
CTRL-C
sudo systemctl start aqualinkd
```
Post the file  aqua.out

Cheers

johnluber, 2018-05-05:
Hi Shaun

Sorry for not getting back sooner.  Been doing some business travel and then had a network outage that got in the way of things here.

I did install the new binary and had a chance to do some usage testing last night.  Unfortunately, the aqualinkd crashed when trying to turn on the pool lights.  Previously, I hadn't been having a problem with this.

I've reverted to the previous version and am attaching the log from the crash.

Thanks!

john
[aqualinkd.log](https://github.com/sfeakes/aqualinkd/files/1976511/aqualinkd.log)

caetanog, 2018-06-11:
Hi, first time post to the site, wanted to start by saying thanks for putting this stuff out there, great work.  I'm actually new to the pool world but come from an Automation background, i have the PI zero config setup but unfortunantly have the PDA that came with the pool, let me know if you need any testers for the PDA version, i have a background in manufacturing systems so i'm quite use to some of these debug environments.

sfeakes, 2018-07-15:
I have this support in the latest version.  It now reports 0 or 32 depending on degC or degF on your phone.  I tried 0 degF and it was unstable, I will play some more.  You should also update the Thermostat configuration for homekit2mqtt (see example in extras directory).  Also supports SWG now, so you can set % generating rather than just viewing it. (again see homekit2mqtt.json example in extras directory)

caetanog, 2018-08-01:
I'm still available to test the PDA version if you need anyone.

sfeakes, 2018-08-02:
I have played with this slowly, but it’s a kinda pain.  Still need to spend a lot more work on it.   Someone recently told me that you can change a chip on the control panel circuit board, and convert the PDA only to RS and PDA functionality.  Said they were around $100 off eBay.  He used that to get option to get AqualinkD working in his setup.

caetanog, 2018-08-02:
Thanks for the quick response, you wouldn't happen to have that persons e-mail would you, there seems to be a wide variety components, i'm wonder how he figured out what to buy. Thanks again for sharing all this stuff, it's awesome. George. ---- On Wed, 01 Aug 2018 22:15:48 -0400 sfeakes <notifications@github.com> wrote ---- I have played with this slowly, but it’s a kinda pain. Still need to spend a lot more work on it. Someone recently told me that you can change a chip on the control panel circuit board, and convert the PDA only to RS and PDA functionality. Said they were around $100 off eBay. He used that to get option to get AqualinkD working in his setup.

sfeakes, 2018-08-04:
Closing this as original issue is fixed.  Opened a new issue for PDA support.

closed 2018-08-04 by sfeakes