aqualinkd/AqualinkD #7: Smartthings integration

shavkaca, 2018-05-29:
Hi,

First of all thanks for a great piece of software, I am running it on a raspberry pi and it works flawlessly.
Probably a dumb question, but I was not able to integrate Aqualinkd with Samsung Smartthings. Could you please post steps.
Thanks!

sfeakes, 2018-06-06:
Glad it's working.  
Sorry, I don't have a smart things hub, so I can't document the steps.  But you have 2 options, since smart things is cloud based home control.  
One would be to allow access directly to the aqualinkd API from the outside world. Since there is no security on aqualinkd, this is not a good idea, unless you wrap the aqualinkd API with a layer of security using an API management tool, or apache etc.
The other option would be to use MQTT, and connect MQTT to smart things, and to aqualinkd. if you google "MQTT bridge Smartthings" there are lots of options to do this.  Then simply configure Smatthings to the MQTT topics used in aqualinkd (listed on the main page).

shavkaca, 2018-06-06:
Thanks for your reply. Will try MQTT when I get a chance and report back.

sfeakes, 2018-11-30:
Closing this. Please use the aqualinkd forum to continue discussion
http://aqualinkd.freeforums.net

closed 2018-11-30 by sfeakes