aqualinkd/AqualinkD #8: Variable speed pump

napertivo, 2018-07-05:
First I want to say THANK YOU!  This is great; I have been looking for something like this for a while.

Have you given any thought on how to handle variable speed pumps?  I have almost everything else working; I did have to move things around a little.  My variable speed pump changes its speed based on what is chosen but there are some extra speed settings that have to be manually chosen.  

I am still looking at the light; I updated the colors to reflect the light I have but selecting it turn my fountain off and on :-).  

[image]
[image]
[image]
[image]

napertivo, 2018-07-06:
Did I see in the other thread that you are sniffing on the rs485 bus?  If so, would I be able to grab what my inside panel is putting on the bus (my Pi is wired to the inside panel) and identify what it sends when those buttons are pressed?

napertivo, 2018-07-06:
Oh, regarding the variable speed pump if it helps, I had to assign the pump to an available aux port (it is not physically cabled to the port), in my case it was aux4.

sfeakes, 2018-07-15:
I have uploaded a more complete serial logger, that may help in deciphering how a variable speed pump works.  It seems they work similar to a Salt Water Generator where they are a complete RS485 device and not a simple switched device.  So I'll need a ton of log information before working out how to support.

napertivo, 2018-07-16:
Thanks I will load it in a few days; I am unable to work on it until next weekend.

napertivo, 2018-08-21:
Finally I was able to run some tests.  Let me know if this helps.
[aqualinkd.log.gz](https://github.com/sfeakes/AqualinkD/files/2304828/aqualinkd.log.gz)

sfeakes, 2018-08-27:
Reading Jandy documentation it looks like there are several different ways to handle variable speed pumps. Looking at your logs, and re-reading this thread. Do you have "High speed" set to "Filter Pump" and "Low speed" set to "AUX 4"?

If so, you just need to configure the UI to these settings.
Since you showed the old UI above, change these line `button_04_label=Waterfall` to `button_04_label=Pump Low Speed`, but from your screen shot in the first post it looks like you have `button_04_label=Air Blower`

So I'm a little confused.  Can you let me know how you control the speed of the pump from a jandy keypad.

napertivo, 2018-08-28:
The pump is connected to the panel via "Filter Pump" and provides the pump on/off.  The controller for the pump, which is on top of the pump is cabled to the rs-485 port as well.  When cabled a new menu item was displayed (attached) showing the pump settings.  
[image]
[image]
[image]

Here is the video I followed programming the pump.  https://www.instructables.com/id/Intelliflo-VS-Pump-to-Aqualink-Jandy-control-Panel/

AUX 4 is used for a JVA controlling one of the returns.  The extra settings were in error while I was trying to configure for the pump in aqualinkd.

napertivo, 2018-08-29:
So AUX 4  is used to map the Solar port to the JVA as described in post #3 of this link so I can control the waterfall "https://www.troublefreepool.com/threads/161435-IAqualink-Programming-JVA-for-Waterfall".  I thought Aux4 was the only one available but I will check again in the light.  I'm not clear on how the pump speeds are related.
[image]

napertivo, 2018-08-29:
OK it was bothering me; flashlights do work in the dark :-).  The fountain JVA is cabled to the Solar port and the Solar port is now assigned to Aux7.  I will update Aqualinkd to reflect the change and collect more data to see if it is more meaningful.

sfeakes, 2018-08-29:
After reading and looking at all the information I can find, I think this will be a large change.  The All button control panel (that aqualinkd mimics) does not support variable speed pumps, (well only in a 2 speed mode when wired in a specific way).  So the only option to to change AqualinkD to mimic some other control panel. The way the one touch works is simply too slow and too painful to mimic, so I'd need to use the iAqualink or PC Interface, neither of which my current control panel supports.  There is too much to debug to be able to do this remotely, so I'm afraid I won't be able to do this correctly until I can get hold of a newer Aqualink RS panel.
I've started working on Aqualink PDA support, and that is very similar to One touch, but the functionality will be stripped down as it simply doesn't support much.  I may be able to cobble something together for variable speed pump but it won't be clean.  But I will need some more logs from the serial logging tool.  I'll give some more detail in a few days.

napertivo, 2018-08-29:
Great, thank you for all your help.  Anything I can do to help please let me know.

tcm0116, 2018-12-26:
How difficult would it be to add support for at least providing the current RPMs and power utilized by the VSP? Below is a portion of a log file that I created while monitoring my iAqualink device (id 0x33) when I entered the Status screen.

It looks like it sends a 0x06 command in the second ACK, which causes the controller to respond with several text strings containing the desired information. I didn't dig through all of the responses, but I think it's pretty much the exact text that is displayed on the Status screen.

```
  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x06|0x19|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x23|0x2a|0x92|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x23|0x00|0x36|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x2d|0x45|0x71|0x75|0x69|0x70|0x6d|0x65|0x6e|0x74|0x20|0x53|0x74|0x61|0x74|0x75|0x73|0x00|0xce|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x2d|0x00|0x40|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x00|0x46|0x69|0x6c|0x74|0x65|0x72|0x20|0x50|0x75|0x6d|0x70|0x00|0x92|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x01|0x53|0x70|0x61|0x00|0x8f|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x02|0x50|0x6f|0x6f|0x6c|0x20|0x48|0x65|0x61|0x74|0x20|0x45|0x4e|0x41|0x00|0x9c|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x03|0x4a|0x61|0x6e|0x64|0x79|0x20|0x65|0x50|0x55|0x4d|0x50|0x20|0x20|0x20|0x31|0x00|0xbb|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x04|0x20|0x20|0x20|0x20|0x52|0x50|0x4d|0x3a|0x20|0x33|0x31|0x35|0x30|0x00|0x00|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x05|0x20|0x20|0x57|0x61|0x74|0x74|0x73|0x3a|0x20|0x31|0x35|0x34|0x30|0x00|0xe6|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x06|0x41|0x71|0x75|0x61|0x50|0x75|0x72|0x65|0x20|0x32|0x35|0x25|0x00|0x40|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x07|0x53|0x61|0x6c|0x74|0x20|0x33|0x35|0x30|0x30|0x20|0x50|0x50|0x4d|0x00|0xfa|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x25|0x08|0x20|0x00|0x92|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x25|0x00|0x38|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x24|0x01|0x00|0x00|0x00|0x00|0x00|0x6a|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x24|0x00|0x37|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x28|0x0c|0x18|0x12|0x00|0x1c|0xbf|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x28|0x00|0x3b|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
```

sfeakes, 2019-01-24:
I'll start working on this soon, posted information in this thread, so closing this out.
https://github.com/sfeakes/AqualinkD/issues/37

closed 2019-01-24 by sfeakes