aqualinkd/AqualinkD #9: Does aqualinkd/solar_heater/ exist for MQTT?

bsheedy, 2018-07-28:
Does  aqualinkd/solar_heater/ exist for MQTT?  As my Aqualink RS6 uses the Solar Pump Extra AUX to run the Waterfall relay.

This is the best implementation of Aqualink web/API access ever... Period.   I installed it on a Pi Zero attached to a USB-RS485 converter, all running through a +5V regulator that is hooked up to the main voltage of the existing Aqualink RS6 panel.  It all fits behind the panel and is running on Wi-Fi.

The MQTT and HTTP API is working well with HASS.IO and I can control everything!  I am even thinking of clearing the programs in the panel and running my own automations through HASS.IO.

Thanks for all your effort!

sfeakes, 2018-07-28:
Glad it’s working well for you.  The solar heater should work, but since I don’t have one I can’t test it.  If you let me know what it’s doing / not doing when you try solar heater I can see if I can fix it.

I actually use my own automation hub to control all the daily automation, and even freeze control since I have far more control that way.

bsheedy, 2018-07-28:
Thanks!  I tried it as aqualinkd/solar_heater and aqualinkd/solar_pump just in case as I kept transposing them in my scripts and i figured it didn't exist, so I used a REST platform call from HASS.IO by using the "http://aqualinkd.ip.address:port?command=Solar_Pump&value=on" call.

sfeakes, 2018-07-28:
I've fixed it now, just uploaded new version. MQTT should be working for Solar Heater.

MQTT
aqualinkd/Solar_Heater/set.    (1 or 0). to set state
aqualinkd/Solar_Heater            State is reported as 1 or 0

WEB API
http://aqualinkd.ip.address:port?command=Solar_Heater&value=on

Also make sure you have a nice label assigned to it in the config.  So for you in /etc/aqualinkd.conf
button_12_label=Waterfall

Hope this works
FYI, If you don't want to install a new version, the bug was that if the DZ_IDX wasn't set it didn't post to either MQTT or Domoticz.  So on your version if you set the below in the config, it should work.
You probably have :-
button_12_dzidx=NONE

change it to :-
button_12_dzidx=999       <- any number other that 0 will work.

closed 2018-08-01 by sfeakes