aqualinkd/AqualinkD #10: Aux key codes for multi-panel Aqualink systems (RS2-14)?

rhisey, 2018-07-31:
Do you happen to know what key codes should be used in aq_serial.h to control additional Aux devices on additional control panels in multi-panel/power center Aqualink systems, like the RS2-14?

I’ve just begun working with aqualinkd on a RS2-14 Dual system, a AquaLinkRS B0029235 REV T.1, and everything on the first physical control panel/power center is responding as expected from what I can see so far. In my case this system has additional aux devices connected to Aux B1, B2 & B3 on the second control panel that I would like to control through aqualinkd.

Oddly enough in this case I also have a 3rd control panel (I recently bought this house from a pool builder) with 3 more aux devices on C3, C4 & C6, which again, I’d like to control through aqualinkd.

Those ‘B’ and ‘C’ designations are as I see them labelled under “Circuit” within the aux label customization screen in the iAqualink web interface.

Just wondering if you know how to handle this before I dive in deeper. Thanks for your hard work in this project!

sfeakes, 2018-07-31:
I don't actually know them. But if I remember the documentation this might be a little more complicated than finding out the button ID's.  Getting the commands is easy, explained that below.  But getting the status might be a little harder. This can only be done from deciphering a particular byte, and I seem to remember that order was one set up to RS6,8,10 and something very different from 12 and above.  For the sake of execution speed, that part of the code is kind-a hairy with a series of masked pointers back to the original received byte.

Anyway, first thing is to get the commands.  The serial logging tool will do that for you.  ( I just uploaded a new version of that, that will make it even easier) So get the latest version then.

Get the ID of your main keypad panel, (assuming 0x08) and your USB port is /dev/ttyUSB0 use the below command.

serial_logger /dev/ttyUSB0 -d -p 0 -I 0x08

That will start logging very packet sent from the controller to the keypad, and from the keypad back to the controller.

Below is an example of a nothing really done.  Panel sent Status, Keypad sent Ack.
```
 To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x00|0x00|0x00|0x00|0x00|0x1c|0x10|0x03|
From 0x08 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
```

Press the key you want to know, and look at the 6th byte printed in the ack. `0x09` in the below means example the menu button was pressed.
```
  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x00|0x00|0x00|0x00|0x00|0x1c|0x10|0x03|
From 0x08 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x09|0x1c|0x10|0x03|
```

The only stuff you are interested in is #5 and #6 in the Ack, #5 is simply some form of status the keypad sends the panel, like Display Busy, but #6 is the actual button press.  All the ones I know so far are listed in aq_serial.h

If you need help getting the ID of your keypads just run serial_logger no options and after it's logged everything on the interface, it'll list out all the ID's that are in use.  If you list those we can take a guess at which one it might be.  But usually 0x08 is the first / main real keypad. (by real I mean with individual buttons for each device, not the PDA or one touch panels)

rhisey, 2018-08-01:
Thanks for your quick response. To clarify, I don’t actually have any RS8 control panels connected to the system. I was referring to the control panel on the front of each power center. The only official method of external control I have is through iAqualink.

I think your point about differences between the models might be valid. When I run your serial logger, without options, I get a lot of warnings before it lists the ID’s found (when running it without any options). Is it normal to see these in your experience?

	 Warning:Serial read too small
	 Warning:Serial read bad checksum, ignoring

I’ve tried different cables just in case, but see these the same warnings in all cases. It will always list some “in use” and “not used” afterwards. but these don’t seem to be consistent each time. It will list some IDs until the next time its run, when it will usually shows slightly different results.

I also noticed that in your web interface (both the classic and the HomeKit style) that the Pool heater and Spa heater temperatures aren’t usually displayed, even though their status seems to update periodically. Occasionally they will flash very quick on screen and go back to “—“

All of these issues make me think that maybe there is more to the packet format on this multi-panel system than your code for a 1-panel currently addresses, but I’m not 100% sure on that. So I’m not sure how to proceed at this point.

Do you know if its possible to use your serial logger to monitor communications between the external iAqualink controller and the main Aqualink panel, instead of an RS8 type controller? Or monitor the status of the buttons on the power center panel itself? Those additional power center panels are connected the main unit over the RS485 connections, so maybe it’s possible to monitor their communications somehow.

sfeakes, 2018-08-01:
You will probably see that message when it first starts, as the first read may not be a complete packet.  But you should not see that constantly.  The checksum will fail if Aqualink didn’t understand what was read. When I have seen constant too small errors in the past, it’s usually due to bad usb cable / bad RS485 adapter or wiring.   You might try flipping the wires on the adapter, see if that helps.  It could also be the way my code is reading the serial port, maybe there is something slightly different about your setup than others.  One other person had a similar issue, and we had to slow down the serial reading in the AqualinkD code for it to work. I had put that down to bad hardware, but maybe something else.

Other thing to try if you have a windows machine you can hook up to the USB adapter, you can download Jandy’s simulator software, if that works reliably with your setup, then we know it’s my code.  

What hardware are you running aqualinkd on?

I’ll give this some more thought, and see if I can write something to track this down.

If you want to slow down the serial reads in AqualinkD, line 867 ‘mg_mgr_poll(&mgr, 0);’ change 0 to something more.  500 would be 1/2 a second.  (Probably way too much). You’ll probably best putting AqualinkD in serial debug mode (config) and running it at 0 to get a base line on the errors, then 100 / recompile and run again.  See if it clears up the errors.

sfeakes, 2018-08-01:
Just a quick thought.  You were not running the serial logger at the same time you were running AqualinkD were you?  That would cause the errors as well.
You can only run one at a time and no other process should be using the same port

rhisey, 2018-08-01:
That may have been it. I just reran the serial logger without any options. The errors from the logger are now gone(other than when it's first started), but I've had to include the -p option to consistently discover the ID's. I used "-p 900". 

Without that option the results are different each time, and it doesn't always show the "<-- can use for Aqualinkd" label.

Here are the results when I use that option:

```
Warning:Serial read bad checksum, ignoring
Notice: ID's found
Notice: ID 0x20 is in use 
Notice: ID 0x28 is in use 
Notice: ID 0x29 is in use 
Notice: ID 0x33 is in use 
Notice: ID 0xa3 is in use 
Notice: ID 0x60 is not used 
Notice: ID 0x51 is not used 
Notice: ID 0x80 is not used 
Notice: ID 0x52 is not used 
Notice: ID 0x18 is not used 
Notice: ID 0x81 is not used 
Notice: ID 0x21 is not used 
Notice: ID 0x22 is not used 
Notice: ID 0x23 is not used 
Notice: ID 0x2a is not used 
Notice: ID 0x2b is not used 
Notice: ID 0x30 is not used 
Notice: ID 0x31 is not used 
Notice: ID 0x32 is not used 
Notice: ID 0x38 is not used 
Notice: ID 0x39 is not used 
Notice: ID 0x3a is not used 
Notice: ID 0x3b is not used 
Notice: ID 0x48 is not used 
Notice: ID 0x50 is not used 
Notice: ID 0x53 is not used 
Notice: ID 0x58 is not used 
Notice: ID 0x68 is not used 
Notice: ID 0x69 is not used 
Notice: ID 0x6a is not used 
Notice: ID 0x6b is not used 
Notice: ID 0x70 is not used 
Notice: ID 0x71 is not used 
Notice: ID 0x72 is not used 
Notice: ID 0x73 is not used 
Notice: ID 0x88 is not used 
Notice: ID 0x89 is not used 
Notice: ID 0x08 is not used  <-- can use for Aqualinkd
Notice: ID 0x09 is not used  <-- can use for Aqualinkd
Notice: ID 0x0a is not used  <-- can use for Aqualinkd
Notice: ID 0x0b is not used  <-- can use for Aqualinkd
Notice: ID 0x40 is not used 
Notice: ID 0x41 is not used 
Notice: ID 0x42 is not used 
Notice: ID 0x43 is not used 
Notice: ID 0x2f is in use 
```

rhisey, 2018-08-01:
Also, to answer your question, I am using a Raspberry Pi Zero W. The RS485 adapter is the same one linked to in your release notes.

sfeakes, 2018-08-01:
Perfect.  Since you have so many things on the RS485 buss I'm sure it's extremely noisy, so that's probably why it will take the serial logger a larger number of packets than normal to decipher what's there and what isn't. 
(You have too many unknown (to me) ID's to figure out which ID we need to log) 
So we need to work out the ID of the iAqualink receiver device that is plugged into the RS485 buss so we can just log those packets rather than everything. It will save a ton of time.  Easiest way would be to simply unplug it so it doesn't respond.  So if you run the serial logger with and without the iAqualink connected, that should tell you the ID.  Once you have the ID, then you cat go back to my post above about how to track down the button codes.

Hope this helps

rhisey, 2018-08-01:
That's a good suggestion. I did try this. Disconnected the iAqualink from the RS485 connection on the power center panel.

When I first ran the serial logger shortly after that I didn't notice any changes. After trying again a couple of minutes later I noticed that "0xa3" and "0x33", both listed as "In Use" in previous attempts, are now gone.  I need to do some further testing to confirm this when I'm back at the panel.

I'm assuming that I'm looking for one (or maybe two in this case) of the "In use" devices to drop off when it's disconnected, right? Or could it potentially be any of these ID's?

Would you expect that device ID(s) to still appear in the logger for a short while after disconnecting?

One thing to note is that when I run the serial logger with the option to filter for a specific ID, it doesn't appear to be filtering correctly and is showing communications from other devices. I updated to the latest version but maybe I configured something incorrectly.

sfeakes, 2018-08-02:
Yes, that's exactly what you're looking for, an ID that disappears from the list when you unplug the iAqualink. 
I would only expect one device ID to be listed for iAqualink.  But the protocol is kinda Archaic and the iAqualink needs to de a lot, so maybe there are two ID's.  Try with more packets, set it to 5000, that should be enough to find everything. (Remember if a device is off, it don't be seen). So something like a Salt Water Generator will only be seen when it is actually physically on, ie when the pump is running.

I made a few changes to the logger, see if that helps.  Ultimately it logs everything, it's only the print command that gets filtered by the -I 0x33.  You can always pipe to a file, then grep the file to filter the output if the serial logger still doesn't;t filter for you.

rhisey, 2018-08-02:
Thanks, Shaun. I think I’ve made some good progress. 

I’ve determined that ‘0xa3’ is the ID of the external iAqualink hardware. When it’s disconnected, it disappears from the logger. 

Also, ‘0x33’ still appears in the logger, however it goes from being listed as “In Use” when the iAqualink hardware is connected to “not used” when it’s disconnected. So I’m guessing that’s related somehow.

When I run the logger to log communications to and from the iAqualink (0xa3), and then enable or disable an Aux within the iAqualink app, I do see the messages coming through to the log, but it’s being listed as “Unknown” instead of “Ack”. In fact, it looks like all messages being sent TO 0xa3 are being listed as “Unknown” type. All messages received FROM 0xa3 are being listed as “Ack”, except when it’s showing a button press(from iAqualink app). 

Here is what I see if I enable Aux 3 in iAqualink (on the first panel):

```
To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x73|0x28|0x10|0x03|
From 0xa3 of type Unknown  | HEX: 0x10|0x02|0x00|0x24|0x73|0x01|0x23|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0xcd|0x10|0x03|
```

Here is Aux B1(represents the first Aux/button on the second panel):

```
To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x73|0x28|0x10|0x03|
From 0xa3 of type Unknown  | HEX: 0x10|0x02|0x00|0x24|0x73|0x01|0x27|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0xd1|0x10|0x03|
```

Here is Aux C4 (represents the fourth Aux/button on the third panel):

```
To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x73|0x28|0x10|0x03|
From 0xa3 of type Unknown  | HEX: 0x10|0x02|0x00|0x24|0x73|0x01|0x32|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0x00|0xdc|0x10|0x03|
```

I know those are kind of random, but I wanted an example from each panel. I am going to make a log of each of these Aux buttons from the iAqualink app. The good news is it appears that even if an Aux is not being used (nothing connected) the iAqualink still sends a message when toggled on/off, so if that’s the case it should be possible to capture all of these.

sfeakes, 2018-08-03:
I'd need to see more to detect what's really going on, as that protocol is different than any of the stuff I've debugged so far on all Jandy devices.  They all wait for something from the master control panel then reply with an ack of a certain type to do something.  That looks to be the same, but the message commands are different (why you get the unknown). But the good thing is the whole message is formatted the same, so at least we are that far.  

You may also want to get some logs from 0x033 just to see what's going on there, it seems that the iAqualink may use 2 interfaces.  I'd assume, one to look like a keypad and one to look like an extended device (like a variable speed pump), and you're picking up the later.

Just for your reference :  
DLE | STX | TO ID | COMMAND | xxxxxxxxxx | CHECKSUM | DLE | ETX

So you can forget about the first 2 bytes, next is the ID of the device the message is intended for, rest can change is size and type, then end with another 3 bytes you can forget about.
It's usually the COMMAND and

rhisey, 2018-08-03:
I’ve attached more data. Some notes: 

1) I wasn’t able to test Aux 1 because of an issue w/ cleaner here, but I tested the rest. So this is AUX 2-6, B1-B8, C1-C8, and “Extra Aux”

2) I have color LED lights on three of these Aux outputs, but these captures just show the basic On/Off toggle message. The iAqualink UI allows you to select a brand of lights. When a specific brand is selected, iAqualink outputs a more customized message. For simplicity I haven't included that in these results. Just the basic On/Off toggle message.

3) I noticed byte 7 in the "From 0xa3" changing (counting up) as each command is executed

I will do some further investigation into device ID 0x33 to see how it might be related.

[iAqualink captures.txt](https://github.com/sfeakes/AqualinkD/files/2255643/iAqualink.captures.txt)

sfeakes, 2018-11-30:
Closing this out, please use the AqualinkD forum to continue this.
http://aqualinkd.freeforums.net

closed 2018-11-30 by sfeakes