aqualinkd/AqualinkD #12: MQTT - Can aqualinkd/Freeze_Protect/setpoint/set be sent data in Fahrenheit?

bsheedy, 2018-08-02:
Is there a way to allow aqualinkd/Freeze_Protect/setpoint/set in MQTT to receive data in Fahrenheit?

I am using Home Assistant (HASS.IO) and the Climate (or thermostat) function only displays and sets the degrees in Fahrenheit.  When I set 32°F in my Home Assistant, the program sends 32 which obviously exceeds the maximum setting in °C for Freeze protection and sets it to 42.  I have sent it 1 - 6 manuall and it works fine.

If I could have the Freeze_Protect, Spa_heater and Pool_Heater "/setpoint/set" receive the values in °F it would simplify my interface immensely. 

Any thoughts would be greatly appreciated.

sfeakes, 2018-08-02:
When you set the ‘ convert_mqtt_temp_to_c = yes ‘ it’s supposed to do (or not do) the conversion on both the post and set, but looks like I messed up the if statement on the set.  I’ll fix in in the next update.  So yes, it will do this for you when I fix the bug.

sfeakes, 2018-08-02:
uploaded a fix.

bsheedy, 2018-08-02:
Thank You!!

How can we honor you for all your efforts?

sfeakes, 2018-08-03:
Better test it work for you first...:)

I've put a donate button on the main page, buy me a beer if you feel like it, or probably my wife a glass or wine for ignoring her while I work on this..:).  But please don't feel like you have to, just sharing the work I've figured out for my own setup.

bsheedy, 2018-08-03:
Done!

The Temps can now be set in °F and everything displays properly.  My Home Assistant now works with every function found in your API's.

I am even more impressed with the simulation.html as I can now have access to the Menu/Enter buttons to  program the panel without getting off the couch!!

Thanks again!

sfeakes, 2018-08-03:
Great, glad it all worked out for you.

Is there any chance you can write some notes on what you did with Home Assistant, so I can add it to the Wiki?

sfeakes, 2018-08-04:
Perfect thanks very much, putting it in the git hub page now.

closed 2018-08-04 by sfeakes