aqualinkd/AqualinkD #13: PDA Support.

sfeakes, 2018-08-04:
Info on PDA support will be posted here.  At present AqualinkD can only mimic an RS All button keypad, So if you have a PDA only control panel, ie one without RS, AqualinkD will not work.

bryan-barnard, 2018-08-15:
If you need any dev/beta support, let me know. I have a Jandy PDA unit.

stulincoln, 2018-09-02:
> Just a quick FYI. I've been working on the PDA support for the last 2
> days, it's been a real pain, probably have a day or so to go before I can
> release the first beta version. So keep your eyes on this thread for an
> update, I *hope* to get something out within a few days.
>
> Quick question for all who are waiting, on my panel everything is listed
> in uppercase, is that the same for you panel. IE. The main menu is
> something like below
>
> POOL MODE     ON
> POOL HEATER  OFF
> SPA MODE     OFF
> SPA HEATER   OFF
>
> and the Equipment menu is similar to below
>
> FILTER PUMP   ON
> SPA          OFF
> POOL HEAT    OFF
> SPA HEAT     OFF
> CLEANER       ON
> AUX2         OFF
> AUX3         OFF
> AUX4         OFF
>
> Can you please confirm your PDA remote show only uppercase for all
> equipment?
> Also that "POOL MODE" on the main screen is the same as "FILTER PUMP" on
> the equipment page?

Correct, mine is all in uppercase. I believe the pool mode is same as
filter pump, correct.

This is awesome, thanks.

Stu

sfeakes, 2018-09-02:
I've posted Beta version.  Please make sure to read the WiKi, and set config accordingly.

ballle98, 2019-01-15:
Thanks for doing this.  I've been playing around with it for a couple weeks now.  One thing that wasn't completely clear on the wiki was the button to PDA label mapping and the config.js.  My first mistake was deleting buttons that were not used and trying to use my own order.  Later after looking at the code I realized that button 1 has to be filter pump, 2 has to be SPA, 10 has to be pool heat and 11 has to be spa heat.  Here is the config that works for me.

`# Labels for standard butons (shown in web UI), and domoticz idx's
button_01_label=Filter Pump
#button_01_dzidx=37
button_01_PDA_label=FILTER PUMP

button_02_label=Spa Mode
#button_02_dzidx=38
button_02_PDA_label=SPA

button_03_label=Cleaner
#button_03_dzidx=39
#button_03_PDA_label=AUX1
button_03_PDA_label=CLEANER

button_04_label=Waterfall
#button_04_dzidx=40 
button_04_PDA_label=AUX2

button_05_label=Spa Blower
#button_05_dzidx=41 
button_05_PDA_label=AUX3

button_06_label=Pool Light
#button_06_dzidx=42 
button_06_PDA_label=POOL LIGHT

button_07_label=Spa Light
#button_07_dzidx=43 
button_07_PDA_label=SPA LIGHT

button_08_label=NONE
#button_08_dzidx=NONE 
button_08_PDA_label=AUX6

button_09_label=NONE
#button_09_dzidx=NONE 
button_09_PDA_label=AUX7

button_10_label=Pool Heater
#button_10_dzidx=44
button_09_PDA_label=POOL HEAT

button_11_label=Spa Heater
#button_11_dzidx=56
button_09_PDA_label=SPA HEAT

button_12_label=Solar Heater
#button_12_dzidx=NONE 
button_09_PDA_label=EXTRA AUX`

ballle98, 2019-03-19:
fixed several PDA issues and added heater and freeze protect support here https://github.com/ballle98/AqualinkD

closed 2019-05-31 by sfeakes

bryan-barnard, 2019-06-03:
@ballle98 @sfeakes Does this update support using the the RPi and the PDA remote? I think the last time I tried this I could only have one of the 2 connected at the same time. It looks like my controller has connections for 2 remotes, but I couldn't not get both to work at the same time. I could only get the RPi to work if I disconnected the PDA 4-pin and installed the RPi in the same position.

sfeakes, 2019-06-03:
Yes it should now work with physical PDA as long as you use a different ID

ballle98, 2019-06-03:
@bryan-barnard I have a setup with a PDA wireless remote and a RPi both connected to the 1 4 pin connector.  The code I have here https://github.com/ballle98/AqualinkD definately supports this, the patches I had were reworked and changed and I haven't had time to validate them.  Please try it and if it does not meet your expectation try the code in my repo.  I don't commit the binnary files so you have to:

`git clone https://github.com/ballle98/AqualinkD; make clean; make install`