aqualinkd/AqualinkD #14: Post Install Instability

mattsntx, 2018-08-13:
First, thank you for creating this program. I have been wanting to do this for some time and it is exactly what I was looking for. I installed last night using a Pi Zero and USB adaptor from Amazon that was linked to in the WiKi (data + to A and data - to B). At this point I am just running the web interface and am having some stability issues. It starts up and runs, but has crashed on me a couple of times. It also only sometimes adds the SWG info and the Freeze protection info. I have reviewed the log (attached) and the only thing that jumps out at me is the "Warning: Serial read too small", but it looks like it just skips over those. Interestingly, the serial logger app won't run. It just gives the first line and freezes there. If I switch the two lines on the USB device, I get errors in the serial logger as expected. If I leave the data + on A and attach the data - to the third post on the device (which appears to be ground) the serial logger will run as expected with the notice of all available ID's. However, the web interface does not work when wired this way so I switched it back to data + to A and data - to B. Can you take a look at my logs and see if you see anything that I can correct. Again, the web interface does work while running. 

Thanks so much for putting this together. Your work is very much appreciated! 

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2284124/aqualinkd.log)
[image]

sfeakes, 2018-08-13:
At first quick glance, it looks like you have a control panel on the same ID as you have AqualinkD set to.
The serial logging tool will log a number of packets quietly, then spit out the result. So maybe that’s why you think it isn’t doing anything.  Leave it running for a minute or two and see it it spits out a result.  You can also put it in debug mode to print out evey packet rather than being quiet (look at the wiki for instructions).

I’ll give it some more thought, but please try that first.

sfeakes, 2018-08-15:
I’m away on business at the moment, I’ll need to put some time aside and go over this in more detail.  I’ll try to find time at the weekend.   But ultimately we need to stop the serial read issues for this to be stable. As long as you’re not running aqualinkd and the serial logging tool at the same time I’m not sure what would be causing this and I’ll need to write some code to see what’s going on.

mattsntx, 2018-08-16:
Ok. I currently have it hooked up to a controller in the house. I may take it outside and hook it directly to the main unit and see if I get less errors there just to eliminate the possibility of cable issues somewhere along the line. Let me know if you find anything when you get a chance.

sfeakes, 2018-08-16:
Out of interest, What distribution & version of Linux are you using?

sfeakes, 2018-08-16:
I've mad a few small changes, please download the latest and see what the logs look like. I simply increased the timeout to wait longer to receive a complete RS485 packet.  Don't think it will fix your problem (although it could), but it'll help rule out a few things.

mattsntx, 2018-08-24:
Sorry I was out of town and could not work on this for a few days. I updated software and seem to be having the same errors. I am attaching the new log file and serial logger output to see if that gives you any insight. 
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2316875/aqualinkd.log)
[Serial_Output.txt](https://github.com/sfeakes/AqualinkD/files/2316876/Serial_Output.txt)

sfeakes, 2018-08-24:
Looks like the serial logger is the new one, but aqualinkd is the old exe.  Just checking you ran the install script?

mattsntx, 2018-08-24:
Sorry - I downloaded it, but did not run the script. I just re-downloaded it again and ran the script this time. Here are the new logs after running it again. Thanks for taking the time to look at this. 
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2316937/aqualinkd.log)
[Serial_logger.txt](https://github.com/sfeakes/AqualinkD/files/2316938/Serial_logger.txt)

sfeakes, 2018-08-25:
Thanks for the logs. Think I've found the problem, and fixed it in the latest update.  Please get latest and try again.

mattsntx, 2018-08-26:
I think that did it! I am no longer seeing errors. I am attaching the log after a re-ran it with the latest update just in case you wanted to look at it. But it looks good to me. Thanks again. 
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2321155/aqualinkd.log)

sfeakes, 2018-08-26:
Log looks good, it didn’t miss / duplicate or re-read any message.  I’ll close this out, thanks for testing.

closed 2018-08-26 by sfeakes