aqualinkd/AqualinkD #15: Changed device_id?

napertivo, 2018-08-16:
I have been unable to play with it for a while and get back to collecting data for a variable speed pump; I finally had time last night.  I look at the pad inside the house and saw that it was frozen, none of the buttons would work.  I unplugged the usb2rs485 board from the rpi since it had a solid red light.  The pad immediately started responding.  I plugged the board back in and it immediately froze again; disconnecting it cleared it again.  It had been successfully working with 0x08 but I am wondering if became a conflict after we had a power outage the other day.  The config has not changed but I am going to try another address and see if the symptoms change.

napertivo, 2018-08-16:
Ok, I don't think it has anything to do with the device_id; it locks up as soon as connected to the USB.  I'm thinking maybe something fried the board so I ordered another.  The odd thing is that the red light comes on steady and never flashes.

sfeakes, 2018-08-19:
It doesn't sound like the device ID.  Sounds like either the RS 485 to USB adapter or something with the PI going nuts on the USB adapter.  It does depend on the board#, but the red light usually mean activity.  So solid red would be a constant send / receive or communication error.  Next thing to test would be to stop AqualinkD and plug / unplug the USB into RPi and see if the results are the same or different.

napertivo, 2018-08-19:
Yeah I think it is the board.  The solid red light that comes on does it even without iaqualinkd running.  I'm not sure it is the same led that would flicker with activity but can't tell until the replacement comes in later today.  Not sure if it sheds any light on the issue but if I remove the 5v jumper from the board it no longer locks up the panel.

sfeakes, 2018-08-19:
The 5v jumper is purely for where the board gets power.  Either power from USB or power from RS485, if you power from RS rather than USB you need to connect 3 wires. So if you only have 2 wires connected and remove that jumper, it will try to power from RS, but can't because you haven't connected the RS power wire.

sfeakes, 2018-08-19:
did it come up as /dev/ttyUSB1 ?

This should tell you if the PI found the board at least.
```dimesg | grep usb```

Then from the output that, it could take you down some other paths.

What you are looking for is something like the below, they are the important lines.
```
...
[    1.643842] usb 1-1: Product: FT232R USB UART
[    1.646974] usb 1-1: Manufacturer: FTDI
[    1.649988] usb 1-1: SerialNumber: A50285BI
.....
[   11.227839] usb 1-1: Detected FT232RL
[   11.414036] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

Complete output 
```
[    0.260806] usbcore: registered new interface driver usbfs
[    0.260976] usbcore: registered new interface driver hub
[    0.261228] usbcore: registered new device driver usb
[    0.462333] usbcore: registered new interface driver smsc95xx
[    0.944991] dwc_otg 20980000.usb: DWC OTG Controller
[    0.947698] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.950465] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    0.958591] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.961311] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.963946] usb usb1: Product: DWC OTG Controller
[    0.966496] usb usb1: Manufacturer: Linux 4.9.67+ dwc_otg_hcd
[    0.969008] usb usb1: SerialNumber: 20980000.usb
[    0.979226] usbcore: registered new interface driver usb-storage
[    1.016547] usbcore: registered new interface driver usbhid
[    1.019295] usbhid: USB HID core driver
[    1.400903] usb 1-1: new full-speed USB device number 2 using dwc_otg
[    1.637350] usb 1-1: New USB device found, idVendor=0403, idProduct=6001
[    1.640662] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.643842] usb 1-1: Product: FT232R USB UART
[    1.646974] usb 1-1: Manufacturer: FTDI
[    1.649988] usb 1-1: SerialNumber: A50285BI
[    9.657445] usbcore: registered new interface driver brcmfmac
[   11.074557] usbcore: registered new interface driver usbserial
[   11.074684] usbcore: registered new interface driver usbserial_generic
[   11.074826] usbserial: USB Serial support registered for generic
[   11.227093] usbcore: registered new interface driver ftdi_sio
[   11.227238] usbserial: USB Serial support registered for FTDI USB Serial Device
[   11.227839] usb 1-1: Detected FT232RL
[   11.414036] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

closed 2018-08-19 by sfeakes

napertivo, 2018-08-19:
Thank you!  That did get me pointed in the right direction.  So, in the interest is helping others, I will share my stupidity.  Surprisingly if you incorrectly connect the power and serial cables to a Pi Zero it will actually power up just fine however the serial connection will not work; I had cabled the serial and power backwards and did not notice.

So I am back to my original troubleshooting and cleanup.  It is operating again and appears fine however tailing the syslog it is not as clean as it should be.  This is what I am seeing:

Aug 19 14:55:24 RPiAqualink aqualinkd: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Aug 19 14:55:29 RPiAqualink aqualinkd: Control Panel B0029221 REV T
Aug 19 14:55:29 RPiAqualink aqualinkd: Getting freeze protection setpoints
Aug 19 14:55:29 RPiAqualink aqualinkd: Serial read bad checksum, ignoring
Aug 19 14:55:35 RPiAqualink aqualinkd: Serial read bad checksum, ignoring
Aug 19 14:55:38 RPiAqualink aqualinkd: Getting pool & spa heat setpoints from aqualink
Aug 19 14:55:40 RPiAqualink aqualinkd: Serial read bad checksum, ignoring
Aug 19 14:55:46 RPiAqualink aqualinkd: Serial read bad checksum, ignoring

Tailing the aqualinkd log I am seeing this:
Aug-19-18 15:08:53 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Message length 24
Aug-19-18 15:08:54 PM Info:   RS Message :- '3:07 PM'
Aug-19-18 15:08:54 PM Debug:  time not checked, will check in 2819 seconds
Aug-19-18 15:08:54 PM Debug:  RS time is accurate '3:07 PM 08/19/18 Sun'
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:54 PM Debug:  RS Received STATUS length 12.
Aug-19-18 15:08:54 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:55 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:55 PM Debug:  RS received packet of type Status length 12
Aug-19-18 15:08:55 PM Debug:  RS received packet of type Status length 12

Is 12 a normal length?  If so I can filter those out.

sfeakes, 2018-08-20:
Yes 12 is the normal length.  If you get the latest version there are a few more logs and fixes that *may* help with the checksum error.
The checksum is always because Aqualinkd didn’t get the complete packet, not get the complete packet is usually for one of two reasons.
 1) another device is taking some of the packet (device ID conflict). 
 2) some problem with AqualinkD reading (bug in my software, or bad serial connection).