aqualinkd/AqualinkD #16: Jandy Heater Control Only

Astroman, 2018-08-19:
Since I have an IQPump01 hooked up to my VS pump, I only need to control my Jandy JXi heater. Is there any reason I couldn't simply hook up to the RS-485 in the heater directly and use this software for remote control of just the heater?

Thanks for this great project!

sfeakes, 2018-08-19:
It wouldn't work as is.  RS485 works from on the principle of one device controlling the network and many devices acting as slaves to the controller.  In this case, the control panel is the controller and all equipment including AqualinkD are slaves.  Each device talks through the controller to other devices, they can't talk directly.  So without a control panel, this would not work.
In saying that, it is quite simple to create a controller. Before I got my control panel comunicating with my salt water generator, I did exactly what you want to do.  Used a version of AqualinkD to act as the control panel / RS485 server to the salt water generator.  Then I could control the Salt Water Generator.   But each device is different in how and what they communicate, so you'd need to spend hours deciphering that since none of it is published.

There is this project that can control some RS485 devices directly (ie act as a RS485 server), not sure if they have what your looking for with the JXi heater though.
https://github.com/tagyoureit/nodejs-poolController

Astroman, 2018-08-19:
Thanks for the info. Still hoping Janet will come out with a simple heater control like the IQPump01.

closed 2018-08-20 by sfeakes

timelery, 2020-03-06:
> Thanks for the info. Still hoping Janet will come out with a simple heater control like the IQPump01.
> […](#)

Did you ever find a solution to controlling the Jandy Jxi heater?

Astroman, 2020-03-06:
No, never did. The heater actually has a simple switch closure mechanism for activating the high and low temperatures so I was thinking of hooking up a couple of relays and valve actuators controlled via my home automation system. This wouldn't provide temperature control but I normally don't need to change it anyway.

timelery, 2020-03-06:
> No, never did. The heater actually has a simple switch closure mechanism for activating the high and low temperatures so I was thinking of hooking up a couple of relays and valve actuators [...]

I did the exact thing against the fireman switch and that allow sme to swith the spa and pool heater on. However, I was hoping to tap into the RS485 bus so I could read the temperature coming from the Jxi. With the ultimate goal of a simple push notification when the spa hit temperature.

If you come across any info on how to properly read the messages coming from Jxi over RS485 please let me know and I will do the same.

timelery, 2020-03-06:
> > No, never did. The heater actually has a simple switch closure mechanism for activating the high and low temperatures so I was thinking of hooking up a couple of relays and valve actuators controlled via my home automation system. This wouldn't provide temperature control but I normally don't need to change it anyway.
> 
> I did the exact thing against the fireman switch and that allows me to switch the spa and pool heater on. However, I was hoping to tap into the RS485 bus so I could read the temperature coming from the Jxi. With the ultimate goal of a simple push notification when the spa hit temperature.
> 
> If you come across any info on how to properly read the messages coming from Jxi over RS485 please let me know and I will do the same.

@sfeakes Do you have any recommendations on how to begin with trying to decode the Jxi messages? Starting with the response below?

[image]

timelery, 2020-03-06:
I think a great first step would be to figure out how to construct the message to send to the Jxi heater to put it into Remote Mode like below.

[image]

Astroman, 2020-03-06:
I haven't looked into the protocols at all. I keep hoping Zodiac will come out with a simple device for heater control like they have for the pump.

When you hooked up to the contacts to turn the heater on and off, was it simply a matter of a switch closure between a common terminal and the low or high temperature terminal? Or did the temp terminal need to be connected to ground or a voltage?