aqualinkd/AqualinkD #17: Can't find right ID and nothing happening

stulincoln, 2018-08-26:
I have gone through the process but all I see on the interface are 'serial bad checksum' errors. When I am running the serial logger to get available ports I don't see anything like you have in the documentation, mine all seem to start at around 0x3 and higher. I did eventually see 0x00 is there so I tried that. Once the service is running I mainly see "RS received packet of Type Ack length 3" or "RS received packet of type Unknown '0x1f' length 12"

A couple of questions I have:

* when I connect my adapter onto my Jandy board, do the wires just go in the same two slots my current Jandy PDA is going into as it is already in there. I have four slots so haven't touched the outside two.

* Do I need to somehow map the id's for each device then?

Thanks for your help.

Stu

sfeakes, 2018-08-26:
First check you have the latest version, I updated it yesterday.  The ID of 0x00 is the main control panel, so don't use that.  It sounds like you may have the RS485 to USB adapter connected incorrectly. Each control panel is different, but you should be connecting it to the Data - and Data + lines, not the power - and power + lines, these are generally the middle two connections on the 4 post connector at the top-left of the control panel.  See the wiki for a picture.  The other issue you may have is connecting them backwards, so you could try reversing the wires.
I've never looked at the Jandy PDA, but I would assume it uses the same two wires, I'll look at the documentation to the now and let you know.

sfeakes, 2018-08-26:
If this is the Jandy PDA you have :-
https://www.jandy.com/~/media/zodiac/global/downloads/h/h0574200.pdf
Then on page 7 where it has the 4 wires to connect, you should be using the yellow and black wires.

sfeakes, 2018-08-27:
That all looks correct, I’ll check the wiring when I’m at home tomorrow night.  Can you post the logs of the output of the serial logger.  0x58 isn’t a valid ID for AqualinkD, it can use either of these, 0x0a, 0x0b, 0x08, 0x09 due to the way the controls panel communicates. (It mimics an all button keypad).  Also do you know the information of your control panel?  It is an AqualinkRS panel?

sfeakes, 2018-08-27:
Just looking at some Jandy documentation 
  Pin 2 (Black) is Data +
  Pin 3 (yellow) is Data -

On your RS485 to USB adapter.
  A is Data -
  B is Data +

So, from the pic's it looks like you have it wired backwards.

Also when you are running the serial_logger make sure AqualinkD is not running otherwise you will get a lot of errors as they will conflict with each other fighting to read data off the adapter.
```
sudo systemctl stop AqualinkD
sudo ./release/serial_logger /dev/ttyUSB0
```
The question about cleaning up the AqualinkD install, just run the install script and it'll overwrite all files except the config files.

stulincoln, 2018-08-28:
I went ahead and changed it to 0x08 and fixed the wiring.  I confirmed this is the PDA-PS6. 

Stopped the service and ran the serial logger with these results below. This is strange as I thought I was seeing more happen on here. As you can see, I don't see any of those 0x0 addresses. Could that be my problem?

Thanks

Stu

Notice: Logging serial information!
Please wait.Notice: ID's found
Notice: ID 0x60 is in use
Notice: ID 0x6a is not used
Notice: ID 0x6b is not used
Notice: ID 0x38 is not used
Notice: ID 0x39 is not used
Notice: ID 0x3a is not used
Notice: ID 0x3b is not used
Notice: ID 0x50 is not used
Notice: ID 0x78 is in use
Notice: ID 0x51 is not used
Notice: ID 0x52 is not used
Notice: ID 0x53 is not used
Notice: ID 0x80 is not used
Notice: ID 0x58 is not used
Notice: ID 0x59 is not used
Notice: ID 0x5a is not used
Notice: ID 0x5b is not used
Notice: ID 0x68 is not used
Notice: ID 0x69 is not used
Notice: ID 0x81 is not used
Notice: ID 0x30 is in use

sfeakes, 2018-08-28:
You can get more info with the -d option.   But the controller is never probing for a all button panel address.  So it looks like you have the PDA only Aqualink and not AqualinkRS.  This software doesn’t work with the PDA only Aqualink, it was designed for the AqualinkRS.  One other person has asked for PDA support, and I started working on it, but have not had time to finish it.  I’ll post some updates when I finish it.

stulincoln, 2018-08-28:
Thanks, I just realized that we my issue as well. Looking forward to seeing the PDA support and happy to test anything with you. Thanks for your time in troubleshooting, will make another coffee donation.

Thanks

Stu

closed 2018-09-02 by sfeakes