aqualinkd/AqualinkD #18: Issue reading temp setpoints

abarbaccia, 2018-08-29:
Thank you for the great work on this! It's a nice addition to my home automation setup.

I am having trouble with some of the components involving temperatures. See below the status I get from the API. For example, pool temp is not showing however I see the data on the RS bus. 

{"type": "status","status":"AquaPure OFF","version":"6524 REV GG","aqualinkd_version":"1.1","date":"08/29/18 WED","time":"9:37 AM","pool_htr_set_pnt":"-999","spa_htr_set_pnt":"-999","frz_protect_set_pnt":"38","air_temp":"82","pool_temp":" ","spa_temp":" ","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "on","Spa_Mode": "off","Aux_1": "off","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "Freeze_Protect": "enabled"}}

Daemon logs
https://pastebin.com/7JjBGwBD

It seems, given the status from API, that it is not being pulled off the bus correctly to begin with.   

This will also show up with the pool heater set points. I can toggle on / off the pump, cleaner, heater, lights without issue. I can also read the Pool Air Temp value. 

Happy to help debug and test any changes. My c is lacking however :)

abarbaccia, 2018-08-29:
Worth noting the very old version number on the panel. Suspect that has something to do with it!

abarbaccia, 2018-08-29:
> I see the problem. All other panels I've come across report "Pool Temp"
> and "Spa Temp", yours states "Water Temp". I'm looking for Pool or Spa in
> the message and not Water. I'll make a change in a few days to account for
> this.
> Do you have a spa? As I'm wondering if I need to use "water temp" and set
> that to "pool temp" or "spa temp" depending on the status of the "spa mode"
> switch. If you have a spa, if you could turn that on, and post another set
> of logs? That would be helpful, as then I can see if there is any change in
> the message.

That was quick!  Thanks!!   No, I do not have a spa, only pool. Sorry about
that.
Will this change account for also the temperature setpoints on the pool
heater as well? This seems to be acting up as well (reporting a large neg
value on the temp, not letting me set other values, etc).

sfeakes, 2018-08-29:
Yes I will have to account for the heater setpoint as well.  Can you post the logs from when you have tried to set that, as I’m guessing it also states water and not pool in the message for setting the temperature but I’d like to know for sure.

abarbaccia, 2018-08-29:
Logs attached
https://pastebin.com/FxjEXDC9

abarbaccia, 2018-08-29:
> Looking at the logs, it seems your system is not listing the "heater setpoint" in the way other control panels do. So far I have only seen the below message:-
>
> POOL TEMP IS SET TO xx
> SPA TEMP IS SET TO xx
> MAINTAIN TEMP IS SET TO xxx
> So I was expecting 'WATER TEMP IS SET TO xx' on yours rather than POOL & SPA But there are two values that came back
>
> TEMP1 (HIGH TEMP) IS SET TO 82F
> TEMP2 (LOW TEMP) IS SET TO 74F
> I'm not sure what they represent, do you know what you have set to 82F and 74F ?

Ok, interesting..

I looked and it seems there is no spa functionality on my controller. 

There are two buttons for the heater. Temp1 and temp2. I can set the target temps through the programming functions. 

On the panel there are buttons to trigger temp1 and temp2. If you push them, the controller will turn heater on until that temperature is achieved. At most only one can be active at a time. 

For starters, if we can mimic the buttons and sensors for temp1 and temp2 that would be a huge step forward. That way I can see heater status and turn on/ off. 

If we can also implement a way to change the setpoit, then I can build it as a climate control device in home assistant (eg just take over temp2 for this purpose). 

How are you thinking about it? 

_____________________________________________________________________________________________________

Andrew Barbaccia 
Project Leader 

THE BOSTON CONSULTING GROUP
446 Springfield Ave
07901 Summit, New Jersey ▪ United States of America

Mobile +1 2015638617
barbaccia.andrew@bcg.com

_____________________________________________________________________________________________________

Read BCG's latest insights, analysis, and viewpoints at bcgperspectives.com

abarbaccia, 2018-08-29:
Also, note, turning on the pool heater switch will activate temp1 currently.

sfeakes, 2018-08-31:
I've uploaded a new version that should work for reading your water temp and heater setpoint now, but I don't have enough information for the setting of the heater setpoint.  I played with my control panel hoping if I put it in single device mode (ie no spa) it would act like yours, but it didn't, so I have been unable to fully test the changes.
Please test the latest version, hopefully you not get pool temp and heater setpoint showing correctly, and once running try to set the heater setpoint (I'm sure it will fail) but post the logs and I'll be able to fix that as well.

abarbaccia, 2018-08-31:
I am away from the house this weekend, but quickly tested over a terminal.
It correctly pulled the water temp from the bus -- NICE!. Unfortunately the
heater set point is still showing -999. When I get back home after the
holiday weekend I can run a few more tests, including changing the heater
setpoint on the panel so you can see how that works in the logs.
Thanks again!

sfeakes, 2018-08-31:
OK, figured out the heater set point (I can't count).

abarbaccia, 2018-09-04:
Hi 

I was able to test and capture some logs:
http://paste.ubuntu.com/p/dG9MmZKjQX/

A few different pieces in here .. please let me know if you would like to track them in separate issues or keep consolidated in this one.  We may want to call this "RS6 support" since I think this is driving the differences vs RS8 panels. 

1)   The pool temp set point.  This still does not seem to be captured correctly. Please let me know how else I can help you debug. 

2)  The process to change the TEMP1 setpoint. This is through the programming menu. 

3)  Also, not in the logs, but the heater mode may need further tweaking. I am able to turn the heater on correctly ... light comes on, heater_enabled says true.  However, when I try to turn off it remains enabled. This is true in the software and the panel so probably a slightly different message may be required for RS6.

Thanks and looking forward to get your thoughts!

abarbaccia, 2018-09-04:
A bit more info on #3.  It seems I have hit a corner case. Here is the situation. If you have both the pump and heater enabled .. then turn the pump off. The heater should stay enabled while the pump turns off .. however the Pool_Heater gets turned off in MQTT / HomeAssistant. Ironically the sensor "Heater enabled" continues to show correctly so not sure what is going on. Either way, because heater turns off with pump, it prevents actually turning off the heater unless you first turn the pump back on (where both will turn on).  This can be logged into a separate issue since it probably affects all platforms.

sfeakes, 2018-09-04:
I had a quick look, I will take some more time during the week and give you a better reply. But quickly.
From the last log you posted, it looks like it’s working (except reading heater set point).
Can you try the built in web ui, as that may give you a better idea. There may be some issues with how hass.io is configured.
Start AqualinkD and let it run for a minute before doing anything, as it has to program the control panel. Build in web ui will let you know then it’s finished.
There are 4 modes to each led (on, off, enable & flash). The web ui will show you these, in the form of on off (obvious) enable will be green for a heater and on will be red.  Flash will state delay, and flash.
The pump status will not effect the heater off or enabled, you should be able the change those independently, are you saying that you can’t change them independently on the keypad or in AqualinkD?

You can also use ‘mosquito_sub -t “aqualinkd/#” -v’ to see what gets posted to MQTT

abarbaccia, 2018-09-04:
Thanks for the debugging tips. I’ll test this week to see what’s being posted to MQTT.

To your question, yes I can control pump and heater independently. 

More to come!

sfeakes, 2018-09-15:
This is fixed in latest update. Now supports "Pool OR Spa Only" mode not just "Pol and Spa" mode.

abarbaccia, 2018-09-18:
Fantastic. Tested and confirmed it can read the values correctly! Thanks

Should this commit also allow me to change the temp setpoint? If not, no worries, it works great with the two setpoints i have configured already.

sfeakes, 2018-09-18:
Is should work for setpoints as well now.  I can’t test it, but I’ve had one other person confirm it workes.

AUPoolUser, 2018-09-19:
Hi All,

I am unable to set the Temp1 or Temp2 setpoints in the Rev.MMM Pool Only Version of Aqualink.

Here is the log of trying to change the temperature via the webpage:

Sep-19-18 13:32:24 PM Info:   URI request: '/config.js '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/background.jpg '
Sep-19-18 13:32:24 PM Info:   WS: Message - Key 'command' Value 'GET_DEVICES' | Key2 '(null)' Value2 '(null)'
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_3-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_2-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_2-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_1-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Filter_Pump-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_3-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_4-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_4-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_5-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_5-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_6-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_1-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Solar_Heater-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_6-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Solar_Heater-on.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_7-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Filter_Pump-off.png '
Sep-19-18 13:32:24 PM Info:   URI request: '/hk/Aux_7-on.png '
Sep-19-18 13:32:24 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:32:26 PM Info:   RS Message :- 'WATER TEMP 18 C'
Sep-19-18 13:32:26 PM Info:   MQTT: Published id=77: aqualinkd/Temperature/Spa 18.00
Sep-19-18 13:32:26 PM Info:   MQTT: Message publishing acknowledged (msg_id: 77)
Sep-19-18 13:32:26 PM Info:   MQTT: received (msg_id: 77), looks like my own message, ignoring
Sep-19-18 13:32:26 PM Info:   MQTT: Published id=78: aqualinkd/Temperature/Spa -999.00
Sep-19-18 13:32:26 PM Info:   MQTT: Message publishing acknowledged (msg_id: 78)
Sep-19-18 13:32:26 PM Info:   MQTT: received (msg_id: 78), looks like my own message, ignoring
Sep-19-18 13:32:27 PM Info:   WS: Message - Key 'command' Value 'Pool_Heater' | Key2 '(null)' Value2 '(null)'
Sep-19-18 13:32:27 PM Info:   WS: button 'Pool_Heater' pressed
Sep-19-18 13:32:27 PM Info:   Queue send '0x12' to controller
Sep-19-18 13:32:27 PM Info:   RS Message :- 'TEMP1 DISABLED'
Sep-19-18 13:32:28 PM Info:   MQTT: Published id=79: aqualinkd/Pool_Heater 0
Sep-19-18 13:32:28 PM Info:   MQTT: Published id=80: aqualinkd/Pool_Heater/enabled 0
Sep-19-18 13:32:28 PM Info:   MQTT: Message publishing acknowledged (msg_id: 79)
Sep-19-18 13:32:28 PM Info:   MQTT: received (msg_id: 79), looks like my own message, ignoring
Sep-19-18 13:32:28 PM Info:   MQTT: Message publishing acknowledged (msg_id: 80)
Sep-19-18 13:32:28 PM Info:   MQTT: received (msg_id: 80), looks like my own message, ignoring
Sep-19-18 13:32:29 PM Info:   RS Message :- '09/19/18 WED'
Sep-19-18 13:32:31 PM Info:   RS Message :- '1:31 PM'
Sep-19-18 13:32:33 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:32:35 PM Info:   RS Message :- 'WATER TEMP 18 C'
Sep-19-18 13:32:35 PM Info:   MQTT: Published id=81: aqualinkd/Temperature/Spa 18.00
Sep-19-18 13:32:35 PM Info:   MQTT: Message publishing acknowledged (msg_id: 81)
Sep-19-18 13:32:35 PM Info:   MQTT: received (msg_id: 81), looks like my own message, ignoring
Sep-19-18 13:32:35 PM Info:   MQTT: Published id=82: aqualinkd/Temperature/Spa -999.00
Sep-19-18 13:32:35 PM Info:   MQTT: Message publishing acknowledged (msg_id: 82)
Sep-19-18 13:32:35 PM Info:   MQTT: received (msg_id: 82), looks like my own message, ignoring
Sep-19-18 13:32:37 PM Info:   RS Message :- '09/19/18 WED'
Sep-19-18 13:32:37 PM Info:   WS: Message - Key 'command' Value 'Pool_Heater' | Key2 '(null)' Value2 '(null)'
Sep-19-18 13:32:37 PM Info:   WS: button 'Pool_Heater' pressed
Sep-19-18 13:32:37 PM Info:   Queue send '0x12' to controller
Sep-19-18 13:32:37 PM Info:   WS: Message - Key 'parameter' Value 'Pool_Heater' | Key2 'value' Value2 '40'
Sep-19-18 13:32:37 PM Info:   Queue send '0x09' to controller
Sep-19-18 13:32:38 PM Info:   RS Message :- 'PRESS ^?~ FOR MORE OPTIONS... PRESS ENTER* TO SELECT'
Sep-19-18 13:32:38 PM Info:   Queue send '0x18' to controller
Sep-19-18 13:32:39 PM Info:   RS Message :- 'PRESS ^?~ FOR MOR'
Sep-19-18 13:32:39 PM Info:   RS Message :- 'PROGRAM      ^?~*'
Sep-19-18 13:32:39 PM Info:   Queue send '0x18' to controller
Sep-19-18 13:32:39 PM Info:   RS Message :- 'REVIEW       ^?~*'
Sep-19-18 13:32:39 PM Info:   Queue send '0x18' to controller
Sep-19-18 13:32:39 PM Info:   MQTT: Published id=83: aqualinkd/Filter_Pump 0
Sep-19-18 13:32:40 PM Info:   RS Message :- 'SET TEMP     ^?~*'
[... 108 lines cut ...]
Sep-19-18 13:33:55 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:33:57 PM Info:   RS Message :- 'PUMP OFF'
Sep-19-18 13:33:59 PM Info:   RS Message :- '09/19/18 WED'
Sep-19-18 13:34:01 PM Info:   RS Message :- '1:33 PM'
Sep-19-18 13:34:03 PM Info:   RS Message :- 'POOL NAME'
Sep-19-18 13:34:05 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:34:07 PM Info:   RS Message :- 'PUMP OFF'
Sep-19-18 13:34:09 PM Info:   RS Message :- '09/19/18 WED'
Sep-19-18 13:34:11 PM Info:   RS Message :- '1:33 PM'
Sep-19-18 13:34:13 PM Info:   RS Message :- 'AIR TEMP 18 C'
[... cut ...]
Sep-19-18 13:34:27 PM Info:   RS Message :- '1:33 PM'
Sep-19-18 13:34:29 PM Info:   RS Message :- 'POOL NAME'
Sep-19-18 13:34:31 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:34:33 PM Info:   RS Message :- 'PUMP OFF'
Sep-19-18 13:34:35 PM Info:   RS Message :- '09/19/18 WED'
Sep-19-18 13:34:37 PM Info:   RS Message :- '1:34 PM'
Sep-19-18 13:34:39 PM Info:   RS Message :- 'AIR TEMP 18 C'
Sep-19-18 13:34:41 PM Info:   RS Message :- 'PUMP OFF'
Sep-19-18 13:34:43 PM Info:   RS Message :- '09/19/18 WED'

sfeakes, 2018-09-19:
Can you post the complete log as an attachment?   The code that detects pool or spa rather than pool and spa doesn’t look like it’s working, to it’s trying to set a pool setpoint rather than Temp1.   That code gets executed at startup.  So if you can start aqualinkd wait for programming to finish, then try to set a setpoint.   Then post the entire log, that would be really helpful.

AUPoolUser, 2018-09-19:
Hi Shaun - Log as requested.

[aqualinkd.log.pdf](https://github.com/sfeakes/AqualinkD/files/2397524/aqualinkd.log.pdf)

sfeakes, 2018-09-19:
Thanks.
This message, looks like you are not on the latest version.
Sep-19-18 23:53:13 PM Notice: Aqualink Daemon **v1.2.1** 

V1.2.2 has some fixes to automatically detect "Pool AND Spa" vs "Pool OR Spa".  So I think this may be your problem.

AUPoolUser, 2018-09-19:
Ooops - here is the updated log for v1.2.2
[aqualinkd2.log.pdf](https://github.com/sfeakes/AqualinkD/files/2399076/aqualinkd2.log.pdf)
....

AUPoolUser, 2018-09-23:
Shaun.
Testing on 1.2.2d of (Pool Only) Aqualink RS-P8:

I’ve done some basic testing
•	Setting Temp1 works well under various situations.
o	I tried setting temp1 lower than temp 2 and it tries to program for a minute or so but nothing happens which is what you would expect.
•	Setting Temp2 works well under various situations.
o	If I try to set temp2 above temp1 nothing happens – so I can’t confuse it.
o	Although I am getting various “connection error!” problems playing with Temp2. I will try to recreate something that replicates the same each time and report back. BTW I fix it by rebooting the pi!

I’d say this 1.2.2d version is an outstanding success.

Shuan, you are a genius!! THANK YOU SO MUCH.

closed 2018-10-08 by sfeakes