aqualinkd/AqualinkD #19: Powering RaspberryPi in control panel

napertivo, 2018-09-03:
I currently have my RaspberryPi sitting on top of my Aqualink RS OneTouch panel in the house but I was thinking of installing it within the control panel.  Has anyone cabled it in the panel?

sfeakes, 2018-09-03:
I had a PI setup in my control panel outside, (before I got my SWG working with the Control Panel I had a version or AqualinkD controlling the SWG).  I had a lot of instability due to the heat in the summer. I do live in a very hot climate, and the panel got direct sunlight for the majority of the day though.

napertivo, 2018-09-04:
Thank you I'm going to try this.  Well first powering it from the OneTouch so I don't have a power cord hanging down the wall.  I'll think about moving it to the panel later, I'm wondering if the heat will be an issue for me as well (Dallas).

sfeakes, 2018-09-15:
This is great.  I always wondered if I could make it all fit in the RS6 keypad. Know I know.

LogoiLab, 2020-06-14:
If people have interest in this I could design a dedicated board that can sit right next to the original that the Pi can just snap into. No more dongles or electrical tape.

lehighkid, 2020-07-18:
I am interested in this.  I already have a buck converter and a 12 to 5v step down.  I do not have an interior panel so I am looking to install in the control panel.  If I understand, the low voltage rs485 power is 12v and I can tap into that, correct?

sfeakes, 2020-07-18:
Be careful, not all are 12v. (Most are though).  So make sure to use a meter if your DC/DC converter expects 12v rather than a range. If installing in panel, you already have 110v, so you can simply add a USB adapter if needed. There is nothing wrong with this approach, and I'm currently doing something very similar. BUT Pi's can be really finicky on power, and if you have a power issue, the first thing that will start acting up is USB and CPU speed. That will have detrimental effects on AqualinkD as RS485 and timing is key. So if you go down this path, just make sure to spec everything appropriately.

lehighkid, 2020-07-18:
Thank you. I used a voltmeter and it was DC voltage around 15V. I was able to step it down to 5V/3A by connecting 1(+)/4(-). I then connected 2(+)/3(-) to my adapter. So far so good. 
[image]
[image]
[image]
[image]

peril99, 2020-08-03:
I left the pi's outside of the box, and have one of these tucked in the aqualink panel for power.

[image]

I have aspirations of more advanced pool  functions so added in a a 120vac -> 20vdc convertor, and then bolted up the dc dc step down.

"Aiposen 110V/220V AC to DC 24V 2A 48W Switch Power Supply Driver,Power Transformer for CCTV Camera/Security System/LED Strip Light/Radio/Computer Project(24V 2A)"

https://www.amazon.com/gp/product/B01B1PRE42/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

and  

"BINZET DC Converter Step Down Regulator 5V Regulated Power Supplies Transformer Converter (5V 3A 15W)"
https://www.amazon.com/gp/product/B00J3MHRNO/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1