aqualinkd/AqualinkD #20: Enhancement: MQTT LWT status

abarbaccia, 2018-09-05:
MQTT has a really nice feature that you can set a "last will" message when you make a connection. In the event aqualinkd drops the connection ungracefully, the MQTT server will publish the message on aqualinkd behalf. It's really useful for Online / Offline status of clients.

More available here:
https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

Expect it would be a minor change when opening the connection.

sfeakes, 2018-09-05:
Thanks for the suggestion. AqualinkD already publishes with the retain flag, so anything connecting should get the last set of values AqualinkD posted directly from the MQTT server.  So I’m not sure what this adds over that from an AqualinkD perspective.  If a client dies, AqualinkD really doesn’t care about knowing that. At the time the client reconnects it will get the latest messages AqualinkD sent.
What problem / or potential issue do you see this solving?

abarbaccia, 2018-09-06:
Sorry for the poor description! Client = aqualinkd, server = mqtt broker. 

This would let mqtt broker know if aqualinkd is running, or if it crashed. When daemon starts you post “online” message to status topic. At the same time you also set the last will message to be “offline”. When stopping the daemon you should post “offline” to the status message. If aqualinkd dies unexpectedly (eg WiFi drops, error, etc), then mqtt server will post the last will message on aqualinkd behalf. It’s a nice feature of mqtt tbh.

sfeakes, 2018-09-07:
Now I understand what you are after.
I've made the changes, Last Will Message is now supported.
/aqualinkd/Alive (0 or 1)

abarbaccia, 2018-09-07:
That's fantastic. Thanks so much!! I'll test it this weekend. Much
appreciated!

abarbaccia, 2018-09-08:
works like a charm! Thanks!!

closed 2018-09-08 by abarbaccia