aqualinkd/AqualinkD #21: ALRS 4 P & S  8158 JJ

jsrichie, 2018-09-07:
Perfect DYI Job Love the work. But seems I need help getting mine online.
I checked and verified my version of the chip as ALRS 4 P & S  8158 JJ.
My PCB, on the other hand, is a question: 
Jandy Ind Power Center PCB8124A Assembly #8125

I have a Heater and Salt Gen Attached all of which are seen and controlled by the Square Remote.
LAARS LX400
Goldline 2003 / Cell shows as NatureSoft(a replacement cell)
I get no errors and the process runs without issue.

I can not get the serial logger to anything but:
Notice: Logging serial information!
Please wait.^CNotice: Stopping!
Notice: ID's found
Notice: ID 0x7c is in use 
I waited overnight to allow all my equipment to run.

Pi zero
USBtoRS485 wiring verified. 

I seem to match the firmware needed but can't seem to find the right combo
Any Ideas.

sfeakes, 2018-09-07:
If serial logger is blank, it’s not connecting / seeing anything on the RS485 network.  Until you get something in the logger nothing will work.
First check you have USB connected, can you post output of ‘dmesg | grep USB’.
If that works and get something similar to the output listed in the wiki, try the serial logger with the -d option.

jsrichie, 2018-09-07:
pi@raspberrypi:~/software/AqualinkD $ dmesg | grep USB
[    0.850267] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.871118] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.881013] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.921525] hub 1-0:1.0: USB hub found
[    1.153749] usbhid: USB HID core driver
[    1.386376] usb 1-1: new full-speed USB device number 2 using dwc_otg
[    1.805846] usb 1-1: New USB device found, idVendor=0403, idProduct=6001
[    1.805865] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.805873] usb 1-1: Product: FT232R USB UART
[   15.368965] usbserial: USB Serial support registered for generic
[   15.629387] usbserial: USB Serial support registered for FTDI USB Serial Device
[   15.629743] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
[   15.663242] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0

sfeakes, 2018-09-07:
So that's all good, Linux is recognising the USB adapter, and can see the serial chip in the adapter.  Can you check you are seeing something on USB0 with the post above.
`sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0`

If that's blank, then check the wiring.  If you are at a total loss, try reversing the A and B (Data+ and Data-) wires on the USB adapter.

jsrichie, 2018-09-07:
reversed the A and B and were good:
Notice: Logging serial information!
Please wait.Notice: ID's found
Notice: ID 0x10 is not used 
Notice: ID 0x38 is in use 
Notice: ID 0x50 is in use 
Notice: ID 0x40 is in use 
Notice: ID 0x11 is not used 
Notice: ID 0x12 is not used 
Notice: ID 0x13 is not used 
Notice: ID 0x18 is not used 
Notice: ID 0x19 is not used 
Notice: ID 0x1a is not used 
Notice: ID 0x1b is not used 
Notice: ID 0x20 is not used 
Notice: ID 0x21 is not used 
Notice: ID 0x22 is not used 
Notice: ID 0x23 is not used 
Notice: ID 0x28 is not used 
Notice: ID 0x29 is not used 
Notice: ID 0x2a is not used 
Notice: ID 0x2b is not used 
Notice: ID 0x30 is not used 
Notice: ID 0x31 is not used 
Notice: ID 0x32 is not used 
Notice: ID 0x33 is not used 
Notice: ID 0x39 is not used 
Notice: ID 0x3a is not used 
Notice: ID 0x3b is not used 
Notice: ID 0x48 is not used 
Notice: ID 0x49 is not used 
Notice: ID 0x4a is not used 
Notice: ID 0x4b is not used 
Notice: ID 0x51 is not used 
Notice: ID 0x52 is not used 
Notice: ID 0x53 is not used 
Notice: ID 0x08 is not used  <-- can use for Aqualinkd
Notice: ID 0x09 is not used  <-- can use for Aqualinkd
Notice: ID 0x0a is not used  <-- can use for Aqualinkd
Notice: ID 0x0b is not used  <-- can use for Aqualinkd
Notice: ID 0x41 is not used 
Notice: ID 0x42 is not used 
Notice: ID 0x43 is not used 
Notice: ID 0x00 is not used 

Now a question? the RS panel has buttons on the right side: 
Pool Mode
Spa Mode
Spa Drain
Spa Fill

where to these show and how to button them?
Thanks

sfeakes, 2018-09-08:
First two are fixed all others (8 in total) are configured in the config.  Last part of the file.  Let me know if the wiki doesn’t explain it well.

closed 2018-09-11 by sfeakes