aqualinkd/AqualinkD #25: homebridge and aqualinkd on two different machines

alanfromtoronto, 2018-10-06:
Great project and I think I got way close to the end but can't get mqtt messages from aqualinkd to homebridge machine.

when I check the daemon.logs on the homebridge machine I just get the following: 
Oct  1 06:26:03 raspberrypi homekit2mqtt[637]: 2018-10-01 06:26:03.220 #033[30m#033[42m<info> #033[49m#033[39m mqtt reconnect
Oct  1 06:26:04 raspberrypi homekit2mqtt[637]: 2018-10-01 06:26:04.225 #033[30m#033[42m<info> #033[49m#033[39m mqtt reconnect
Oct  1 06:26:05 raspberrypi homekit2mqtt[637]: 2018-10-01 06:26:05.230 #033[30m#033[42m<info> #033[49m#033[39m mqtt reconnect
Oct  1 06:26:06 raspberrypi homekit2mqtt[637]: 2018-10-01 06:26:06.239 #033[30m#033[42m<info> #033[49m#033[39m mqtt reconnect

[homebridgeconfig.json.txt](https://github.com/sfeakes/AqualinkD/files/2453156/homebridgeconfig.json.txt)
[homekit2mqtt.defaults.txt](https://github.com/sfeakes/AqualinkD/files/2453157/homekit2mqtt.defaults.txt)
[homekit2mqtt.service.txt](https://github.com/sfeakes/AqualinkD/files/2453158/homekit2mqtt.service.txt)

alanfromtoronto, 2018-10-06:
Thanks in advance

192.168.1.107 is the aqualind
192.168.1.196 is the homebridge

[aqualinkd.conf.txt](https://github.com/sfeakes/AqualinkD/files/2453270/aqualinkd.conf.txt)

sfeakes, 2018-10-06:
Here is at least one problem, you're pointing to two different MQTT servers.

Aqualinkd is configured to
mqtt_address = 192.168.1.196:1883

Homekit's mqtt is
"mqtt": {
                "host": "192.168.1.107",
                "port": 1883,
                "topic": "aqualinkd"
          },

They need to be the same MQTT server.  I would assume you need to change 192.168.1.107 to 192.168.1.196.  But that's a guess, both should be pointing to your MQTT server, (the machine running mosquito).

alanfromtoronto, 2018-10-06:
Yes that fixed it!

sfeakes, 2018-10-06:
Great, glad that worked.

closed 2018-10-06 by sfeakes