aqualinkd/AqualinkD #27: Using AqualinkD without controller

Kranzy, 2018-10-17:
Hi, I have a Zodiac Tri Chlor unit that has the ability to interface with a controller. 

Is it possible to setup and run this software connected to the controller connection of the chlorinator to get updates and control the chlorinator? Such as pump on/off etc?

I realise this isn't an issue with the current code however I wasn't sure the best way to ask the question.

AUPoolUser, 2018-10-18:
Hi Kranzy, This software is designed to work primarily with an Aqualink RS system. There is also an Aqualink Tri unit - but I don't think anyone has tried it to see whether the RS485 communications is identical to the Aqualink RS unit - I doubt it given that the Aqualink PDA and Aqualink RS which have the same hardware are even different. The Aqualink Tri I would assume is different - but someone needs to test it and confirm it.

BUT in regards to the Tri Chlorinator (which I think is mainly sold in Australia, Europe and South Africa), I understand from Zodiac that it does connect to an Aqualink RS unit and communicates with it. Whether it connects to this software again is a try.

Sorry I cant answer your question definitively, but I though a little bit of extra  information may give you some extra information that sways your decision one way or another to try it.

Good Luck - please post your results if you try!

sfeakes, 2018-10-18:
AqualinkD will not work without a control panel.  It’s designed to be an RS485 client, and a control panel is the RS485 server.  It can control salt water generators, but only through the control panel.

Now with all the testing and debugging I’ve done, I have created a RS485 server that mimics the control panel, and that can control a Hayward Goldline, AquaRite chlorinator, and some other devices directly.  But I’ve never relaeased that software / project.  I’ll have to read up on the Zodiac tri system and see if it uses the exact same RS485 protocol (If I can find any documentation).   You will have to introduce power to the RS485 buss, as a USB adapter can’t do that, but that would be the only hardware issue.

Kranzy, 2018-10-18:
I am happy to work with you in the testing of that. 

Ill see if I can find any documentation that might help.

sfeakes, 2018-10-18:
Ok.  Give me some time to write some details up on how to use it, and I’ll post as a different project in alpha.  As long as you get the RS485 link sorted we should be able to tell very quickly if it’s going to work, assuming you’re using a raspberry pi.

Kranzy, 2018-10-18:
No worries, appreciate your help. I will be running it from a raspberry pi.

sfeakes, 2018-10-18:
If this is what you have, then I'm quite sure we can get it working.
https://www.jacksonsleisure.com/content/pdf/salt_chlorinator/ZodiacTRi%20manual.pdf

Page 34 & 35 cover the connection & configuration.
You'll need to configure it to "Polaris EOS/ Jandy Aqualink controllers"
Looks like you will also need to introduce power on the RS485 as well.  (So just connect data +/- to the USB 2 RS485 adapter, (A & B on the Tri), and connect external DC power adapter to (0v and +v) on the Tri.   --  **** I think it's 12v DC, but don't take that for granted ****

There is also a warning about Jandy version L on page 33.  I'm not sure what the change was from a SWG, so may need to dig into that.

sfeakes, 2018-10-18:
https://github.com/sfeakes/AquapureD

Really quick and dirty, way too much code, but the executable to test are there.  Just get the repo and try to run (per instructions) once you get the Pi connected to the Zodiac Tri

Kranzy, 2018-10-19:
> If this is what you have, then I'm quite sure we can get it working.
> https://www.jacksonsleisure.com/content/pdf/salt_chlorinator/ZodiacTRi%20manual.pdf
>
> Page 34 & 35 cover the connection & [...]

Its pretty much the same unit, mine maybe slightly newer? Link to my manual is below. I'm going to start getting a list of the components I need then I'll start testing. 

Any raspberry pi will work?

https://www.zodiac.com.au/Content/cms/products/pdf/Zodiac%20TRi-XO%20Manual.pdf

sfeakes, 2018-10-19:
Any Pi will work.  While the codebase is large, it's optimized for small devices, so even the Pi Model A/B will run this with ease.  Only thing to note is some Pi's like the Zero have no onboard Wifi and only 1 USB port, so you'll need a USB hub to run WiFi & USB2RS485, and the Pi is notoriously bad with many hubs.  Pi Zero W is the one I recommend, as it's the cheapest with onboard WiFi.

USB to RS485 converters, I've personally tested these two, both work flawlessly.
https://www.amazon.com/OctagonStar-Converter-Adapter-Interface-FT232RL/dp/B01LCFRR3E/
https://www.amazon.com/gp/product/B00NKAJGZM/
But I'm sure just about any will work.

Kranzy, 2018-10-25:
Moving further discussions to here:
https://github.com/sfeakes/AquapureD/issues/1

closed 2018-10-25 by Kranzy

benville32, 2019-07-12:
> AqualinkD will not work without a control panel. It’s designed to be an RS485 client, and a control panel is the RS485 server. It can control salt water generators, but only through the control [...]

 I have an Aquarite SWG which has a controller box.  I'm assuming the AqualinkD would have no problem interfacing.  I have seen some threads where physical Aqualink devices don't work correctly with Aquarite and causes them to lock.  Seems to be an interoperability done on purpose by Hayward.
Has anyone seen that?
Anyone else used this with Aquarite?