aqualinkd/AqualinkD #28: Bad receive packet

akalderon, 2018-11-09:
I keep getting the following error.  Any hints on how to troubleshoot what may be wrong?  Thanks

Nov 09 10:26:30 homebridge2 aqualinkd[17375]: Serial read bad checksum, ignoring
Nov 09 10:26:30 homebridge2 aqualinkd[17375]: Bad receive packet  | HEX: 0x10|0x02|0xaf|0x10|0x02|0x0a|0x02|0x00|0x11|0x00|0x00|0x00|0
Nov 09 10:26:41 homebridge2 aqualinkd[17375]: Serial read bad checksum, ignoring
Nov 09 10:26:41 homebridge2 aqualinkd[17375]: Bad receive packet  | HEX: 0x10|0x02|0xaf|0x10|0x02|0x0a|0x02|0x00|0x11|0x00|0x00|0x00|0
Nov 09 10:26:52 homebridge2 aqualinkd[17375]: Serial read bad checksum, ignoring
Nov 09 10:26:52 homebridge2 aqualinkd[17375]: Bad receive packet  | HEX: 0x10|0x02|0xac|0x10|0x02|0x0a|0x02|0x00|0x11|0x00|0x00|0x00|0
Nov 09 10:27:03 homebridge2 aqualinkd[17375]: Serial read bad checksum, ignoring
Nov 09 10:27:03 homebridge2 aqualinkd[17375]: Bad receive packet  | HEX: 0x10|0x02|0xac|0x10|0x02|0x0a|0x02|0x00|0x11|0x00|0x00|0x00|0
Nov 09 10:27:14 homebridge2 aqualinkd[17375]: Serial read bad checksum, ignoring
Nov 09 10:27:14 homebridge2 aqualinkd[17375]: Bad receive packet  | HEX: 0x10|0x02|0xac|0x10|0x02|0x0a|0x02|0x00|0x11|0x00|0x00|0x00|0

sfeakes, 2018-11-10:
Usually this is due to a conflicting address on the RS485 bus.
Does anything work at all, or do you only get these messages?

Also, can you run the serial logger and post the output?
Make sure to stop the aqualinkd service (as that will conflict).
`sudo systemctl stop aqualinkd`
Then run the logger in debug mode to print every packet
`sudo ./release/serial_logger /dev/ttyUSB0 -d`
Then if you can post that output, it may give us a clue as to what's going on.

akalderon, 2018-11-11:
Yes, generally it is working albeit frequently missing a command and then crashing, requiring a restart of the service.  
Here is the output from the serial logger.  Any assistance would be greatly appreciated - thank you!
--
Notice: Logging serial information!

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x51 of type    Probe | HEX: 0x10|0x02|0x51|0x00|0x63|0x10|0x03|

  To 0x80 of type    Probe | HEX: 0x10|0x02|0x80|0x00|0x92|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x52 of type    Probe | HEX: 0x10|0x02|0x52|0x00|0x64|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x53 of type    Probe | HEX: 0x10|0x02|0x53|0x00|0x65|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x58 of type    Probe | HEX: 0x10|0x02|0x58|0x00|0x6a|0x10|0x03|

  To 0x33 of type Unknown  | HEX: 0x10|0x02|0x33|0x30|0x75|0x10|0x03|
From 0x33 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

  To 0xa3 of type Unknown  | HEX: 0x10|0x02|0xa3|0x53|0x08|0x10|0x03|
From 0xa3 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x1f|0x00|0x32|0x10|0x03|

  To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|
[... 279 lines cut ...]
Notice: ID 0x51 is not used 
Notice: ID 0x80 is not used 
Notice: ID 0x60 is not used  <-- can use for Aqualinkd (PDA mode only)
Notice: ID 0x52 is not used 
Notice: ID 0x53 is not used 
Notice: ID 0x58 is not used 
Notice: ID 0x68 is not used 
Notice: ID 0x69 is not used 
Notice: ID 0x6a is not used 
Notice: ID 0x6b is not used 
Notice: ID 0x70 is not used 
Notice: ID 0x71 is not used 
Notice: ID 0x72 is not used 
Notice: ID 0x81 is not used 
Notice: ID 0x73 is not used 
Notice: ID 0x88 is not used 
Notice: ID 0x89 is not used 
Notice: ID 0x08 is not used  <-- can use for Aqualinkd
Notice: ID 0x09 is not used  <-- can use for Aqualinkd
Notice: ID 0x0a is not used  <-- can use for Aqualinkd
Notice: ID 0x0b is not used  <-- can use for Aqualinkd
Notice: ID 0x40 is not used 
Notice: ID 0x41 is not used 
Notice: ID 0x42 is not used 
Notice: ID 0x88 is in use

sfeakes, 2018-11-11:
From that it doesn't look like there is an address conflict, also didn't list any bad checksums.
Can you let it run until it finishes for complete results. (ie don't press ctrl -c).
If the output is too great, post as file or run without -d option

Can you give me a list of the equipment you have on the RS485 bus?  It's kind-a looking like bad wiring or USB adapter at the moment.  Can you let me know what adapter your using and may want to check the wiring for perfect connection.

akalderon, 2018-11-12:
I edited the post and pasted a full output (without ctrl-c).  
On the bus there is an Aquacal Heater/Chiller as well.  Wiring seems fine on both ends from checking it again just now. 

Adapter is: https://www.amazon.com/gp/product/B078X5H8H7/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1

sfeakes, 2018-11-13:
So when you are running the serial logger you are not seeing any bad packets?  That's very strange.
How often / frequently do you see them in the aqualinkd log?
What hardware / OS are you using.

Also can you try running the serial logger for longer (without the debug) just to see it it picks up any bad packets?  `-p 5000` will run for 5k packets.

akalderon, 2018-11-14:
Traveling this week and will do so.ovwr the weekend.  Thank you so much for sticking with me.

akalderon, 2018-11-22:
So I ended up rewiring with a serial cable that has better shielding.  Most of the bad packet errors are no longer there but I do get the following.  Everything seems to work fine despite this log and I don't really see the service crashing yet beyond the initial instance below which seems to be related to aqualinkd not being able to set the RS clock?  

Nov 22 12:28:57 homebridge2 aqualinkd[628]: Control Panel B0029221 REV T.2
Nov 22 12:28:57 homebridge2 aqualinkd[628]: Getting pool & spa heat setpoints from aqualink
Nov 22 12:29:01 homebridge2 aqualinkd[628]: Getting freeze protection setpoints
Nov 22 12:29:41 homebridge2 aqualinkd[628]: RS time is NOT accurate '12:27 PM 11/22/18 Thu', re-setting on contr
Nov 22 12:29:41 homebridge2 aqualinkd[628]: Setting time on aqualink
Nov 22 12:29:49 homebridge2 aqualinkd[628]: Serial read timeout
Nov 22 12:29:49 homebridge2 aqualinkd[628]: Bad receive packet  | HEX: 0x10|0x02|0x0a|0x03|0x00|0x4d|0x69|
Nov 22 12:29:54 homebridge2 aqualinkd[628]: __Aqualink daemon looks like serial error, resetting.__
Nov 22 12:30:01 homebridge2 aqualinkd[628]: Control Panel B0029221 REV T.2
Nov 22 12:30:01 homebridge2 aqualinkd[628]: AQ_Programmer Could not set numeric input 'MINUTE', not found
~

sfeakes, 2018-11-27:
Glad it's all working (except setting time).
Can you try turning on debug mode in the config then starting aqualinkd, wait a few mins, and post the log.  That way I hope I can see what's causing the problem.

akalderon, 2018-11-27:
Log as requested below:
[https://pastebin.com/CCQnHnG1](url)

sfeakes, 2018-11-27:
Great, thanks for that, I actually found a different issue from your logs, and I've fixed that.  But it looks like the time on the Pi and time on Control panel match, so it didn't try to set to the time on the control panel.  AqualinkD will set the time on the Aqualink control panel to match the Pi and that's what was failing. 
Sorry to ask you again, but when you get time, can you change the time on either the Pi or the Control panel so they do not match, and run it with debug again and post the logs.

akalderon, 2018-11-28:
In link below.  I changed the time on the Aqualink side and aqualinkd did successfully set it.

Again - many many thanks
[https://pastebin.com/SmedRri6](url)

sfeakes, 2018-11-28:
Posted an update to fix this.

closed 2018-11-28 by sfeakes