aqualinkd/AqualinkD #29: I'm getting a Stack: e6 message in the web interface

brianmccarty, 2018-11-20:
Do i need to do anything for this messgae?

[image]

sfeakes, 2018-11-20:
That message has come from the control panel, I’m not sure what it means.  Do you get it all the time, or only once you’ve done something?

Just a thought, I do reformat messages to look nicer, so maybe there is a character on the original received message that’s making my code put out some garbage. If you use the simulator interface no reformatting is done, so maybe try the simulator and see if you get that message.

brianmccarty, 2018-11-20:
Thanks for the quick reply.  I'm on an AquaLink 8157 Rev K.   Attached is the scroll of the simulator page.  The Stack: E6 is still there, perhaps it's placement will give you an idea of where I can look further.

here is a video clip.
https://youtu.be/x5OCUuJT3Fs

sfeakes, 2018-11-20:
That’s definitely a message from the control panel to the keypad.  If you have an RS allbutton keypad I’m sure you’d see the same message.  But it’s not a message I’ve ever seen created from the control panel before, what other equipment do you have connected?, I’m wondering if it’s from an accessory.

But nothing to worry about, just like to know why / what’s causing it.

brianmccarty, 2018-11-20:
Yeah me too.  This evening I'll take a look at things.  I'm curious what
the OneTouch is displaying on that line where the Stack message appears as
it cycles through it's messages.

sfeakes, 2018-11-27:
It looks like when you ran the serial logger, you also had aqualinkd running, so there was a lot of conflicts happening on the RS Buss.  So I can't make much from that.

On the second, can you turn on debug logging in the aqualinkd configuration, then re-start aqualinkd, and post the log. That way I may get a better idea or what's happening.

brianmccarty, 2018-11-27:
I'm in the process of wiring the pi up to the Aqualink RS directly with a
couple feet of cable and disconnecting the OneTouch panel (which has a bad
display anyway).  My hope is to eliminate the errors of any possible
cabling faults, and will recheck the address of the devices.   Once I get
this setup I'll capture logs and post them

sfeakes, 2018-11-27:
Just a kind-a shot in the dark. It looks like you might have a pool heater connected to your RS Buss.  Going over LX heater error codes. There is one,  "_**E06** message replaced flue **stack** sensor_" in the Pentair listings. Probably common on many other brand LX heaters as well.  Have you looked for any error codes on your heaters LCD panel recently?

brianmccarty, 2018-11-27:
I do have a heater connected and I have not checked the panel for any error
codes, but I will check that out, thank you.

closed 2019-05-31 by sfeakes