aqualinkd/AqualinkD #30: AqualinkD crashes regularly if I cycle a device on and off a few times

brianmccarty, 2018-11-22:
For example, if I switch the pool light on and off with a 5 second delay between the switch. After a few cycles the AqualinkD service stops.  Tailing the aqualinkd.log shows it freezes.  

Here is a dump from the syslog 

Nov 21 20:38:53 raspberrypi systemd[1]: Starting Aqualink RS daemon...
Nov 21 20:38:53 raspberrypi aqualinkd: Aqualink Daemon v1.2.3
Nov 21 20:38:53 raspberrypi aqualinkd: Config log_level         = 7
Nov 21 20:38:53 raspberrypi aqualinkd: Config socket_port       = 80
Nov 21 20:38:53 raspberrypi aqualinkd: Config serial_port       = /dev/ttyUSB0
Nov 21 20:38:53 raspberrypi aqualinkd: Config web_directory     = /var/www/aqualinkd/
Nov 21 20:38:53 raspberrypi aqualinkd: Config device_id         = 0x0b
Nov 21 20:38:53 raspberrypi aqualinkd: Config override frz prot = NO
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_server       = (null)
Nov 21 20:38:53 raspberrypi systemd[1]: Started Aqualink RS daemon.
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_dz_sub_topic = (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_dz_pub_topic = (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_aq_topic     = (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_user         = (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_passwd       = (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Config mqtt_ID           = aqualinkd_b827ebd8f3
Nov 21 20:38:53 raspberrypi aqualinkd: Config idx water temp    = 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config idx pool temp     = 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config idx spa temp      = 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config idx SWG Percent   = 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config idx SWG PPM       = 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config PDA Mode          = NO
Nov 21 20:38:53 raspberrypi aqualinkd: Config deamonize         = YES
Nov 21 20:38:53 raspberrypi aqualinkd: Config log_file          = /var/log/aqualinkd.log
Nov 21 20:38:53 raspberrypi aqualinkd: Config light_pgm_mode    = 0.00
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Filter_Pump   = label Filter Pump     | PDAlabel FILTER PUMP
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Spa_Mode      = label Spa Mode        | PDAlabel SPA
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_1         = label Spa Jets        | PDAlabel AUX1
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_2         = label Waterfall       | PDAlabel AUX2
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_3         = label Spa Light       | PDAlabel AUX3
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_4         = label Pool Light      | PDAlabel AUX4
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_5         = label NONE            | PDAlabel AUX5
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_6         = label NONE            | PDAlabel AUX6
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Aux_7         = label NONE            | PDAlabel AUX7
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Pool_Heater   = label Pool Heater     | PDAlabel POOL HEAT
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Spa_Heater    = label Spa Heater      | PDAlabel SPA HEAT
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Config BTN Solar_Heater  = label Solar Heater    | PDAlabel EXTRA AUX
   | dzidx 0
Nov 21 20:38:53 raspberrypi aqualinkd: Starting web server on port 80
Nov 21 20:38:53 raspberrypi aqualinkd: Starting MQTT client to (null)
Nov 21 20:38:53 raspberrypi aqualinkd: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Nov 21 20:38:54 raspberrypi aqualinkd: Control Panel 8157 REV K
Nov 21 20:38:54 raspberrypi aqualinkd: Getting pool & spa heat setpoints from aqualink
Nov 21 20:39:00 raspberrypi aqualinkd: Getting freeze protection setpoints
Nov 21 20:39:40 raspberrypi aqualinkd: Serial read bad checksum, ignoring
Nov 21 20:39:40 raspberrypi aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0xf8|0x09|0x00|0x00|0x0a|0x10|0x03|
Nov 21 20:39:53 raspberrypi aqualinkd: Serial read bad checksum, ignoring
Nov 21 20:39:53 raspberrypi aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0xf8|0x09|0x00|0x00|0x0a|0x10|0x03|
Nov 21 20:39:55 raspberrypi aqualinkd: Serial read bad checksum, ignoring
Nov 21 20:39:55 raspberrypi aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0xf8|0x09|0x00|0x01|0x0b|0x10|0x03|
Nov 21 20:39:59 raspberrypi aqualinkd: Serial read bad checksum, ignoring
Nov 21 20:39:59 raspberrypi aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0xf8|0x09|0x00|0x00|0x0a|0x10|0x03|
Nov 21 20:40:49 raspberrypi systemd[1]: aqualinkd.service: Main process exited, code=killed, status=11/SEGV
Nov 21 20:40:49 raspberrypi systemd[1]: aqualinkd.service: Unit entered failed state.
Nov 21 20:40:49 raspberrypi systemd[1]: aqualinkd.service: Failed with result 'signal'.

sfeakes, 2018-11-25:
It's the bad packets that are causing the crash.  I would suggest checking all the connections / wire on the RS485 bus.
The below message are the ones you need to get rid of.  They are usually from either two devices using the same address or from bad wiring.
```
aqualinkd: Serial read bad checksum, ignoring
aqualinkd: Bad receive packet | HEX: 
```

sfeakes, 2018-11-30:
Closing this out as it looks like it's fixed from your logs in the other thread.  Please let me know if it's not fixed.

closed 2018-11-30 by sfeakes