aqualinkd/AqualinkD #31: AqualinkD & Homekit

ericklil, 2018-11-24:
Hello,

What great works you have done!
Running HapNodeJS on a Raspi 3, could one run AqualinkD on the same unit or is a separate Raspi suggested?

sfeakes, 2018-11-25:
You can run it all on a single Pi3 easily.

ericklil, 2018-11-25:
Many Thanks, look forward to it!

ericklil, 2018-11-26:
Is there a way to integrate `HapNodeJS` and `AqualinkD` directly instead of using HomeBridge?

sfeakes, 2018-11-26:
Yes.  Use the Homekit2MQTT configuration in the Wiki.
Homekit2MQTT is a plugin for HapNodeJS that allows for direct MQTT integration. It requires a little more configuration but it's a direct integration without the need for homebridge.

https://github.com/sfeakes/AqualinkD/wiki#HomeKit

closed 2018-11-27 by sfeakes