aqualinkd/AqualinkD #32: Homekit2mttq sSerial error

ericklil, 2018-12-22:
Hello,

Could you have look ate the below and see where I've made an error?

I first Set up `Homekit2mqtt` & tested, it fine works with several accessories.
then: Set up AqualinkD as per instructions, tested and confirmed it worked on physical Aqualink.
Created FiterPump Service `aqualinkd/Filter_Pump/set` in `example-homekit2mqtt.json` 
Ran : `sudo aqualinkd -d -c /etc/aqualinkd.conf`

MQTT messages reach the console:
`Debug:  MQTT: topic aqualinkd/Filter_Pump/set 0
Info:   MQTT: MATCH Filter_Pump to topic aqualinkd/Filter_Pump/set
Info:   MQTT: received 'OFF' for 'Filter_Pump', turning 'OFF'
Info:   Queue send '0x02' to controller`
then
` Error :Aqualink daemon looks like serial error, resetting`

Also `sudo systemctl start aqualinkd` does not launch  AqualinkD

Appreciate any feedback...

sfeakes, 2018-12-24:
Can you post the full logs.  It looks like AqualinkD isn't communicating correctly to the RS485 adapter.

If you can post the logs from aqualinkd and also the serial logger that would be helpful, information in in the wiki.  Just make sure you don't run both at the same time.

ericklil, 2018-12-26:
Hello,

Hope that this helps…

Running `Homekit2MQTT` on same server ( RaspberryPi )
Installed `Homekit2MQTT` first, then added `Aqualinkd`

Service does not want to load after initial trouble free start up

Serial Logger does not show any activity leavers a spinning` /` after `"Please wait"`

`pi@Homebridge:~ $ sudo systemctl stop AqualinkD
Failed to stop AqualinkD.service: Unit AqualinkD.service not loaded.
pi@Homebridge:~ $ cd ~/software/AqualinkD
pi@Homebridge:~/software/AqualinkD $ 
pi@Homebridge:~/software/AqualinkD $ udo ./release/serial_logger /dev/ttyUSB0
-bash: udo: command not found
pi@Homebridge:~/software/AqualinkD $ sudo ./release/serial_logger /dev/ttyUSB0
Notice: Logging serial information!
Please wait.z^Z
[1]+  Stopped                 sudo ./release/serial_logger /dev/ttyUSB0
pi@Homebridge:~/software/AqualinkD $ make slog
make: Nothing to be done for 'slog'.
pi@Homebridge:~/software/AqualinkD $ sudo ./release/serial_logger /dev/ttyUSB0
Notice: Logging serial information!
Please wait.`

`pi@Homebridge:~ $ sudo aqualinkd -d -c /etc/aqualinkd.conf
Notice: Aqualink Daemon v1.2.5
Notice: Config log_level         = 7
Notice: Config socket_port       = 80
Notice: Config serial_port       = /dev/ttyUSB0
Notice: Config web_directory     = /var/www/aqualinkd/
Notice: Config device_id         = 0x0a
Notice: Config override frz prot = NO
Notice: Config mqtt_server       = localhost:1883
Notice: Config mqtt_dz_sub_topic = (null)
Notice: Config mqtt_dz_pub_topic = (null)
Notice: Config mqtt_aq_topic     = aqualinkd
Notice: Config mqtt_user         = someusername
Notice: Config mqtt_passwd       = somepassword
Notice: Config mqtt_ID           = aqualinkd_b827eb0248
Notice: Config idx water temp    = -1024
Notice: Config idx pool temp     = -999
Notice: Config idx spa temp      = -999
Notice: Config idx SWG Percent   = 0
Notice: Config idx SWG PPM       = 0
Notice: Config PDA Mode          = NO
Notice: Config deamonize         = NO
Notice: Config log_file          = (null)
Notice: Config light_pgm_mode    = 0.00
Notice: Config BTN Filter_Pump   = label Filter Pump     | PDAlabel FILTER PUMP     | dzidx 0
Notice: Config BTN Spa_Mode      = label Not Known       | PDAlabel SPA             | dzidx 0
Notice: Config BTN Aux_1         = label Vanishing Edge  | PDAlabel AUX1            | dzidx 0
Notice: Config BTN Aux_2         = label Waterfall       | PDAlabel AUX2            | dzidx 0
Notice: Config BTN Aux_3         = label Spa Blower      | PDAlabel AUX3            | dzidx 0
Notice: Config BTN Aux_4         = label Pool Light      | PDAlabel AUX4            | dzidx 0
Notice: Config BTN Aux_5         = label Spa Light       | PDAlabel AUX5            | dzidx 0
Notice: Config BTN Aux_6         = label NONE            | PDAlabel AUX6            | dzidx 0
Notice: Config BTN Aux_7         = label NONE            | PDAlabel AUX7            | dzidx 0
Notice: Config BTN Pool_Heater   = label Pool Heater     | PDAlabel POOL HEAT       | dzidx 0
Notice: Config BTN Spa_Heater    = label Spa Heater      | PDAlabel SPA HEAT        | dzidx 0
Notice: Config BTN Solar_Heater  = label Solar Heater    | PDAlabel EXTRA AUX       | dzidx 0
Notice: Starting web server on port 80
Error:  Failed to create listener
Error:  Can not start webserver on port 80.`

ericklil, 2018-12-26:
and this..

● aqualinkd.service - Aqualink RS daemon
   Loaded: loaded (/etc/systemd/system/aqualinkd.service; enabled; vendor preset: enabled)
   Active: failed (Result: resources) since Wed 2018-12-26 11:26:38 PST; 33s ago
  Process: 20452 ExecStart=/usr/local/bin/aqualinkd $OPTS (code=exited, status=0/SUCCESS)
 Main PID: 20322 (code=exited, status=1/FAILURE)

Dec 26 11:26:38 Homebridge systemd[1]: Starting Aqualink RS daemon...
Dec 26 11:26:38 Homebridge aqualinkd[20452]: Aqualink Daemon v1.2.5
Dec 26 11:26:38 Homebridge aqualinkd[20452]: Config log_level         = 7
Dec 26 11:26:38 Homebridge systemd[1]: aqualinkd.service: PID 20453 read from file /run/aqualinkd.pid does not exist or
Dec 26 11:26:38 Homebridge systemd[1]: Failed to start Aqualink RS daemon.
Dec 26 11:26:38 Homebridge systemd[1]: aqualinkd.service: Unit entered failed state.
Dec 26 11:26:38 Homebridge systemd[1]: aqualinkd.service: Failed with result 'resources'.
~
~
~
~
~
~
~
~
~

ericklil, 2018-12-31:
A USB device plugged in to Raspi changed the serial port... Resolved!

closed 2018-12-31 by ericklil