aqualinkd/AqualinkD #33: Aqualinkd not responding to MQTT commands

tcm0116, 2018-12-26:
I've got aqualinkd and mosquitto running on a RaspberryPi and I'm working on a SmartThings integration. However, it looks like aqualinkd is not subscribing to the "set" MQTT topics.

When I start aqualinkd, i see a message in the log indicating "Starting MQTT client to localhost:1883". However, based on net_services.c, I would expect to then see a "MQTT: Connection acknowledged" message, which appears to occur right before it subscribes to the topics.  However, I never see the connection acknowledged message. aqualinkd does publish status to mosquitto, so it looks like the connection is good.

Any thoughts on why this may be occurring?

tcm0116, 2018-12-26:
Disregard. My SmartThings device handler wasn't properly sending the command to my proxy. It appears to be working as expected.

closed 2018-12-26 by tcm0116

hollapm, 2019-08-12:
@tcm0116 Were you able to integrate with Smartthings, and if so could you share your device handler?

Thanks!

napertivo, 2019-08-19:
I'd be interested in your device handler as well if you would like to share it.

Thanks

tcm0116, 2019-08-19:
I got the device handlers uploaded to GitHub. You should be able to add the repo to SmartThings and pull the device handlers down. Let me know if you have any issues. 

https://github.com/tcm0116/smartthings-aqualink

You'll also need to install the smartthings-mqtt-proxy package via npm on a device on your network. This allows the device handlers and AqualinkD to interact.

https://www.npmjs.com/package/smartthings-mqtt-proxy

Here's the config file I use for the proxy:

```
---
mqtt:
    # Specify your MQTT Broker's hostname or IP address here
    host: localhost
    # Preface for the topics $PREFACE/$DEVICE_NAME/$PROPERTY
    preface: smartthings

    # Other optional settings from https://www.npmjs.com/package/mqtt#mqttclientstreambuilder-options
    # username: AzureDiamond
    # password: hunter2

    # Don't retain the values, as this can cause AqualinkD to receive old commands when it restarts
    retain: false

    # Port number to listen on
    port: 8080
```

Ideally, the system that the MQTT proxy is installed on has a static IP address because you need to provide the IP address, the port used by the proxy (8080 is the default), and the MAC address (used as a unique identifier) to the pool controller device handler. 

One thing to keep in mind is that the pool controller handler is currently designed to work with my specific pool configuration, which is a pool/spa combo with a heater. If yours is different, we can work on adding some configuration options. I'd like to add the ability to control the pool light and the chlorine generator at some point.