aqualinkd/AqualinkD #34: Error:  Aqualink daemon looks like serial error, resetting.

bryangauvin, 2018-12-30:
Any idea whats wrong here?

 sudo /usr/local/bin/aqualinkd -d -c /etc/aqualinkd.conf
Notice: Aqualink Daemon v1.2.5
Notice: Config log_level         = 7
Notice: Config socket_port       = 88
Notice: Config serial_port       = /dev/ttyUSB0
Notice: Config web_directory     = /var/www/aqualinkd/
Notice: Config device_id         = 0x09
Notice: Config override frz prot = NO
Notice: Config mqtt_server       = localhost:1883
Notice: Config mqtt_dz_sub_topic = (null)
Notice: Config mqtt_dz_pub_topic = (null)
Notice: Config mqtt_aq_topic     = aqualinkd
Notice: Config mqtt_user         = 406XXXXXX
Notice: Config mqtt_passwd       = 406XXXXXXX
Notice: Config mqtt_ID           = aqualinkd_b827eb29b2
Notice: Config idx water temp    = 0
Notice: Config idx pool temp     = 0
Notice: Config idx spa temp      = 0
Notice: Config idx SWG Percent   = 0
Notice: Config idx SWG PPM       = 0
Notice: Config PDA Mode          = NO
Notice: Config deamonize         = NO
Notice: Config log_file          = (null)
Notice: Config light_pgm_mode    = 0.00
Notice: Config BTN Filter_Pump   = label Filter Pump     | PDAlabel FILTER PUMP     | dzidx 0
Notice: Config BTN Spa_Mode      = label Spa Mode        | PDAlabel SPA             | dzidx 0
Notice: Config BTN Aux_1         = label Cleaner         | PDAlabel AUX1            | dzidx 0
Notice: Config BTN Aux_2         = label Aux_2           | PDAlabel AUX2            | dzidx 0
Notice: Config BTN Aux_3         = label Aux_3           | PDAlabel AUX3            | dzidx 0
Notice: Config BTN Aux_4         = label Pool Light      | PDAlabel AUX4            | dzidx 0
Notice: Config BTN Aux_5         = label Aux_5           | PDAlabel AUX5            | dzidx 0
Notice: Config BTN Aux_6         = label Aux_6           | PDAlabel AUX6            | dzidx 0
Notice: Config BTN Aux_7         = label Aux_7           | PDAlabel AUX7            | dzidx 0
Notice: Config BTN Pool_Heater   = label Pool Heater     | PDAlabel POOL HEAT       | dzidx 0
Notice: Config BTN Spa_Heater    = label Spa Heater      | PDAlabel SPA HEAT        | dzidx 0
Notice: Config BTN Solar_Heater  = label Solar_Heater    | PDAlabel EXTRA AUX       | dzidx 0
Notice: Starting web server on port 88
Notice: Starting MQTT client to localhost:1883
Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827eb29b2
Info:   MQTT: Connection acknowledged
Info:   MQTT: Subscribing to 'aqualinkd/#'
Info:   MQTT: Subscription(s) acknowledged
Info:   MQTT: Published id=1: aqualinkd/Alive 1
Debug:  MQTT: Ignore aqualinkd/Alive 1
Debug:  MQTT: Ignore aqualinkd/SWG 0
Debug:  MQTT: Ignore aqualinkd/SWG/enabled 0
Debug:  MQTT: Ignore aqualinkd/Filter_Pump 0
Debug:  MQTT: Ignore aqualinkd/Spa_Mode 0
Debug:  MQTT: Ignore aqualinkd/Aux_1 0
Debug:  MQTT: Ignore aqualinkd/Aux_2 0
Debug:  MQTT: Ignore aqualinkd/Aux_3 0
Debug:  MQTT: Ignore aqualinkd/Aux_4 0
Debug:  MQTT: Ignore aqualinkd/Aux_5 0
Debug:  MQTT: Ignore aqualinkd/Aux_6 0
Debug:  MQTT: Ignore aqualinkd/Aux_7 0
Debug:  MQTT: Ignore aqualinkd/Pool_Heater 0
Debug:  MQTT: Ignore aqualinkd/Pool_Heater/enabled 0
Debug:  MQTT: Ignore aqualinkd/Spa_Heater 0
Debug:  MQTT: Ignore aqualinkd/Spa_Heater/enabled 0
Debug:  MQTT: Ignore aqualinkd/Solar_Heater 0
Info:   MQTT: Message publishing acknowledged (msg_id: 1)
Info:   MQTT: received (msg_id: 1), looks like my own message, ignoring
Debug:  MQTT: Ignore aqualinkd/Alive 1
Debug:  ++ Websocket joined
Info:   WS: Message - Key 'command' Value 'GET_DEVICES' | Key2 '(null)' Value2 '(null)'
Debug:  WEB: homebridge used 1545 of 3000
Error:  Aqualink daemon looks like serial error, resetting.
Error:  Aqualink daemon looks like serial error, resetting.
Error:  Aqualink daemon looks like serial error, resetting.
Error:  Aqualink daemon looks like serial error, resetting.
Error:  Aqualink daemon looks like serial error, resetting.

sfeakes, 2018-12-31:
Aqualink is not getting anything from the RS458 adapter.
Take a look at the wiki, specificity the section about testing the connection.  Section “USB Device on RaspberryPi / Linux computer” should help.

bryangauvin, 2018-12-31:
I have looked at that section. 

Here is what I get. When I run this command, nothing returns if I am in Auto mode on the pump. However if in service mode it spews out the random characters. Any idea?

sudo -s eval 'stty raw -echo < /dev/ttyUSB0; cat -vte /dev/ttyUSB0'

sfeakes, 2018-12-31:
Do you have a Jandy RS control panel connected to the RS485 bus, or is it just a pump?  AqualinkD communicates with a control panel, so you should see junk using that command all the time (as long as the RS control panel is powered up). A box like this.

[image]

bryangauvin, 2018-12-31:
Yes, I have that exact setup. My iAqualink although showing online was not controlling anything which is why I started this project. 

From what you're saying I'm now thinking the controller may have gone bad. Do you concur?

bryangauvin, 2018-12-31:
OK, so I just turned on the filter pump & heater manually in service mode & the software is reporting them on. So its getting readings, just not on auto mode

bryangauvin, 2019-01-06:
Ordered an new PCB, replaced it. Same issue. Aualinkd shows the status of devices when running in service mode (IE: reports pump running, spa mode, ect...) However in auto mode nothing. Any idea what could be causing this @zsblevins @sfeakes

sfeakes, 2019-01-06:
I'm at a complete loss on this one.  I've never seen anything quite like it.  If I understand this, in auto mode you get nothing on the RS485 bus and in service mode, you get the correct information. (of course in service mode you can't control anything, only read information, so you can
t keep it in service mode)
When you changed the board, did you also change the chip, or did you keep the original chip?  You can flash firmware to that chip, I wonder if for some reason it's corrupt in some way, but that would only explain it if you kept the same chip while swapping the boards.

bryangauvin, 2019-01-06:
I have tried both a new chip and the existing chip with the same results.

bryangauvin, 2019-01-06:
I've asked the question on the trouble-free pool forum as well since this is clearly not your software, but something hardware related. 

I also have a scheduled appointment with a repair company on Tuesday. Hate to pay for that crap, but I'm at a loss here. I'll report back if either figures it out so you know in case someone else runs into the issue.

sfeakes, 2019-01-06:
Do you have a keypad (or something that's powered from the RS485 bus) connected?  iAqualink has it's own power I believe. I'm wondering if you loose power and data to the RS485 bus or if it's just the data that goes when you're in normal (not service) mode.

Other thing to try is "time out" mode, that should put it in "service mode" for 3 hours, then revert back to "Automatic mode", so see what happens in 3.5 hours time after you set "time out" mode.

bryangauvin, 2019-01-07:
I have no keypad. iaqualink gets its power from the PCB.  Trying the
timeout mode today

bryangauvin, 2019-01-08:
Ok, so Pool guy just left. No resolution yet, however, the issue appears to be with the drive on the pump creating interference in the signal. Hes going back to the shop to ask the other techs if they have ever seen this or how to fix. Nothing to do with your software. In fact, If I unplug the 4 pin to the pump your software works fine. UGH, FML.

closed 2019-01-08 by bryangauvin

sfeakes, 2019-01-08:
Glad you have at lease tracked it down.  I bet there is a short in the RS bus circuit of the pump.  Look up the pump parts list and see if you can buy / replace the control circuit.  Or just bypass that circuit all together and run it from the control panel power relays.