aqualinkd/AqualinkD #37: Programming / System Setup

ericklil, 2019-01-05:
Happy New Year,
The "older" ` ipaddress/simulator.html` makes it possible to access the `System SetUp` however the connection is lost when scrolling either to fast or selecting some of the settings. Are there limitations to this access? I'd keen to be able to access the set up functions as not to have to rely on `OneTouch` or the `iAqualink` to set up and or change settings. Have you looked into this? Could I help if there is interest to maybe a`OneTouch` styled interface?

sfeakes, 2019-01-05:
Limitations are purely due to me trying to figure out what the hell the Aqualink RS protocol is...:). If you can list what settings you are trying to access (ie each keypress you use to get them) that would be very helpful.  I wrote the simulator really quickly so I'm sure I've missed some of the protocol commands between the keypad and controller as the protocol is very different when you need the controller to wait for user input.

ericklil, 2019-01-07:
I hear you, the protocol is daunting at time, you have done very great works thus far though.
Hoping not to have to buy a OneTouch or iAqualink.
I have accès to both of these units through work but they are so much slower than AqualinkD, that it would be great to add the full functionality of either unit.
Have seen that this person <https://github.com/earlephilhower/aquaweb> was able to mimic some OneTouch functionality but I am unfamiliar with any .py 
I could try to log all traffic from 0x33 & 0xa3 which is the iAqualink. can you give me some hints as to get the “cleanest” readouts?

sfeakes, 2019-01-08:
Onetouch and PDA are total crap when you look at the protocol, i wouldn’t waste any time or money on either. I’d have quit the job rather than implement that shit if I was working for Jandy. iAqualink seems to have corrected some of the protocol limitations, but I don’t have access to either the appropriate control panel of the devise to sniff what’s going on.  I’d like to see what’s going on with the iAqualink stuff, but it will probably be really difficult to implement anything as I don’t have anything to test against.  Getting variable speed pump seems to be what most people want, and I can’t do that with the allbutton.  Give me some time to change the serial logger code and I’ll Write up some instructions on using it to log iAqualink.  I’ll also take another look at my simulator code.  I did look at aqua web a long time ago, I’ll take another look and see how they implement the simulator.

ericklil, 2019-01-08:
Yeah we are dealing with dinosaur type of systems, One can grow a beard while waiting for the PDA to execute, that is if you haven’t fallen a sleep while waiting for the start up..

As for pump speed control, I have successfully made a ESP 8266 based device where I send pumps speed control `RS 485` adaptor for the `Intelliflo VF` and `VS` and others then hooked it up via `MQTT` to Homekit. But that is directly to the pump and not via the Aqualink.

I have an iAqualink at my disposal that I can sniff or perhaps send to you to try out, if you care. I know that iAqualink runs via iobridge <http://www.iobridge.com/> however I’d be fine with a local implementation as I could use Homekit to control (but not edit) my equipment.

tcm0116, 2019-01-24:
I have an iAqualink and am willing to help with the effort to get AqualinkD working using that interface.

sfeakes, 2019-01-24:
I made some changes to the serial logger to help with this.

Please remember any time you run the serial logger, you need to STOP aqualinkd.  `sudo systemctl stop aqualinkd` you can start it again by replacing stop with start in that command.

First thing we need to do is figure out what's on your RS485 buss and the ID's.
Make sure your pump is on and run.
`sudo ./release/serial_logger /dev/ttyUSB0`
Wait for it to finish.
We are interested in any ID's that state "is in use".  I hope to see 0x33 & 0xa3 which is iAqualink, but we also need to see the traffic from the pump(s). I'm not sure on the pump ID's, so that's what we need to work out. Hopefully @ericklil can tell us what ID's are used. Searching on line It looks like Pantair use ID's starting from 0x60, but that's a PDA device for Jandy, so we need to figure that out the ID first. If you can cross reference any ID's that are used against my notes on the Jandy protocol linked below, that should give you the answer. https://github.com/sfeakes/AqualinkD/wiki/Jandy-Aqualink-RS485-protocol
So if you see ID 0x50 you know it's a Aquarite Salt Water Generator, or 0x88 would be your heater.

Once you have figured out the ID of the pump(s) we can start logging what happens. So run the serial_logger in debug mode against the ID's and run it a few times with known device changes, making notes of what you've done & see.  The smaller the runs the better at the moment.  For example.

Turn pump off, start serial logger, turn pump on, stop serial logger, save output. The command for serial logger would be
`sudo ./release/serial_logger /dev/ttyUSB0 -d -i 0x33 -i 0xA3 -i 0x<pumpID>`  <--- replace pumpID
Then just hit ctrl-c to finish logging.

If you could start with these sequences.
#1 Turn pump off, start serial logger, turn pump on, stop serial logger.
#2 Turn pump on, start serial logger, turn pump off, stop serial logger.
#3 Turn pump on, start serial logger, change pump RPM, stop serial logger. (make note of RPM)
#4 Turn pump on, start serial logger, change pump RPM (different RPM to previous), stop serial logger. (make note of RPM).
#5 Turn pump off, start serial logger, turn pump on, change RPM, turn pump off, stop serial logger.

If you can post all that information, I'll see if I can figure out what's going on.

 @ericklil can you send me a copy, or point me to where I can download, the source code of the ESP 8266 software you mention above.  And if you have any documentation on the protocol, that would be very helpful. Also do you know if the pentair pumps work with the Jandy equipment?

With my (at the moment) limited knowledge of how iAqualink and VSP works, I'm thinking I will implement this by using the iAqualink protocol to set pump RPM's, but read the pump's status directly from the pump rather than iAqualink. That's a very similar implementation to how I do Salt Water Generator implementation, as there is so much more info you can get directly from the device rather than the control panel. but you still use the control panel to control the device so everything still works if you remove decide to take AqualinkD out of the chain.

ericklil, 2019-01-25:
Thank you for your input on this, would be great to slay this beast...
I have been able to address the Pentair VS or VF pumps using the following commands:

  `0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x04,0x01,0x00,0x01,0x1A`  //Release pump's Display, enables local control
  `0xFF,0x00,0xff,0xA5,0x00,0x60,0x10,0x04,0x01,0xff,0x02,0x19` // Attention (Disable pumps display, enables remote control)
`0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x01,0x04,0x03,0x21,0x00,0x08,0x01,0x46` // Run speed one
`0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x01,0x04,0x03,0x21,0x00,0x10,0x01,0x4E` // Run Speed Two
`0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x01,0x04,0x03,0x21,0x00,0x18,0x01,0x56` // Run Speed Three
`0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x01,0x04,0x03,0x21,0x00,0x20,0x01,0x5E`// Run Speed Four
`0xFF,0x00,0xFF,0xA5,0x00,0x60,0x10,0x01,0x04,0x03,0x21,0x00,0x00,0x01,0x3E` // Stops PUMP

@sfeakes I wrote a little Arduino [sketch](https://www.dropbox.com/sh/88lhbk9nf87dt09/AACvsfCnHvz77DFXsyrhlTsra?dl=0), please have a look at that. This an older protocol- that still works- but can be made better as the VS pumps will accept 10% steps for speed increase decrease. I will need to record that on `0x33 & 0xa3` ( the iAqualink) I will also see to hook up multiple `VS Pumps` and see how the addressing is handled on the pump side. The pump installer has to change the pump address from a default 1 to 2.3 or 4, which is the maximum number of `VS Pumps` that Aqualink can handle

The `ESP 8266 /MQTT/Homekit` [sketch](https://www.dropbox.com/sh/li5qdvkruj2lagj/AABkwyYVrfX1WHUPxJiCwlsba?dl=0) 
& a 'RS485 Serial Module' [Amazon Link](https://www.dropbox.com/sh/li5qdvkruj2lagj/AABkwyYVrfX1WHUPxJiCwlsba?dl=0)
 using the`fan service`. Again in this situation the connected Serial module is hooked up directly to the Pump ( yellow as Green Wires and not using the Aqualink.

I am out of town until early feb, will certainly try the serial logging as you suggest, then..

ericklil, 2019-01-25:
> Interesting, the Pump ID's are 0x60 as I saw in some Pantair documentation. I'm assuming those pumps were never on the same bus as an RS control panel? as 0x60 is the PDA remote to an RS control panel Just want to check I'm not missing something in the protocol.
> Did you ever interpret the response message / status message from the pumps that came back after you sent the above? Took a look at the sketches, seems like you only sent information and ignored the reply, is that correct or did I miss something?

Yes I did ignore the reply, I can sniff it, when I get back though...
They were indeed not on the same bus as the RS controller.

sfeakes, 2019-01-25:
Great to know they were not on the same bus.  I think I have all the pieces to the puzzle, just need to get some of the log information from a few posts above, and figure out if the RS control panel "set's and forgets" the pumps, or keeps them in "control mode all the time".  I'm fairly sure it's the latter, but hopefully the logs will tell me.

ericklil, 2019-01-28:
In order for the Pentair pumps to stay “tuned” to the controller they need to receive a “heartbeat” signal at least once every 20 seconds. If not they will return to their local control mode. Obviously the RS controller is providing such heartbeat. The RS sends a status request, which I had decoded, but not used, in the ESP 8266 sketch. 

Kind Regards
Cordialement,

Erick Oeseburg

mdgnys, 2019-01-31:
Hey guys, I appreciate you working on this. Sfeakes, you've done incredible work. I wish I could also sniff my iaqualink, as my motor seems to use a different kind if speed control (I could enter the rpm). Unfortunately, my interest in this project was sparked by my iaqualink device getting water damage during our recent freak storm. If I am able to get it working again (I'm not super hopeful), I'll definitely sniff away at it.  
Thanks for a great project!