aqualinkd/AqualinkD #40: co-exist with physical PDA

ballle98, 2019-01-15:
I forked the project and started to work on this myself here https://github.com/ballle98/AqualinkD/issues/4 but I thought it would be good to get some advice.

As it stands I can't leave AqualinkD running because it never gives up control so that another PDA can take control.  On a real PDA if you don't press any buttons for 30-60 seconds it will go to sleep and stop responding to status messages.  The the controller starts sending probes again.  Even if AqualinkD is using a seperate ID (like 62) and the PDA is using 60, the PDA will not work because no probes are sent for 60-63.

I'm thinking that if there have been no program actions for 60 seconds and there are no threads active it would be safe to stop acking until the force status update timeout.

sfeakes, 2019-01-15:
I'll have to think a little about how to implement this. I don't have a PDA to test with so I had no idea it gives up control.  There is some code that does something similar when you put it in simulator mode, it might be best to piggy back off that.  Can make this work with the built in UI, But you'll loose all the state posting to other systems like Homekit and other automation hubs. Maybe put it on a timer, so every hour it probes for changes. Then you'll have to deal with it being very slow to act on the first request if it wakes up from a UI / Automation hub input.  It probably needs to be a configuration option.

ballle98, 2019-01-31:
Uploaded a logfile here showing the interaction between a standard PDA 0x60 and aqualinkd 0x62.  I started the PDA first and aqualinkd is shut out until the PDA goes to sleep.  Then I let aqualinkd run and the PDA is shutout.  I changed the code to print all serial messages and timestamp then HH:MM:SS.MS

https://gist.github.com/ballle98/ecd22c1ef49415f804bd27dc1443c819

ballle98, 2019-01-31:
created a branch and pull request for a proposal https://github.com/sfeakes/AqualinkD/pull/45

sfeakes, 2019-05-31:
Pulled your branch, changed it some, but kept most of functionality.