aqualinkd/AqualinkD #41: Not turning on heater

bryangauvin, 2019-01-17:
Hello,
    Finally got this up and running. Everything is working except turning the heater on. When I use the interface it shows it turns on, On the jandy rs it lights up the light but the heater doesnt turn on.

    When I look the iAqualink OneTouch settings it shows a Heat Pump On as one of the values. Im not seeing a Heat Pump in the AqualinkD interface.

    What am I missing here?

sfeakes, 2019-01-17:
If the RS keypad heater lights up, then that's the correct functionality.  AqualinkD uses the RS naming, so I think what you might be looking for is "solar heater" on the RS Keypad.  To confirm, turn on the heater on the OneTouch, and look at what's lit on the RS keypad.  If it's the solar heater, then just uncomment solar heater in the web configuration file.
On a default installation that's the file in
`/var/www/aqualinkd/config.js`

In the web UI section of the wiki (although not well described)
https://github.com/sfeakes/AqualinkD/wiki#AqualinkDweb

bryangauvin, 2019-01-17:
When I turn it on via the iaqualink the heater light comes on and solar heater stays off

bryangauvin, 2019-01-17:
[image]

bryangauvin, 2019-01-17:
Thats what it shows in iaqualink when I turn on the spa mode

bryangauvin, 2019-01-17:
this is what AqualinkD shows while the heater is running 

[image]

sfeakes, 2019-01-18:
Oh.  One other thing I see looking at the images, and not knowing your exact pool equipment setup. Is maybe the heat pump is configured to only come on while heating and not when "up to temp", in which case the Orange background vs Green background on the temperature icon within Spa heat / Pool Heat gives you that indication.
Example, I have a gas heater and no heat pump. Green background means the heater is ENABLED but not actually heating. Orange means ENABLED & HEATING.  Ie orange means on and heating, green means on but not heating as water it up to temperature.   
Without knowing your pool equipment setup and how it's all configured it's extremely hard to tell if it's acting correctly.   If you have an all button keypad then AqualinkD should mimic the LED's on that exactly.

bryangauvin, 2019-03-04:
Any idea how to figure this out and adjust so it works?

bryangauvin, 2019-03-31:
Hello @sfeakes ,
   Im just getting back to working on this. Here is what I am seeing in Jandys website for what turns on when I turn on the spa. Aqualinkd is not showing the heat pump as a option. Please advise what I have to adjust to see that icon.

[image]

mdetweiler, 2020-08-16:
I'm having a similar issue I think.  I got everything working as far as I can tell, but the button config is confusing since the first button on my panel box controller seems to be a service mode selector, but AqulinkD uses button #1 as Filter pump.  In any case, when I turn on spa mode at my wired Jandy remote panel, the spa heater comes on.  When I turn on spa mode in AqualinkD, it says spa mode is on but no heater.

sfeakes, 2020-08-16:
If I understand your problem correctly.  When you hit "spa mode" on jandy keypad, it sets pump(s) / JVA's to Spa setting AND also turns on spa heater.   But when you hit "spa" button in AqualinkD it sets pump(s) / JVA's to Spa setting but does not turn on spa heater?

If so, then that's correct.  Spa mode is actually different between different Jandy control panels.  OneTouch & IAqualinkTouch both have both Spa and Spa Mode options, but Allbutton keypad only has Spa.   You should just be able to hit the spa heater button on AqualinkD to turn on the spa heater.    

I do agree AqualinkD saying "Spa Mode" when it's actually "Spa" is confusing.  When I wrote this I was going by equipment and Jandy documentation I had available. Jandy change this with the introduction of other keypads,  and it's only recently (last few months) I realized that "Spa" and "Spa Mode" are actually different things with Jandy.

So : -   
"Spa" is Pumps & JVA's
"Spa Mode" is Pumps, JVA's & heater.

AqualinkD is Spa even though the button says Spa Mode.

If you are using a home automation hub, you can configure a "Spa Mode" button that selects both AqualinkD's Spa Mode and Spa Heater buttons.

mdetweiler, 2020-08-17:
Thanks very much for the quick reply and detailed explanation!  I should have mentioned that I really appreciate the work you put into this project!  Very nice job!

sfeakes, 2020-08-28:
Just posted V2.1.1.  That has a config option to link Spa Mode and Spa Heater.  Please see documentation / release notes.  I've restricted it to just webUI as to not confuse home automation implementations.