aqualinkd/AqualinkD #43: Freeze Protect Indicator or Data in MQTT

bsheedy, 2019-01-20:
I am using all MQTT options in my HASS.IO implementation and haven't had a chance until now to check the status of Freeze Protection.  I am polling aqualinkd/Freeze_Protect as an indication that it is on or off but it never changes to 1.   Please advise if AqualinkD does not relay the state of Freeze Protection to MQTT.  If not, it would be a great feature request as Home Assistant is the only way I can see my Freeze Protect status when travelling (As I wouldn't post AqualinkD to the internet as recommended.)

sfeakes, 2019-01-20:
It does (well should) relay the Freeze Protection information to MQTT.  It's on and off depending on if the control panel has turned on Free Protection due to the outside temperature being below the freeze proctection setpoint. It's not on / off depending on the "enameled" state like a heater. Well that's the intended functionality, I haven't tested it in a while, I'll do that soon and fix if there is a bug.

bsheedy, 2019-01-20:
Freeze Protection when activated on the RS6 Panel reports properly on all of the AqualinkD web pages (simple, controller...) perfectly.  It does not publish '1' on aqualinkd/Freeze_Protect in MQTT, and there is no evidence of aqualinkd/Freeze_Protect/enabled (Or aqualinkd/Freeze_Protect/enabeled) ever being published in MQTT.

bsheedy, 2019-01-20:
BTW - Thank you for all of the attention and response you give to this forum.  It should be noted that you are saving everyone 100's of dollars in unnecessary upgrades that are unfairly priced.

sfeakes, 2019-01-21:
I posted a quick fix, if you could check that would be great. Have a bunch of other updates to add, but wanted to get this out of the way first.

bsheedy, 2019-01-21:
Will DO!  Just waiting for the temp to get to 37 F, as it is not something I can trick the controller into doing - Freeze Protection will definitely be on in a few hours and I should have an answer in the morning (Houston time).  Thanks!

sfeakes, 2019-01-21:
I've obviously got some more debugging to do.  What you describe is how it's "supposed" to work.
aqualinkd/Freeze_Protect/enabled - should remain at 1 as long as you have devices assigned to Freeze Protect in the control panel.
aqualinkd/Freeze_Protect/ - should switch between 1 and 0 depending on if it's below the threshold.

I'll spend some more time on this and hopefully fix soon.

bsheedy, 2019-01-21:
No rush at all!

Consider that the only way you know it is 'activated' on an RS panel is through the text from the display on the console saying "Freeze Protection is Enabled" - it may not be present in a serial code.

But you know far more than I...  I am just grateful that 99.99999% of the functions are natively working in Home Assistant - and this one is not a deal-breaker.

sfeakes, 2019-01-24:
OK, it finally got cold enough while I had some time to debug this.  It's fixed and I've uploaded a new version.  I also added the battery state to MQTT topic (info in wiki).

bsheedy, 2019-01-24:
I have confirmed that "aqualinkd/Freeze_Protect/" changes to '1' when the temperature is below the threshold and that "aqualink/battery" works as designed.   I have noticed that freeze protection messages no longer appear any of the web pages (controller.html, simulator.html...) when it is activated.

Thanks!

sfeakes, 2019-01-24:
Great.  I've fixed the message in the web pages, and uploaded that fix.

closed 2019-01-24 by sfeakes