aqualinkd/AqualinkD #48: Homebridge-AqualinkD doesnt respond to turning or off a device

BigDogJonx, 2019-02-24:
After upgrading to the latest release I noticed that Homekit integration directions was listed and decided to install MQTT (mosquitto) and installed homebridge-aqualinkd since I already have Homebridge installed and running other things.  I got the config.json updated and got my phone to show the devices and what seems to be the actual air temp.  But if I turn on any device it shows its on in Homekit but the device isnt on.  State changes dont show up on homekit either if I turn on the device from AqualinkD homepage or actual RS panel.  

Not sure if I did anything wrong on MQTT side.  I noticed it said the following after installing it "Make sure to configure and make sure it starts on boot."  What specifically do we need to configure?  I have it set for starting on boot by running the command "sudo systemctl enable mosquitto.service".

sfeakes, 2019-02-24:
Homebridge-AqualinkD get's it initial list of devices and state directly from AqualinkD, after that it uses MQTT for realtime info.
So, it seems you have Homebridge-AqualinkD picking up the initial devices, (so that's configured correctly). You need to make sure both AqualinkD and Homebridge-AqualinkD are communicating with the same MQTT server and using the same topics.

So check /etc/aqualinkd.conf (MQTT server parts) and Homebridge config.json (Homebridge-AqualinkD MQTT part).  If they both look accurate, then use mosquito-sub to see if information is being posted correctly to MQTT.  `mosquitto-sub -t aqualinkd/# -v`.

BigDogJonx, 2019-02-24:
Thanks @sfeakes I got it working.  Turns out when I moved in the new aqualinkd.conf file (since I came from a really old version and noticed alot changed, I moved in the label parts into the new conf file) I didnt adjust the MQTT section.  Noticed it was all commented out.  So I uncommented out the mqtt_address line and mqtt_aq_topic line, restarted AqualinkD and boom, got it to now show the right air temp, and validated by turning on my airblower button for Spa, turned it off, worked.  Thanks again.

closed 2019-02-24 by BigDogJonx