aqualinkd/AqualinkD #49: SWG switching on/off

scroundy, 2019-02-27:
Love this project. Thank you so much for this great code. I have a Rev T2 PDA PS4 combo board. Everything works pretty well. However the aquapure panel beeps about every minute switching from the set aquapure % with "cell on" to 0% and cell off. It seems like it is conflicting or something somehow. If I turn off or stop aqualinkd then the system works fine. Any ideas?

sfeakes, 2019-03-01:
This is very odd, I can't see how / why it would do this at the moment.  AqualinkD doesn't change any messages between the control panel and aquapure panel, it just sniffs those messages.
On the Aquapure panel, there is a LED for "remote controlled", does that stay lit or does it flash on and off?
Also does the PPM and Percent show up correctly in the UI of AqualinkD?

scroundy, 2019-03-02:
My aquapure panel doesn't have a "remote controlled" LED. it is like this one: https://goo.gl/images/P9hsUg
The PPM and percent do show correctly. Although aqualinkd UI normally shows as generator off.

I validated this only happens when aqualinkd service is on. When I stop the service or operate the panel in service mode this doesn't happen. 

Are there some specific lines of code that I could comment out to ignore setting aquapure % or even reading aquapure to see if that helps?

sfeakes, 2019-03-02:
I posted an update last night that will allow you to configure AqualinkD not to even read messages from the SWG.  If you get the latest version and put the below in aqualinkd.conf it won't read anything directly from the SWG, only messages the Control Panel send to AqualinkD
```
read_all_devices = no
```

sfeakes, 2019-03-02:
Just out of interest, you don't have anything telling AqualinkD to set the SWG % to 0 do you?  Like something (like a home automation hub) sending a MQTT or HTTP message AqualinkD.  If you turn on debugging in the aqualinkd.conf file that might tell you.  Or post the log and I can take a look, as there is a lot of information to decipher.  Just make sure the log is going when it sets the Aquapure to 0.

scroundy, 2019-03-02:
I don't have MQTT interacting with SWG - I don't think. I'm using Home Assistant but I don't have anything in there around SWG.
Here's a few minutes of my log:  https://drive.google.com/open?id=1V5e1eWMwLT-9QIMCfr6w0SgpuvDWE2C_

scroundy, 2019-03-02:
try this: https://drive.google.com/open?id=1V5e1eWMwLT-9QIMCfr6w0SgpuvDWE2C_

sfeakes, 2019-03-02:
In that log, there was nothing that looked abnormal, Aquaprure was reporting 20% and 3200PPM the entire time, MQTT did disconnect and re-connect twice, but that didn't effect anything.  Did the SWG do the beeping / reset to 0 during the time that log was active?

BTW, if anything (Home Assistant) posts MQTT to aqualinkd/SWG/Percent/set or aqualinkd/SWG/Percent_f/set then AqualinkD will try to set the SWG percent to that value posted.

scroundy, 2019-03-03:
Yes it did have the problem. What are about a lines like these?
Line 100: 
Mar-02-19 13:21:38 PM Info:   MQTT: Published id=6: aqualinkd/SWG/enabled 0
Mar-02-19 13:21:38 PM Info:   MQTT: Published id=7: aqualinkd/SWG 0
line 3024:
Mar-02-19 13:28:30 PM Debug:  MQTT: Ignore aqualinkd/SWG/setpoint -6.666666

Although I did play around with this and it seems like the problem occurs even if I disable mqtt parameters in the aqualinkd.conf file. The one consistency is that when AqualinkD is running it occurs. If I turn off AqualinkD then it doesn't happen anymore. If I run aquaweb (python script from github) it doesn't have the problem. Not sure.

scroundy, 2019-03-03:
Here's another version of the log with me trying different things.
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2922715/aqualinkd.log)

sfeakes, 2019-03-03:
> Yes it did have the problem. What are about a lines like these?
> Line 100:
> Mar-02-19 13:21:38 PM Info: MQTT: Published id=6: aqualinkd/SWG/enabled 0
> Mar-02-19 13:21:38 PM Info: MQTT: Published id=7: [...]

First two messages are simply AqualinkD posting SWG information to the MQTT topics. The 3rd message is simply a message AqualinkD sent and the MQTT server sent it back, so it was ignored.
These messages are the ones AqualinkD receives from the RS Control Panel
```
Mar-02-19 13:30:44 PM Info:   RS Message :- 'AquaPure 20%'
Mar-02-19 13:30:46 PM Info:   RS Message :- 'Salt 3200 PPM'
```
All of those states 20% and 3200 PPM, I'd have expects AqualinkD to see 0% messages, or for AqualinkD to tell the Control Panel to set the % to 0.  But there was nothing like that.

The other thing to try is set logging to DEBUG_SERIAL, then you should get every RS message AqualinkD receives and sends.
Did you try the latest software with `read_all_devices = no` ?

sfeakes, 2019-03-03:
> Here's another version of the log with me trying different things.
> [aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/2922715/aqualinkd.log)

Interesting thing in this log, is you tried to set the SWG to 20%, and it didn't finish cleanly.  Since it was already 20% it's hard to say if it would have worked or not. It looks like the setting / changing worked, but the "confirm" message on your panel is different to others.
This doesn't explain your problem though, but does show your SWG is slightly different to any others I've come across.
Below is the part of the log where it tried to set the SWG %.
```
<something MQTT / HTTP told AqualinkD to set SWG to 20%, so it started a thread to do that>

Mar-02-19 19:00:43 PM Debug:  Thread 21660784 is active
Mar-02-19 19:00:43 PM Notice: Setting setpoint of Salt Water Generator to 20
Mar-02-19 19:00:43 PM Debug:  programming SWG percent to 20

<below the thread is running through the control panel menus to find & select the right menu for SWG> 

Mar-02-19 19:00:43 PM Info:   Queue send '0x09' to controller
Mar-02-19 19:00:43 PM Debug:  waitForMessage PRESS ENTER* TO SELECT 3
Mar-02-19 19:00:43 PM Debug:  Programming mode: loop 1 of 3 looking for 'PRESS ENTER* TO SELECT' received message '03/02/19 Sat'
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Message length 24
Mar-02-19 19:00:43 PM Debug:  Kicking thread
Mar-02-19 19:00:43 PM Debug:  Programming mode: loop 2 of 3 looking for 'PRESS ENTER* TO SELECT' received message '03/02/19 Sat'
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Lng Message length 24
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Lng Message length 24
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Lng Message length 24
Mar-02-19 19:00:43 PM Info:   RS Message :- 'Press ~ for more options... Press Enter* to select'
Mar-02-19 19:00:43 PM Debug:  Kicking thread
Mar-02-19 19:00:43 PM Debug:  Programming mode: loop 3 of 3 looking for 'PRESS ENTER* TO SELECT' received message 'Press ~ for more options... Press Enter* to select'
Mar-02-19 19:00:43 PM Debug:  Programming mode: String MATCH
Mar-02-19 19:00:43 PM Debug:  Programming mode: found message 'PRESS ENTER* TO SELECT' in 'Press ~ for more options... Press Enter* to select'
Mar-02-19 19:00:43 PM Info:   Queue send '0x18' to controller
Mar-02-19 19:00:43 PM Debug:  waitForMessage (null) 1
Mar-02-19 19:00:43 PM Debug:  Programming mode: loop 1 of 1 waiting for next message, received 'Press ~ for more options... Press Enter* to select'
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Lng Message length 24
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Message length 24
Mar-02-19 19:00:43 PM Info:   RS Message :- 'Press ~ for mor'
Mar-02-19 19:00:43 PM Debug:  Kicking thread
Mar-02-19 19:00:43 PM Debug:  Find item in Menu: loop 1 of 25 looking for 'SET AQUAPURE' received message 'Press ~ for mor'
Mar-02-19 19:00:43 PM Info:   Queue send '0x18' to controller
Mar-02-19 19:00:43 PM Debug:  waitForMessage (null) 1
Mar-02-19 19:00:43 PM Debug:  Programming mode: loop 1 of 1 waiting for next message, received 'Press ~ for mor'
Mar-02-19 19:00:43 PM Info:   RS Message :- 'Program      ~*'
Mar-02-19 19:00:43 PM Debug:  Kicking thread
Mar-02-19 19:00:43 PM Debug:  Find item in Menu: loop 2 of 25 looking for 'SET AQUAPURE' received message 'Program      ~*'
Mar-02-19 19:00:43 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:43 PM Debug:  RS Received STATUS length 12.
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Message length 24
Mar-02-19 19:00:44 PM Info:   RS Message :- 'Review       ~*'
Mar-02-19 19:00:44 PM Debug:  Kicking thread
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:44 PM Debug:  RS Received STATUS length 12.
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:44 PM Info:   Queue send '0x18' to controller
Mar-02-19 19:00:44 PM Debug:  waitForMessage (null) 1
Mar-02-19 19:00:44 PM Debug:  Programming mode: loop 1 of 1 waiting for next message, received 'Review       ~*'
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Message length 24
Mar-02-19 19:00:44 PM Info:   RS Message :- 'Set Temp     ~*'
Mar-02-19 19:00:44 PM Debug:  Kicking thread
Mar-02-19 19:00:44 PM Debug:  Find item in Menu: loop 3 of 25 looking for 'SET AQUAPURE' received message 'Set Temp     ~*'
Mar-02-19 19:00:44 PM Info:   Queue send '0x18' to controller
Mar-02-19 19:00:44 PM Debug:  waitForMessage (null) 1
Mar-02-19 19:00:44 PM Debug:  Programming mode: loop 1 of 1 waiting for next message, received 'Set Temp     ~*'
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Status length 12
Mar-02-19 19:00:44 PM Debug:  RS Received STATUS length 12.
Mar-02-19 19:00:44 PM Debug:  RS received packet of type Message length 24
Mar-02-19 19:00:44 PM Info:   RS Message :- 'Set Time     ~*'
Mar-02-19 19:00:44 PM Debug:  Kicking thread
Mar-02-19 19:00:44 PM Debug:  Find item in Menu: loop 4 of 25 looking for 'SET AQUAPURE' received message 'Set Time     ~*'
Mar-02-19 19:00:44 PM Info:   Queue send '0x18' to controller
[... 35 lines cut ...]
Mar-02-19 19:00:45 PM Debug:  Programming mode: String MATCH
[... cut ...]
Mar-02-19 19:00:45 PM Debug:  RS received packet of type Message length 24

<This is where yours is different. We are looking for a message :- >
< "POOL SP IS SET TO">
< but your returns (ignore case) >
< "Pool set to ">

Mar-02-19 19:00:45 PM Info:   RS Message :- 'Pool set to 20%'
Mar-02-19 19:00:45 PM Debug:  Kicking thread
Mar-02-19 19:00:45 PM Debug:  Programming mode: did not find 'POOL SP IS SET TO'
Mar-02-19 19:00:45 PM Debug:  Thread -1227299728 finished
```

I'll make a change to fix this, but it won't fix your main problem.

scroundy, 2019-03-03:
Thank you for looking into it. I did try read_all_devices = no and it didn't seem to make a difference. It is most likely a problem with my board or sensor - since my pump has been running today and isn't having the problem.  I'm going to close this so you can spend time on variable speed pump settings :)

closed 2019-03-03 by scroundy