aqualinkd/AqualinkD #50: AqualinkD I think is randomly turning on my Jandy ePump

BigDogJonx, 2019-03-01:
Ive had this for about 12months now, and somehow something recently for the past week has been happening which is somehow something is turning on my ePump (Jandy Variable Speed) to full speed, but here is the kicker, the RS8 keypad has nothing on, shows its all off.  Login into iAqualink app and shows its off.  But clearly pump is on since I hear it running and I have an energy monitor that shows its pulling almost 1800watts.  Then out of a hunch I decided to check the WEB portion of the iAqualink app, it basically logs you into the Web view.  Then I clicked Status, and there I see ePump is on, running full tilt, yet it says "No devices are currently on" below it.  Now if I simply turn ON the pump from the app and turn it off, it actually turns off, but at random intervals of Id say 0-12hrs it turns itself back on.  So out of a hunch I rebooted my RPi3, still did the same thing next day.  Then this time I disconnected the RS485 wiring block off the USB module (I bought this one off Amazon I believe was the recommended one - https://www.amazon.com/gp/product/B01LCFRR3E/ref=oh_aui_search_asin_title?ie=UTF8&psc=1).  

After disconnecting the wiring block off the USB serial adapter, its now been 3days and no issue of it randomly turning on.  

So now trying to figure out what on the RPi3 side is causing this.  Only other thing the RPi3 hosts is my Homebridge for Homekit.  Dont use it for anything else.

Here are the pictures of the app side to help understand.
https://ibb.co/YdfM1hp
https://ibb.co/vX4X1mC
https://ibb.co/LPzJmVM

sfeakes, 2019-03-02:
AqualinkD doesn’t even send commands directly to the pump, so this is very strange. It actually doesn’t even look at any messages from the pump either, it’s the next thing on my list to try to work out though.
Since iAqualink / keypad doesn’t show the pump as on, I’m assuming AqualinkD doesn’t either?

From my understanding, The RS485 pumps have 2 modes, local control and remote control, it should be in remote control (so the RS control panel can control it). Other RS485 devices like salt water generator have an LED that tells you if it’s in local or remote control. Does the pump have some form of indicator (like an LED) that shows the state?  When the pump is on but the control panel doesn’t show it’s on, I’d check that first, and see if the pump thinks it’s being controlled remotely or locally.  If it’s in local then this would explain why the pump is on and the control panel doesn’t know about it.  Still that Doesn’t explain why it does into local control mode, but it may help in figuring it out.

BigDogJonx, 2019-03-02:
First off let me State I’m not knocking or saying it’s the fault of AqualinkD as I don’t have any proof of that.  Now that I know it’s related to something on the Rpi side now need to investigate that more. I loved the AqualinkD not to mention it was fine for 11months. 

So to answer some Questions. No I don’t see it on from the AqualinkD side either. As a matter of fact even the Jandy SWG isn’t even on as the LCD display isn’t on. The ePump doesn’t have a controller on the pump like other models do. This one is purely a controller less meaning just be used with a RS panel. I’ll have to look at the manual again to see if there is any LEDs near the pump body. 

My next option was to plug the plug connector back on the USB serial adapter and then enable debug mode on the AqualinkD and maybe setup and alert on my energy monitor to look for the 2kW spike and then see if I can see anything on the logs to indicate what it saw or did it send any messages. And if I’m there at home maybe then analyze what’s going on physically at the pump if it has some local indication of what mode it’s in. 

Other option is also to completely blow away my RPi and reload it from scratch for the homebridge and AqualinkD which should be easy since I have the conf files. I feel pretty confident something from this RPi is sending something on the USB and some message somehow is being intercepted by the pump to go into full tilt.

sfeakes, 2019-03-02:
If it is Pi related, then it probably is something AqualinkD is doing. Unless you have installed some other software on the Pi that can do the RS458 protocol. So I don't think re-building the pi will help (but it can't hurt). Maybe use a new CF card, that way you can always go back. The way the RS485 protocol works, it is very hard to interject a "bad" command, so loose connection / some interfereance etc will always stop things from working, never interject an erroneous command.  

I don't fully understand the RS485 pump(s) protocol yet, but what I do know is you can basically send a command and forget (ie turn on and go into local control mode). I would expect that command to do exactly what you are seeing. Or (turn on, and stay in remote mode), where the RS485 control panel has to re-send the command (or an acknoladgement) every X seconds, and if not received the pump will turn off. That's now the RS control panel expects to control a pump.

So assuming there is not a problem with the pump itself, and the Pi is forcing a message to make the pump do this, it would seem the pump is receiving the "set and go into local control" command.

If you want to monitor the buss to see what's going on, you can use the serial logger that's part of AqualinkD to do that.  Information is in the Wiki. It will generate a ton of information though, so you may want to reduce the output to just the ID of the pump.  Remember you can't run AqualinkD and the serial logger at the same time, (unless you have 2 USB to RS485 adapters).

I use the same adapter you posted for my system, and have a few others I use for testing, I can send you one if you want, and you can send it back when finished. But for the price they are on amazon, it's probably not worth it. The one you linked is the best I've come across so far, the below link is a cheaper one that's also very good, if you did want to test with a different adapter, or run AqualinkD at the same time as the serial logger.
https://www.amazon.com/gp/product/B00NKAJGZM/ref=oh_aui_search_asin_title?ie=UTF8&psc=1

BigDogJonx, 2019-03-02:
Yea, thats what I was thinking (as far as logging went).  Good to know we can filter it to the pump only.  How do I know which one is the pump vs the SWG vs the heater?  I didnt know we could see specific ones.

Im also going to try a few more tests before I do the logging.  One would be turn off AqualinkD service and leave it connected just to rule out AqualinkD or not.  If it doesnt happen with AqualinkD service stopped, then that tells me has to be AqualinkD.  Then next steps would be to do the logging to understand whats causing it.  

I dont have any other software using RS485, the only other software we have on here is Homebridge and Homebridge plugins (Wemo, Alarm, Cameras) which are all IP based.  So dont think that is affecting anything.

sfeakes, 2019-03-02:
https://github.com/sfeakes/AqualinkD/wiki/Jandy-Aqualink-RS485-protocol

The above is everything I know about the Jandy RS485 protocol. (not written very well I know, mainly my notes). AND it may well be wrong in places, so don't take it as gospel.

Each set of devices usually have 4 ID's they can use. (except iAqualink remote & PDA remote).
SWG are 0x50 to 0x53.
Pumps are 0x78 to 0x7b. (THIS IS A GUESS, I DON'T HAVE ANYTHING TO TEST).
So that's id's of 0x78, 0x79, 0x7a & 0x7b

To start with I'd run the serial logger in standard mode so it reports all devices it sees on the buss, making sure to turn the pump on / off while it's calculating. Then at the end it should spit out all the ID's it saw, and hopefully you'l see one of those.
Or set the serial logger to report just on those ID's, turn the pump on / off / change speed etc, and hopefully you'll see one of those ID's coming up in the logs.  

Remember the RS485 Control panel is constantly polling every ID, so just seeing a PROBE and no ACK, means the ID is not used, or device is off.

Once you figure out the ID, then set serial logger to just that ID and let it run as long as you want.

BigDogJonx, 2019-03-05:
Just want to give an update here. Before jumping into logging I decided to cold boot the RPi3. I know before I said I rebooted it which I did and problems still occurred. This time I decided to unplug it. Unplug USB. Wait a minute and plug it in. It’s much much better. I had one occurrence only but not sure entirely what to make if it just yet. I noticed pump came on and again nothing on the display so I did the normal turn pump on and then back off. But this time wouldn’t let me turn it off because it said need to wait for cool down mode which meant somehow now the heater was on too. That was about 3days ago. Hasn’t happened since and haven’t rebooted or cold booted again. 

This is going to be some long troubleshooting as I try to rule out areas. But leaning more now on it’s something RPi related now as somehow clearly the cold boot clearly reduced the occurrences bigtime. 

Dixit

BigDogJonx, 2019-03-13:
Sorry been awhile. Been doing some other tests. Unfortunately this is not a quick process since it’s on some random instance this pump situation happens. I decided to another test which I mentioned above which was cold booted RPi and it was better but eventually after about 5-7days it started doing the same thing again. 

So then yesterday night I decided let’s now try another thing to rule out things one by one. This time I decided let me just turn off the AqualinkD service, which I did last night. Low and behold just noticed moments ago which has now been 22hrs later the pump is again randomly back on. 

So again checked my RPi just to make sure it wasn’t AqualinkD and AqualinkD is definitely not running. So therefore I guess I must now rule it out. Now I just decided let’s see what happens if I simply now unplug the terminal block off the USB RS485 device which that terminal block connects to my RS8 keypad about 20ft away. Not changing anything else. Let’s see what happens tomorrow. If nothing happens, then my hunch is something on RPi is causing this which is not AqualinkD. 

If it still happens. Clearly ruling out anything RPi related and now must be some wiring or possibly the board or somehow the damn iAqualink itself going crazy. Which is going to suck as that board alone New now is $500 or so. Gotta love having a pool. Hahha. 

Dixit

dethpickle, 2019-05-28:
On the off chance you're still having issues here, I wanted to add a possibility I didn't see above.  If you're using MQTT with aqualinkd, there's always a chance you have some stray _retain_ information in your queue that might cause a status change when the MQTT server cycles.  If so, it'd sure seem from the outside like aqualinkd is sending a command when it's really just parroting the old command retained in the MQTT queue.

I've not had any issue like this with aqualinkd, but I have had similar issues with normal MQTT based automation and my shaky knowledge of how and when to use retain and QOS levels with queuing.

If you think this might be relevant, have a look at this: https://youtu.be/NjKK5ab0-Kk
This is a shot in the dark, but I hope it helps.

BigDogJonx, 2019-05-30:
@dethpickle Excellent info there.  I'll look into it, for now Ive been keeping the RS485 wire disconnected from the serial to usb adapter.  Im going to replug it in and try seeing if this retain information is causing it.  I AM using MQTT since I wanted it connected to homekit, which was done recently, prior (about 6-8months ago) it wasnt setup with a MQTT broker.  So seems like this could be the cause.  

Only thing here that even on this stray retain message somehow being sent, my thing is how is it somehow telling the RS8 panel to turn the pump on, but neither the RS8 keypad or even if I log into iAqualink app shows the pump is running.  Only way is by looking at Status on the iAqualink app or web browser and somehow that knows its on because that I believe is essentially just scanning the RS485 line and showing what devices are online/offline (like it doesnt show my SWG when pump is off because its offline as expected).  Thats the part throwing me for a loop is how somehow it actually sends some commend that the keypad/app/panel has no idea about and makes it seem nothing is on, but clearly it is (and in my cause its a variable speed pump and of all things causing it to run at full tilt).

sfeakes, 2019-05-31:
The more I think about this, the more I think the pump is loosing connection with the control panel and goes into "stand alone" mode, and for some reason that mode's default is pump on.  Ig any of the Jandy equipment turned on the pump then the status would be posted in iAqualink & AqualinkD, since you're just getting RPM in iAqualink, I think the pump it turning itself on.  VSP's have two modes, one it "let someone else control me" the other "local control". For "someone else control me" the control panel has to ping the VSP a few times a second, if it misses those pings it will do into "stand alone mode".  Next time it does this, check the LCD of the pump, see if it gives you any information.

BigDogJonx, 2019-06-01:
@sfeakes, only thing is we dont have a "standalone" pump model.  So doesnt have any LCD on the back of the pump.  Its meant to only work with an RS/iAqualink based panel.  They sell the pump in basically two models, one with and without that built in LCD/Keypad on the motor body itself.  

Lastly only thing to sort of go debunk the theory of it losing connection would be what Ive tested and proven is the cause which is something on the RPi side.  I mentioned earlier on my findings that by simply disconnecting the RS485 wire going into the Serial to USB adapter for the RPi, I never got the pump ever randomly starting itself up.  So in my mind that rules out anything on the pump side or loose wiring on the RS485 side on the outside (any of the equipment being the heater, SWG, Pump, RS8 keypad, which is actually inside).

BigDogJonx, 2019-09-25:
Sorry its been so long on this @sfeakes 
But In think Ive confidently figured out the source of the issue after months or playing around and testing as the issue itself is not reproducable on demand. 

Remember I said It must be something with the RPi because even with the service off it was happening.  Then disconnecting the RS485 plug to the Serial to USB Adapter (https://www.amazon.com/gp/product/B01LCFRR3E/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) fixes it?
Well I did one test where I didnt disconnect the RS485 plug to the adapter, I left that on, but instead disconnected the USB cable going from that adapter to the RPi and guess what, the issues kept happening.  But if you remove the RS485 plug going to the Serial/USB adapter, it stops.  So somehow this Serial/USB Adapter even with it not plugged in causes some weird havoc on that serial line and causes my ePump to turn on randomly at full tilt.

So Im going to close this as I dont see this a fault of the iAqualinkD in anyway.

closed 2019-09-25 by BigDogJonx