aqualinkd/AqualinkD #53: Request Spa Drain

jsrichie, 2019-04-01:
The RS8 has buttons on the right side for Spa Drain / Spa Fill
We have the ability to Spa Fill but NOT Spa Drain. 
Can we get the Spa Drain as a button same a Fill?
Other options, I hate having to go out to side to drain and its the only thing beside getting in the Spa I have to do 👍 outside due to this great project !!!!!!

sfeakes, 2019-04-01:
It will depend on you installation, but these are probably Aux 6 and Aux 7.  Look in the configuration, all buttons on the RS8 should be supported, there are 12 buttons in total that can be configured. First 2 are usually fixed to pump & spa, 10 & 11 are usually fixed to spa and pool heat. Leaving 3 to 9 and 12 open for user configuration. The other 1,2,10 & 11 can be configured differently with some knowledge, but I haven’t seen any installers do that yet.
AqualinkD by default hides unconfigured buttons, so you need to configure them in AqualinkD.conf and make sure the UI is not hiding them in config.js.   Info on both is in the wiki.

jsrichie, 2019-04-01:
The Button under the Valve that cycles Pool Mode / Spa Mode / Spa Drain / Spa Fill so It would appear to be a hex code that I can not find to ID and assign.

sfeakes, 2019-04-01:
Oh, do you mean the buttons on the control panel that’s located by the equipment, and not the RS8 keypad buttons?  I don’t think there is any way to control those. AqualikD mimics the keypad, so if you can’t do it through the keypad, AqualinkD can’t do it.  I’d look to see if they is a way to do it through the keypad, if you don’t have a keypad then try the simulator UI in AqualinkD.

john1158way, 2019-04-16:
@jsrichie  I am interested in the SPA controls too.   Have you been able to find a way to switch modes using the simulator?   I will have a look.   I would like to get some background as your setup seems a little different than mine.  I have a Jandy Aqualink RS running  6521 REV 5  for my setup SPA mode that I can set up with AqualinkD puts the SPA in re-circulation mode.  SPA fill (suction from pool, discharge through SPA) is not available to me in AqualinkD nor is SPA drain (suction from SPA discharge to pool) in pool mode a slipstream of water on the discharge side goes through the SPA and an overflow water feature is enabled so I do not drain my SPA very often as the water is recirculated through the pool in normal operation.   In the manual I see that DIP switch Aux 3 can be used for SPA overflow which would probably mimic SPA fil and I wonder if that is how you get SPA fill.  Let's see if we can figure this out.

john1158way, 2019-04-20:
@jsrichie I have looked through the various menu options in the simulator and using my touch panel.  I did not see a place to set the drain mode.  What I have moved on to is to look at  my aux  outputs to power the valve actuators to the correct position for the modes.

john1158way, 2019-04-21:
while looking at how to wire in the drain mode to the valves I came across the menu path diagram for a One Touch Jandy controller and it looks like through the One Touch there are more mode settings.  https://www.jandy.com/-/media/zodiac/global/downloads/0748-91071/6594.pdf page 39.  So if a One Touch can send codes for additional modes then AqualinkD should be able to send codes.  To confirm and find the codes We just need to find someone operating a One Touch that can run the serial logger

sfeakes, 2019-04-23:
It's not quite as simple as that, One touch and PDA are really archaic protocols, so it won't be one simple command or even command sequence to send. But, if it is supported then it's possible.  But I don't see anywhere in the flow chart that would indicate spa drain / spa valve etc. I think the best option is one you already mentioned, just plug the valve into one of the spare JVA sockets and assign that.

john1158way, 2019-04-24:
thanks Sean,  did not realize that One Touch was so problematic.  When looking at the AUX method I believe I have to wire a relay into the valve wiring as the JVA controller keeps one of the 3 valve actuator wires wire hot all the time and uses a physical limit switch to stop the valve when it rotates to the proper position.  Looking at JVA documentation the JVA has a common wire and two hot (24v) wires. The valve position is determined by which of the wires is hot. Right now I have SPA mode which recirculates the SPA and Pool mode which draws on the pool and splits the discharge between Pool Jets and the Spa with an overflow from the SPA creating a water feature. I was thinking that I would use and AUX connection to to power an extra relay I have in my control box. I would put this relay in the wiring path of the discharge valve actuator. With the AUX off button the power to the valves would be routed  through the normally closed contacts as per normal but when the relay is pulled in the power is switched to opposite in the Normally Open Contacts. So if in Pool mode when the relay pulls in it switches the discharge from Pool/SPA combo to 100% SPA and I get SPA Fill

john1158way, 2019-04-24:
If there is a way to control valves directly through an Aux setting that keeps the existing control methods working as well that would be great to know.

john1158way, 2019-04-24:
found first blocker to my thinking,  my relay does not have normally closed contacts that are connected when the relay is not pulled in only normally open that connect when the relay pulls in.  I  would need to mount a different relay in box  that has double pole normally closed contacts.

stephengpowell, 2019-04-26:
Hello. I ran across this post and have the Jandy RS8 pool and spa. I am not clear on what I need to implement. Do I need the USB interface and Raspberry PI Zero

sfeakes, 2019-04-26:
Yes.  You need a USB2RS485 converter and a raspberry pi (any model will work).  But you should check the version# of the RS8 for what’s supported.  if PDA only # you might not want to bother.

ballle98, 2019-04-26:
@stephengpowell this is a little off topic here, there is a forum http://aqualinkd.freeforums.net for questions and discussions, but since you asked the question...  One more thing you need to consider is how you want to connect aqualinkd to Alexa.  I'm using ha-bridge on a Pi 2B+ (similar performance to zero) with aqualinkd and it's fine but if you need more resource intensive bridges and gateways (openhab etc.) they may require the extra memory in a Pi 3.

More information on connection options is here: https://github.com/sfeakes/AqualinkD/wiki#Generic

john1158way, 2019-05-28:
I have implemented my Spa Drain / Spa Fill work around.    I purchased a 24v coil DPDT relay and wired the Jandy return valve control wires (red and white) through the normally closed contacts and cross wired the the red / white wires through the normally open contacts.  I then wired the coil to aux 5.   So now with Aux 5 off the Pool mode and SPA mode work normally.  In Pool mode when Aux 5 is engaged the return valve reverses from discharge to pool to discharge to Spa. the fill mode.   In SPA mode when Aux 5 is engaged the valve reverses to discharge to pool the drain mode.    What I am going to do next is add code to the web page to label the Aux 5  feature with a pop up and confirmation the user wants the SPA drain / fill action as well as executing the correct combination of mode and Aux 5.

nmoscone, 2020-06-23:
> I have implemented my Spa Drain / Spa Fill work around. I purchased a 24v coil DPDT relay and wired the Jandy return valve control wires (red and white) through the normally closed contacts and [...]

> I have implemented my Spa Drain / Spa Fill work around. I purchased a 24v coil DPDT relay and wired the Jandy return valve control wires (red and white) through the normally closed contacts and [...]

Hi John,

Would you mind sharing a wire diagram? I’d like to do the same to my controller?

john1158way, 2020-06-24:
[image]

the return valve is the valve that returns water from the filter to the pool or spa

john1158way, 2020-06-24:
note I checked the voltage on my JVA control and it is 48v not 24v and used a DPDT relay with a 48v coil

nmoscone, 2020-06-24:
Thank you I really appreciate it.  mine is the JVA 2440 and it says 24v but I will double check with a meter first just to be sure.

This is a nice work around!  I have been walking down using service mode to have the spa over flow effect into the main pool but is a pain having to go to the main panel / lose the other buttons when in service mode!

Nick