aqualinkd/AqualinkD #54: HTTP Api matching on default PDA button names rather than assigned name in aqualind.conf

flyrdm, 2019-04-02:
I debugged the code section in net_services and it is matching on default button names for the HTTP Api. Is this intended behavior? 

**** If I do the following:

curl http://localhost?command=Spa_Bubbles&value=off
[1] 4220
dietpi@PoolPi:~$ Unknown command

**** Now if I try:

dietpi@PoolPi:~$ curl http://localhost?command=Aux_2&value=off
[1] 4230
dietpi@PoolPi:~$ Ok

**** Relevant conf section:

##button_04_label=Waterfall
button_04_label=Spa Bubbles
#button_04_dzidx=40
#button_04_PDA_label=AUX2

**** Relevant logging:

Apr-01-19 18:59:07 PM Info:   URI request: '/?command=Spa_Bubbles'
Apr-01-19 18:59:07 PM Info:   WEB: Message command='Spa_Bubbles'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Filter_Pump'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Spa_Mode'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_1'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_2'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_3'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_4'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_5'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_6'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Aux_7'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Pool_Heater'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Spa_Heater'
Apr-01-19 18:59:07 PM Info:   WEB: Message request button name match input/btn??: 'Spa_Bubbles' / 'Solar_Heater'
Apr-01-19 18:59:07 PM Info:   RS Message :- 'Air Sensor OK'

sfeakes, 2019-05-22:
This is a known issue, should be fixed in the next PDA release.

sfeakes, 2019-05-22:
I was just looked at this again, you have the names incorrect, that's why it didn't match.  (I was confusing this with a slightly different issue on the PDA).
You have `Spa Bubbles` as the name and `Spa_Bubbles` in the request.  space vs underscore.

You also need to configure the PDA label. `#button_04_PDA_label=AUX2` (you have this commented out)  This is the label that the PDA displays as the button name.

closed 2019-05-31 by sfeakes