aqualinkd/AqualinkD #57: MQTT 1.6

abarbaccia, 2019-05-06:
My MQTT was upgraded to v1.6 and aqualinkd refused to connect (socket errors). Rolling back to MQTT 1.5 resolved this issue.   I have no reason to use v1.6 but wanted to flag if others are experiencing the issue.

dethpickle, 2019-05-17:
I'm seeing this too.

```
May 17 14:11:23 raspberrypi aqualinkd[511]: Starting MQTT client to 10.1.2.5:1883
May 17 14:11:23 raspberrypi aqualinkd[511]: MQTT Connection closed
May 17 14:11:33 raspberrypi aqualinkd[511]: Starting MQTT client to 10.1.2.5:1883
May 17 14:11:33 raspberrypi aqualinkd[511]: MQTT Connection closed
May 17 14:11:43 raspberrypi aqualinkd[511]: Starting MQTT client to 10.1.2.5:1883
May 17 14:11:43 raspberrypi aqualinkd[511]: MQTT Connection closed
```

Also on v1.6:
```
[14:09:44] INFO: Starting mosquitto MQTT Server...
1558127384: mosquitto version 1.6.2 starting
```

Unfortunately, my MQTT is part of another package where I don't have direct control of there version.  That package had been recently updated and I was just chasing down why this one client was no longer working.  All of my other MQTT clients continue to work.

dethpickle, 2019-05-17:
I can also confirm that the MQTT package I was using (Frenk's MQTT Server & Web client add-on for HASSio) updated the MQTT version 5 days ago, just before I started seeing this issue.

https://github.com/hassio-addons/addon-mqtt/releases
See under **Version Updates**

sfeakes, 2019-05-18:
I've just uploaded a fix for this, please test.
There is a bug in V1.6 of mosquitto where it can't take the full length of the client name, so I simply reduced the client name by 1 character.

dethpickle, 2019-05-18:
Is there a way I can do this in Hassio?  A CLI to load a specific version or is there a way to directly add a specific version as a "repository" in the Hass.io add-on store?

And, thanks!

sfeakes, 2019-05-18:
You shouldn't need to change hassio at all, just change AqualinkD to the latest version. (I put in a fix in AqualinkD to make it work with mosquitto v1.6).

Depending on how you installed AqualinkD will depend on how you upgrade it, look in the wiki.

dethpickle, 2019-05-18:
Oh geez...   I'm an idiot.  You're right.

dethpickle, 2019-05-18:
Pulled the new version and ran the install.sh again.  It's working beautifully.

```
May 18 14:45:53 raspberrypi systemd[1]: Starting Aqualink RS daemon...
May 18 14:45:53 raspberrypi aqualinkd[1993]: Aqualink Daemon v1.2.6f
May 18 14:45:53 raspberrypi aqualinkd[1993]: Config log_level         = 7
May 18 14:45:53 raspberrypi aqualinkd[1994]: Starting web server on port 80
May 18 14:45:53 raspberrypi aqualinkd[1994]: Starting MQTT client to 10.1.2.5:1883
May 18 14:45:54 raspberrypi aqualinkd[1994]: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
May 18 14:45:54 raspberrypi systemd[1]: Started Aqualink RS daemon.
May 18 14:46:01 raspberrypi aqualinkd[1994]: Control Panel B0029223 REV T.2
May 18 14:46:01 raspberrypi aqualinkd[1994]: Getting freeze protection setpoints
May 18 14:46:08 raspberrypi aqualinkd[1994]: Getting pool & spa heat setpoints from aqualink
```

Everything switching and updating as expected.  Thank you!

sfeakes, 2019-05-18:
Perfect, thanks for letting me know.

closed 2019-05-22 by sfeakes