aqualinkd/AqualinkD #58: 0 setting on Aquapure not reflecting on Web(API) or MQTT

niharmehta, 2019-05-07:
I noticed that when I set my SWG to 0, the Web Page, Web API, and MQTT continue to reflect the last value vs. the 0% value.  It reports correctly  at any other setting I have set via the RS.   Also cannot seem to publish a new SWG setting using MQTT or Web API.  Seems like the set command is ignored. 

When looking at logs, if a set a percent setting using the RS,  the logging reflects the update, however when using RS to set to 0, the log does not seem to reflect the new setting although it is indeed set on the RS. 

**MQTT Ignoring Set Command.**

` pi@aquad:/var/log $ tail -f /var/log/aqualinkd.log |grep MQTT
May-07-19 01:15:54 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent/set 10 
May-07-19 01:16:14 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent/set 25 
May-07-19 01:16:19 AM Debug:  MQTT: Ignore aqualinkd/SWG_Pool/Percent/set 25 
...
May-07-19 01:20:41 AM Debug:  MQTT: Ignore aqualinkd/SWG_Pool/Percent/set 0

niharmehta, 2019-05-07:
I have Control Panel B0029221 REV T with iAqualink as my only 'panel'  and Hayward SWG running in Aquapure Emulation mode.     I also do wait a couple of minutes until the SWG is updated in the web UI until I issued commands. 

This is really odd, and having difficulty confirming definitely what the fault states are.    As you said I checked with mosquitto_sub and it seemed to work.   The correct value was reflected in the RS and the web. I may be seeing two potential issues.  Really odd is this morning, the 0 value working with the Web API and seemed to show correctly on the RS. 

I think one issue may be related to how my system/RS is set up.   I can set the Aquapure percentage for Spa and Pool separately. However when I was testing the MQTT commands, it is only updating the Pool value in the Aquapure config on the RS. However, when I am in Spa mode, and send the MQTT call, the Pool value gets updated, UNLESS the pool setting selected is the same as the Spa current setting if the Spa is currently running.   
For example, the Spa Mode is currently running at 15% SWG.  Below, when the commands are run, the Pool SWG values are updated, unless I set it to the same % as currently running Spa Mode.  

`May  7 09:07:23 aquad aqualinkd: Setting SWG % to 0
May  7 09:07:23 aquad aqualinkd: Setting setpoint of Salt Water Generator to 0
May  7 09:07:37 aquad aqualinkd: Setting setpoint of Salt Water Generator to 15
May  7 09:07:37 aquad aqualinkd: Setting setpoint of Salt Water Generator to 15
May  7 09:07:37 aquad aqualinkd: SWG  15s already 15, not changing
May  7 09:08:37 aquad aqualinkd: Setting setpoint of Salt Water Generator to 20
May  7 09:08:37 aquad aqualinkd: Setting setpoint of Salt Water Generator to 20
May  7 09:08:37 aquad aqualinkd: Setting SWG % to 20
May  7 09:08:37 aquad aqualinkd: Setting setpoint of Salt Water Generator to 20`

Next, I  think it is possible  some of the issues may also be related to the 'read_all_devices' flag . When set to 'no',   I can make the MQTT publish commands, which are reflected in the web app, Web API,  but NOT on the RS at all.  The Debug logs show the command was accepted by the aqualinkd, and the webUI is showing the value. However RS does not change.   When setting the read_all_devices= yes" , it does seem to work. So this may be a critical setting for SWGs to work 

Hope this all helps.  I have not yet been able to recreate the 'ignored' error yet. I will keep testing later.

sfeakes, 2019-05-07:
There is a lot going on here which will add to the confusion in the abnormalities your seeing.

1) There is a bug in setting SWG percent to 0 through the web API where it will be set to 0 but it won't be reflected in AqualinkD.
2) SWG is coded to just support one setting, the pool.
3) read_all_devices will intercept messages between teh SWG and control panel to post more accurate information in AqualinkD, as what you can get from the control panel itself is not that great.

So,
1) I found the bug and will post a fix in the next update.
2) I never knew you could set SWG separately between pool & spa, I'll have to research and re-code this. But it will be some major changes to the UI's, so not sure when that will be available.
3) I think this may be a red herring as #1 will be hit n miss as to if it works depending on some other things.

I short, you'll probably have to wait until I post fix for #1, and #2 I may need a lot of debug information from you to fix depending on what I can find from manuals.

niharmehta, 2019-05-07:
Thanks.. makes sense. I normally do not need to make a lot of changes to the SWG, so it is not critical. It is more important to me to capture any possible telemetry from the salt cell so I know it is working and producing, not producing, salt readings, and errors/faults because of temp, salt, or cleaning needed.  Any changes can be done in the iAqualink UI if needed.   VSP and the OneTouch Macros would be a higher priority for me. 
I am happy to provide you secure access to my pi and RS system if you need to poke around.

sfeakes, 2019-05-07:
I think I've figured out how to ass spa swg easily.  It shouldn't take too long, but I also need to merge in some other PDA code that will take some time before posting update.  Try to get it done at the weekend.

Thanks for the offer of access to your system, I may take you up on that when I get time.  I really need to get VSP working and an understanding of the iAqualink protocol, and since I have neither I'm SOL without buying new hardware.

BTW, you should get ALL status SWG information posted to MQTT aqualinkd/SWG/fullstatus with `read_all_devices= yes`, far more than the control panel tells you.  Look in the MQTT section of the wiki for what each number represents.

I'll update this thread once I post the fixes, so I'll leave it open for now.

sfeakes, 2019-05-18:
I've just uploaded a fix for this, please test and report back.

niharmehta, 2019-05-18:
The good news is that it does seem to set 0% via MQTT and the Web API . 

 However, it defaults to the Pool setting for the SWG and I did not see a way to specify the Spa setting.  Even when the SPA mode was active, it was changing the Pool setting.  Even the F to C conversion gives unexpected numbers. However, I am not sure if that is normal as I do not use Homekit or IOS. 

The logs seem to have some odd settings being set when the SWG is changed:
May-18-19 02:01:42 AM Info:   MQTT: Published id=279: aqualinkd/SWG/setpoint -6.666666
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent 20
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent_f -6.666666
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/setpoint -6.666666
May-18-19 02:01:42 AM Info:   MQTT: Published id=280: aqualinkd/SWG/Percent 0
May-18-19 02:01:42 AM Info:   MQTT: Published id=281: aqualinkd/SWG/Percent_f -17.777779
May-18-19 02:01:42 AM Info:   MQTT: Published id=282: aqualinkd/SWG/setpoint -17.777779
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent 0
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent_f -17.777779
May-18-19 02:01:42 AM Debug:  MQTT: Ignore aqualinkd/SWG/setpoint -17.777779
May-18-19 02:01:44 AM Info:   MQTT: Published id=283: aqualinkd/SWG/Percent 20
May-18-19 02:01:44 AM Info:   MQTT: Published id=284: aqualinkd/SWG/Percent_f -6.666666
May-18-19 02:01:44 AM Info:   MQTT: Published id=285: aqualinkd/SWG/setpoint -6.666666
May-18-19 02:01:44 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent 20
May-18-19 02:01:44 AM Debug:  MQTT: Ignore aqualinkd/SWG/Percent_f -6.666666
May-18-19 02:01:44 AM Debug:  MQTT: Ignore aqualinkd/SWG/setpoint -6.666666

The more concerning issue is that when AqualinkD is on, and I set the SWG to zero via AqualinkD or the iAqualink,  the panel shows the pump Offline  and/or the Watts are intermittently WAY out of range when setting from 0 to another value.    If I shut down the AqualinkD service, no erroneous readings show up on the panel when making a SWG change to '0' or back to another value

Screenshots attached. 
[image]
[image]

sfeakes, 2019-05-18:
If spa mode is on, then SWG should default to setting & showing spa %, otherwise it’s pool.  I’ll have to take another look into this.

The iAqualink stuff is very odd, did this before the last update?

sfeakes, 2019-05-18:
Fixed the Spa SWG issue.  I've been working on two different code bases for the PDA update and pushed the wrong code.

Please try the latest version, still not sure if it'll fix the iAqualink display issue.

niharmehta, 2019-05-18:
Looks like the the 0% and the Pool/Spa settings are working now. The SWG percent command updates the currently running mode.    Not sure if there is a way to send a swg_percent command  to a specific mode vs. the currently running one only.  

The errors showing on the the iaqualink still continue.  I cleaned and downgraded to 1.2.5 to check if this is a new issue or existing, and it exists in 1.2.5  as well.   Stopping the aqualinkd service seems to resolve it, so I do not think it is wiring or anything on the hardware side.

sfeakes, 2019-05-18:
Glad the SWG is working.  At the moment there is no way to specify SWG for Spa or Pool separately, what ever is active is the SWG that will be set.  I'll look to adding that in the future.

On the iAqualink stuff, can you run the Serial logger when SWG is set to 0, and post the complete output?  (make sure AqualinkD is not running, then run the below)
```sudo ./release/serial_logger /dev/ttyUSB0 -d -p 1000```

niharmehta, 2019-05-18:
[serial_log_1.log](https://github.com/sfeakes/AqualinkD/files/3194573/serial_log_1.log)

Log files attached.  Set Pool mode from 25 to 0, then did the same for Spa mode.   Currently it is running in FilterPump/Pool mode .

niharmehta, 2019-05-21:
After some further testing, it seems like the errors are also happening when not explicitly sending a command either. Every so often when I  check Status on the iAqualink, I see it showing the pump is either offline or Watts are way out of range.  Definitely repeatable when sending a command (including just as I stop the service) , but also sometimes when not explicitly sending a command.

sfeakes, 2019-05-21:
I did look over the logs, and nothing really stood out. The control panel never pinged the address 0x08 which is very odd, I've never seen that before. But I don't *think* that's anything to do with the issue.

If the problem occurs while simply running the serial logger, then that would be indicative of bad wiring or USB2RS485 adapter, (you don't have the ground connected to you, just the two data wires?). But if that was the case, I'd expect to see checksum errors in the logs.  There have been a few cases of strange issues due to wiring, so changing the wiring between the RS485 adapter and where you have connected it on the RS485 bus might be worth a shot.

I really think that the iAqualink device is getting checksum errors when the USB2RS485 is powerd, rather than AqualinkD messing up any of the protocol data. (Say that because the serial logger simply reads data, and you see the problem when running that).  So, I think the following may be your best course of action.
This is all just guessing, slightly informed, but still guess so adjust as you see fit. (try each and retest)

- Look for any error information / logs that you can get to on the iAqualink device.

- Completely power everything off then on again (Control panel / iAqualink / any pumps & heaters on the RS485 bus ).  You'll probably have to do this from the fusebox.

- Change wiring between Control Panel & USB2RS425 adapter, AND between control panel and iAqualink box

- Try a different USB2RS485 adapter & maybe even PSU on Raspberry Pi.

- (may be hard to test with this option).  Disconnect everything from RS485 bus and just connect iAqualink and AqualinkD, preferably right next to control panel on short wires.

niharmehta, 2019-05-21:
I assume that 0x08 is not pinged because there is no RS8 controller on the bus. Only the iaqualink  which is 0x33 or 0xa3? .  The iaqualink is a locked down device.  Other than what it displays, I have no other way of seeing any logs on it. 
I just tested both a swap of the power adapter, and changed the serial adapter as well.  First one was CH340 based and current one is FTDI. I think this would eliminate chip specific driver issues I think. 

I have previously powered off the entire system  (using the breakers) when troubleshooting my other issues. 

The wiring between the iaqualink and the USB adapter is on the same run from the main box  which is about 60ft away. I just joined the D+/- pairs via a wire nut (which I have checked 3 times)  and pass the power pairs to the iaqualink. The pi and the iaqualink are bout 3 inches of wire away from each other.  If the problem is on the wire from the main control box, then I think I would be having greater issues with the iaqualink without the aqualinkd.

It still confuses me how it could be a physical issue if I can stop the problem by stopping the aqualinkd service (leaving pi on and connected) but once we access the serial port, the errors start again.  So I did some additional testing. 

I opened  the transmit jumper on my rs485 adapter, and this stopped the problem from occurring. Adding the jumper back I can recreate the problem. So it definitely seems transmit related on the pi side.  

On another note, it is interesting that with transmit disconnected, I still get updates from the SWG settings in the web UI , but everything else is grey.

sfeakes, 2019-05-22:
The way this RS485 protocol works is the master pings all slave address it can talk to and if it get's a reply it starts communicating. So the fact that there is no RS8 device doesn't make any difference, it should ping that ID. 0x09, 0x08, 0x0a & 0x0b are all RS8 keypad address, it pings 3 of them just not 0x08.  That's what's odd, but again I don't think that has anything to do with the problem.

The reason the SWG get's updated even with transmit disabled is because AqualinkD monitors the communication between the control panel and SWG and reports that information. The control panel does not display 1/2 of the information it get's from the SWG, so I spent a lot of time reversing engineering that protocol to get all the information.  All the other information AqualinkD get's from the RS485 bus (like device status, temperature etc) it needs to reply to a ping, so without transmission it can't get anything else.
(That's what the config item `read_all_devices=yes` does, I intend to do this for all devices in the future). Just set that to no and AqualinkD will only display information from the control panel.

The reason I think it *might* be a physical issue is when you enable transmit you are utilizing power and introducing noise on the RS485 bus.  But I would also expect to see checksum errors on the PI side.  You don't have any termination on your wiring do you? (ie any form of resistors between the data + & - wires).

Would it be easy for you to move the AqualinkD equipment to the control panel and wire it in directly to test?  Just add 2 wires to the terminals on the control panel, no need for T nut or anything.

niharmehta, 2019-05-25:
. I *THINK* it may be resolved based on your recommendation. I Re-stripped and terminated every connection on the control panel side for the SWG, Pump, and iaqualink.  It seems like the Offline/Watt error has gone away for now.  Thank you!

sfeakes, 2019-05-25:
Thanks great news.    
I've just uploaded a new code base, and put in some logging so I can sniff / and work out more of the Protocol for devices I don't own (like Variable Speed Pumps).  Since you have iAqualink & VSP, if you wouldn't mind I'd like to get some logs from your system so I can add support for VSP.  I'm still writing instructions, so don't worry for the moment, I'll update this when I've finished.

niharmehta, 2019-05-25:
Pulled it 15 minutes ago and verified logging.  Happy to help provide logs once you are ready.

closed 2019-08-18 by sfeakes