aqualinkd/AqualinkD #59: Question: Light control?

mikesalz, 2019-05-10:
Hello.  This is a question rather than an issue.  Does the Homebridge-AqualinkD plugin control lights?  And if so, is it simply on/off or is there a way to change light color as well?  Thanks in advance!

sfeakes, 2019-05-10:
Yes it controls the lights, but it's just on / off.  Only the web UI can control the color.  But you can do the manual way of changing color through Homebridge-AqualinkD, ie a series of on/off commands.

closed 2019-05-22 by sfeakes

hollapm, 2019-08-02:
Is there a plan to add multi-light control?  IE for pool lights and spa lights if they are on different switches?

sfeakes, 2019-08-03:
It’s not a feature anyone has asked for until now. I’ll look into it when I get some time.