aqualinkd/AqualinkD #60: Admin panel does not reflect current status.

LogoiLab, 2019-05-10:
The I have set the deck lights and pool lights to on via the Touch Panel that is also wired in. AqualinkD does not display these as on and toggling them does nothing. I have insured a good connection and that everything is working properly I have enabled every button in both the config and the web interface because I thought at first it was a mapping issue. no dice.
[image]

sfeakes, 2019-05-10:
Can you post the log from starting Aqualinkd.  It’s doesn’t look like it sinked with the control panel.

LogoiLab, 2019-05-10:
is that in the debug log? or is it in journalctl?

sfeakes, 2019-05-10:
That’s the right log.  It’s just the part from when you hit some buttons. I’d need to see the log from when AqualinkD actually starts.  Stop AqualinkD, delete the log, start AqualinkD, let it run for a few minutes (no need to do anything in the UI), then post the complete log.   I think you’ll see the problem within the first part of the log when it’s trying to sync with the controller.

LogoiLab, 2019-05-10:
ok cool I'll retry and let you know

LogoiLab, 2019-05-10:
Here is the full fresh log(about 10 mins worth) I see where it gets a PROBE. Control still isn't working.

```
May-10-19 16:28:25 PM Notice: Aqualink Daemon v1.2.6c
May-10-19 16:28:25 PM Notice: Config log_level         = 7
May-10-19 16:28:25 PM Notice: Config socket_port       = 80
May-10-19 16:28:25 PM Notice: Config serial_port       = /dev/ttyUSB0
May-10-19 16:28:25 PM Notice: Config web_directory     = /var/www/aqualinkd/
May-10-19 16:28:25 PM Notice: Config device_id         = 0x0b
May-10-19 16:28:25 PM Notice: Config read_all_devices  = NO
May-10-19 16:28:25 PM Notice: Config override frz prot = NO
May-10-19 16:28:25 PM Notice: Config mqtt_server       = localhost:1883
May-10-19 16:28:25 PM Notice: Config mqtt_dz_sub_topic = (null)
May-10-19 16:28:25 PM Notice: Config mqtt_dz_pub_topic = (null)
May-10-19 16:28:25 PM Notice: Config mqtt_aq_topic     = aqualinkd
May-10-19 16:28:25 PM Notice: Config mqtt_user         = (null)
May-10-19 16:28:25 PM Notice: Config mqtt_passwd       = (null)
May-10-19 16:28:25 PM Notice: Config mqtt_ID           = aqualinkd_b827ebfbe3
May-10-19 16:28:25 PM Notice: Config idx water temp    = 0
May-10-19 16:28:25 PM Notice: Config idx pool temp     = 0
May-10-19 16:28:25 PM Notice: Config idx spa temp      = 0
May-10-19 16:28:25 PM Notice: Config idx SWG Percent   = 0
May-10-19 16:28:25 PM Notice: Config idx SWG PPM       = 0
May-10-19 16:28:25 PM Notice: Config PDA Mode          = NO
May-10-19 16:28:25 PM Notice: Config deamonize         = YES
May-10-19 16:28:25 PM Notice: Config log_file          = /var/log/aqualinkd.log
May-10-19 16:28:25 PM Notice: Config light_pgm_mode    = 0.00
May-10-19 16:28:25 PM Notice: Config BTN Filter_Pump   = label Filter Pump     | PDAlabel FILTER PUMP     | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Spa_Mode      = label Cleaner         | PDAlabel SPA             | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_1         = label Waterfall       | PDAlabel AUX1            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_2         = label Pool Light      | PDAlabel AUX2            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_3         = label Aux_3           | PDAlabel AUX3            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_4         = label Deck Light      | PDAlabel AUX4            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_5         = label Garden Light    | PDAlabel AUX5            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_6         = label Aux_6           | PDAlabel AUX6            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Aux_7         = label NONE            | PDAlabel AUX7            | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Pool_Heater   = label Pool Heater     | PDAlabel POOL HEAT       | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Spa_Heater    = label Spa Heater      | PDAlabel SPA HEAT        | dzidx 0
May-10-19 16:28:25 PM Notice: Config BTN Solar_Heater  = label Solar Heater    | PDAlabel EXTRA AUX       | dzidx 0
May-10-19 16:28:25 PM Debug:  process_id of child process 30902 
May-10-19 16:28:25 PM Notice: Starting web server on port 80
May-10-19 16:28:25 PM Notice: Starting MQTT client to localhost:1883
May-10-19 16:28:25 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
May-10-19 16:28:25 PM Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827ebfbe3
May-10-19 16:28:25 PM Info:   MQTT: Connection acknowledged
May-10-19 16:28:25 PM Info:   MQTT: Subscribing to 'aqualinkd/#'
May-10-19 16:28:25 PM Info:   MQTT: Subscription(s) acknowledged
May-10-19 16:28:25 PM Info:   MQTT: Published id=1: aqualinkd/Alive 1
May-10-19 16:28:25 PM Debug:  MQTT: Ignore aqualinkd/Alive 1
May-10-19 16:28:25 PM Info:   MQTT: Message publishing acknowledged (msg_id: 1)
May-10-19 16:28:25 PM Info:   MQTT: received (msg_id: 1), looks like my own message, ignoring
May-10-19 16:28:25 PM Debug:  MQTT: Ignore aqualinkd/Alive 1
May-10-19 16:28:27 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:28:27 PM Debug:  RS Received PROBE length 7.
May-10-19 16:28:36 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:28:44 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:28:53 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:02 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:11 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:19 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:28 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:37 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:45 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:29:54 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:03 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:12 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:20 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:29 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:38 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:30:47 PM Debug:  RS received packet of type Probe length 7
[... 41 lines cut ...]
May-10-19 16:36:54 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:37:03 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:37:11 PM Debug:  RS received packet of type Probe length 7
[... cut ...]
May-10-19 16:39:23 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:39:31 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:39:41 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:39:49 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:39:58 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:40:07 PM Debug:  RS received packet of type Probe length 7
May-10-19 16:40:15 PM Debug:  RS received packet of type Probe length 7
```

sfeakes, 2019-05-10:
This is odd, it looks like the control center is looking for a device and AqualinkD is responding, but nothing after.  Can you try a different ID,  you have 0x0b set at the moment, can you try 0x0a, 0x08 or 0x09.  One of them may be taken by other devices so run the serial logger to see what’s free.  Information is in the wiki.

Other than that, can you turn on serial debug and post the output from AqualinkD starting again.  It will create a huge file so you may need to zip it and post as attachment.

LogoiLab, 2019-05-10:
Ok will do

LogoiLab, 2019-05-11:
Ok got it functioning. Turns out the ground wire for my system is floating and pulling it down caused the system to ignore the serial to usb device I am using, simply removed the ground line from my serial interface and it started working. Hope this will at least help others with the same issue.

closed 2019-05-11 by LogoiLab