aqualinkd/AqualinkD #61: hourly (exactly) checksum errors

niharmehta, 2019-05-15:
Not sure if this is a bug or not.  However, I have been trying to track this down for days.  Every hour.. right at 48 minutes, I get a burst of checksum errors in the logs.   I have cycled the control center, restarted my pi, blocked iaqualink access, checked pi cron jobs, but without fail at 48 minutes past the hour, I get a chunk of checksum failures.    The only thing that  is maybe odd on the serial captures is possibly the lack of null on the first hex, and the values in the 3rd hex number that seem different than other times.  I am not sure what device is generating the errors as I see both 0x50 and 0x0a in the logs during this time 

Examples of the errors:
```
May 14 16:47:55 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x6c|0x10|0x02|0x50|0x11|0x19|0x8c|0x10|0x03|
May 14 16:48:07 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 16:48:07 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x6e|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 14 16:48:29 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 16:48:29 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x6d|0x10|0x02|0x50|0x11|0x19|0x8c|0x10|0x03|
May 14 16:48:40 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 16:48:40 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x6e|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 14 17:47:50 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 17:47:50 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x73|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 14 17:48:12 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 17:48:12 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x74|0x10|0x02|0x50|0x11|0x14|0x87|0x10|0x03|
May 14 17:48:23 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 17:48:23 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x73|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 14 17:48:34 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 17:48:34 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x74|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 14 17:48:45 aquad aqualinkd: Serial read bad checksum, ignoring
May 14 17:48:45 aquad aqualinkd: Bad receive packet  | HEX: 0x10|0x02|0x73|0x10|0x02|0x50|0x11|0x14|0x87|0x10|0x03|

...
May 7 00:48:53 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 00:48:53 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x40|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 00:49:04 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 00:49:04 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x27|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 01:48:15 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 01:48:15 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x50|0x11|0x0f|0x82|0x10|0x03|
May 7 01:48:26 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 01:48:26 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 01:48:37 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 01:48:37 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 01:48:48 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 01:48:48 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x50|0x11|0x0f|0x82|0x10|0x03|
May 7 01:48:59 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 01:48:59 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 02:48:12 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:48:12 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 02:48:23 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:48:23 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 02:48:33 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:48:33 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 02:48:44 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:48:44 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 02:48:56 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:48:56 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x65|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 02:49:07 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 02:49:07 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x0a|0x02|0x00|0x14|0x00|0x00|0x00|0x32|0x10|0x03|
May 7 03:48:09 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:48:09 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 03:48:20 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:48:20 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 03:48:31 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:48:31 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 03:48:42 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:48:42 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 03:48:53 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:48:53 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 03:49:04 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 03:49:04 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 04:48:19 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 04:48:19 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x4f|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 04:48:30 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 04:48:30 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x4a|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 04:48:41 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 04:48:41 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x32|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 7 04:48:52 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 04:48:52 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x1d|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 05:48:16 aquad aqualinkd: Serial read bad checksum, ignoring
[... 8 lines cut ...]
May 7 05:49:00 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 7 06:48:13 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:48:13 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 06:48:24 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:48:24 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 06:48:35 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:48:35 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x66|0x10|0x02|0x50|0x11|0x00|0x73|0x10|0x03|
May 7 06:48:46 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:48:46 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 7 06:48:57 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:48:57 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x67|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 06:49:08 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 06:49:08 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 7 08:48:15 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 08:48:15 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x50|0x11|0x14|0x87|0x10|0x03|
May 7 08:48:26 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 08:48:26 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|
May 7 08:48:37 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 08:48:37 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 08:48:49 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 08:48:49 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x33|0x30|0x75|0x10|0x03|
May 7 08:48:59 aquad aqualinkd: Serial read bad checksum, ignoring
May 7 08:48:59 aquad aqualinkd: Bad receive packet | HEX: 0x10|0x02|0x68|0x10|0x02|0x0a|0x02|0x00|0x10|0x00|0x00|0x00|0x2e|0x10|0x03|

```

sfeakes, 2019-05-16:
Someone else had / just reported this exact problem, and I believe at the exact same time. (48 min passed the hour). I wonder if you are both on the same version of raspbain and there is something using the IO.  On a raspberry just about everything IO related uses the same chip (network, usb, file, etc), so it can easily become a bottle neck.
http://aqualinkd.freeforums.net/thread/9/bad-packets-checksum-errors-minutes

niharmehta, 2019-05-16:
Same person. I will do some testing to see if i can track it somewhere in the OS. Nothing in syslog or matches any cron jobs.