aqualinkd/AqualinkD #75: Homekit - Salt Level PPM

johnluber, 2019-07-03:
I'm noticing that no matter what the salt level read is from aqualinkd that HomeKit reports a max value of 2700.  For example, the status command from aqualinkd reports my salt level at 3800 (a little high) but HomeKit reports it at 2700.

Is 2700 a hard coded limit or maybe I missed a config change that I need to make?

sfeakes, 2019-07-03:
2700 is not hardcoded, and not a limit.  I've seen similar where for some reason Home app get's stuck, and I've had to close and re-start the app (swipe up in app, then reopen).  But to know for sure look at the mqtt queue to see what was sent.
```mosquitto_sub -t aqualinkd/# -v```

Then look for the values. (these are mine as of today)
```
aqualinkd/SWG/PPM 3200
aqualinkd/SWG/PPM_f 1760
```

PPM is the real value, PPM_f is the value sent to HomeKit. 
PPM_f is a value converted from celsius to fahrenheit, since HomeKit expects every value is will celsius and will automatically convert to fahrenheit if you have your phone set for (US units). There is nothing you can do about that, So if your phone is set for US, AqualinkD converts the value first, only for HomeKit to convert it back again.

johnluber, 2019-07-03:
Interesting…

Here’s the result of running the command you gave showing 2700:

pi@aqualinkd:/etc/mosquitto/conf.d $ mosquitto_sub -t aqualinkd/# -v
aqualinkd/Filter_Pump 1
aqualinkd/Filter_Pump/delay 0
aqualinkd/Spa_Mode 0
aqualinkd/Spa_Mode/delay 0
aqualinkd/Aux_1 1
aqualinkd/Aux_1/delay 0
aqualinkd/Aux_2 0
aqualinkd/Aux_2/delay 0
aqualinkd/Aux_3 0
aqualinkd/Aux_3/delay 0
aqualinkd/Spa_Heater 0
aqualinkd/Spa_Heater/setpoint 37.22
aqualinkd/Spa_Heater/enabled 0
aqualinkd/Aux_4 0
aqualinkd/Aux_4/delay 0
aqualinkd/Aux_5 0
aqualinkd/Aux_5/delay 0
aqualinkd/Pool_Heater 0
aqualinkd/Pool_Heater/setpoint 4.44
aqualinkd/Pool_Heater/enabled 0
aqualinkd/Temperature/Air 35.56
aqualinkd/Temperature/Pool 32.78
aqualinkd/Temperature/Spa -18.00
aqualinkd/Freeze_Protect 0
aqualinkd/Freeze_Protect/setpoint 2.78
aqualinkd/Freeze_Protect/enabled 1
aqualinkd/SWG 0
aqualinkd/SWG/Percent 75
aqualinkd/SWG/PPM 2700.   
aqualinkd/SWG/enabeled 0
aqualinkd/SWG/Percent_f 23.888889
aqualinkd/SWG/enabled 0
aqualinkd/SWG/setpoint 23.888889
aqualinkd/SWG/PPM_f 1482
aqualinkd/SWG/fullstatus 255
aqualinkd/Aux_6 0
aqualinkd/Aux_6/delay 0
aqualinkd/Aux_7 0
aqualinkd/Aux_7/delay 0
aqualinkd/Solar_Heater 0
aqualinkd/Solar_Heater/delay 0
aqualinkd/Alive 1
aqualinkd/Battery 1
aqualinkd/Service_Mode 0

And, here’s the http command status output showing 3800:

{"type": "status","status":"Ready","version":"","aqualinkd_version":"1.3.3d","date":"07/03/19 Wed","time":"2:03 PM","pool_htr_set_pnt":"-999","spa_htr_set_pnt":"-999","frz_protect_set_pnt":"-999","air_temp":"96","pool_temp":"91","spa_temp":" ","swg_percent":"101","swg_ppm":"3800","temp_units":"f","battery":"ok","leds":{"Filter_Pump": "on","Spa_Mode": "off","Aux_1": "on","Aux_2": "off","Aux_3": "off","Aux_4": "off","Aux_5": "off","Aux_6": "off","Aux_7": "off","Pool_Heater": "off","Spa_Heater": "off","Solar_Heater": "off", "SWG": "off"}}

johnluber, 2019-07-03:
I did try restarting aqualinkd, homebridge, and the home app and go the same result.

sfeakes, 2019-07-03:
If you look at other values they are also out between web and mqtt (swg % and temperature). So looks like MQTT is not being updated.  You may want to check the mqtt config, of send debug log

johnluber, 2019-07-03:
I definitely see what you're saying.  It does seem to be at the mqtt  level on the pi.  I see proper values on the web interface and via the status command.

I've setup exactly as described in the install except that I set an IP address instead of loopback in Aqualinkd.conf.  Are we literally supposed to use someusername and  somepassword?  Just want to be sure.

If it helps my config is aqualinkd, mqtt and homebride with aqualinkd.  But, as you said, it seems to be at the mqtt level. So, I need to solve that first.  

The mosquitto.conf file is unmodified from the mosquitto install.  Is that correct?

At one point I had homekit2mosqitto installed but I have removed that.  Perhaps there's something lingering there?

Thanks for helping me sort through this...

sfeakes, 2019-07-04:
That all sounds accurate.  No need to change mosquito config, then just make sure that AqualinkD points to that server.  User and passwd for MQTT are optional, in both AqualinkD and homebridge-aqualinkd configs. Since AqualinkD is not posting to MQTT I think that’s the best place to start, if you want to post /etc/AqualinkD.conf I’ll take a look.
But make sure the following two are set and not commented out

mqtt_address = somehost:port
mqtt_aq_topic = aqualinkd

johnluber, 2019-07-04:
This is perplexing...  This morning I grabbed the aqualinkd.conf off of GitHub and installed it on my pi tweaking just a minimum of settings.  And, then I purged and reinstalled mosquitto.  Still getting the same result.

Here's aqualinkd.conf:

# aqualinkd.conf
#

# The directory where the web files are stored

web_directory=/var/www/aqualinkd/

# Log to file, comment out if you do not want to log to file
#log_file=/var/log/aqualinkd.log

# The log level. [DEBUG_DERIAL, DEBUG, INFO, NOTICE, WARNING, ERROR]
# Pick the highest level, and all levels below will be sent to syslog.
# your syslog settings may be set to only display messages above a certian level
# in which case make sure you use the log_file settings to capture everything
# you want when debugging
# so, NOTICE also prints WARNING & ERROR
#     DEBUG_SERIAL would print everything possible

log_level=DEBUG
#log_level=INFO
#log_level=NOTICE

# The socket port that the daemon listens to
# If you change this from 80, remember to update aqualink.service.avahi
socket_port=80

# The serial port the daemon access to read the Aqualink RS8
serial_port=/dev/ttyUSB0

# If equiptment is in freeze protect mode some commands like pump_off / spa_on are
# ignored.  You can force these to work by setting the below.
override_freeze_protect = no

# Confert Deg F to Deg C when posting to Domoticz or MQTT.
convert_mqtt_temp_to_c = yes
convert_dz_temp_to_c = yes

# default is to use pool water temp as spa water temp when spa is off (and there for not able to report water temp)
# enable below to report 0 as the spa temp when spa is off.
# This is for MQTT cnnections only, WEB socket and WEB API always report TEMP_UNKNOWN (-999) allowing the consumer to
# decide how to report.
report_zero_spa_temp = no

# default is to not report changes to pool temp when the filter pump is off or in spa mode
# enable below to report 0 as the pool temp when the filter pump is off or when in spa mode.
# This is for MQTT cnnections only, WEB socket and WEB API always report TEMP_UNKNOWN (-999) allowing the consumer to
# decide how to report.
report_zero_pool_temp = no

# mqtt stuff
mqtt_address = localhost:1883
mqtt_user = someusername
mqtt_passwd = somepassword
#mqtt_dz_pub_topic = domoticz/in
#mqtt_dz_sub_topic = domoticz/out
mqtt_aq_topic = aqualinkd

# The id of the Aqualink terminal device. Devices probed by RS8 master are:
# 08-0b, 10-13, 18-1b, 20-23, 28-2b, 30-33, 38-3b, 40-43
# Working RS ID's are 0x0a 0x0b 0x09 0x08    <- 0x08 is usually taken
# You can use 0x00 and AqualinkD will find an ID for you, but this makes a slow startup
device_id=0x00

# Please see forum for this, only set to yes when logging information to support
# new devices. Inflrmation will be written to /tmp/RS485.log
#debug_RSProtocol_packets = no
[... 110 lines cut ...]

And, here's the mosquitto log showing socket errors:

1562242997: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1562242997: Config loaded from /etc/mosquitto/mosquitto.conf.
1562242997: Opening ipv4 listen socket on port 1883.
1562242997: Opening ipv6 listen socket on port 1883.
1562243001: New connection from 10.10.30.2 on port 1883.
1562243001: New client connected from 10.10.30.2 as mqttjs_0d9f8cd1 (c1, k60, u'someusername').
1562243003: New connection from 10.10.30.5 on port 1883.
1562243003: New client connected from 10.10.30.5 as aqualinkd_b827ebbb (c1, k5, u'someusername').
1562243041: New connection from ::1 on port 1883.
1562243041: New client connected from ::1 as mosqsub/1268-aqualinkd (c1, k60).
1562243051: Socket error on client mosqsub/1268-aqualinkd, disconnecting.
1562243637: Socket error on client aqualinkd_b827ebbb, disconnecting.
1562243637: New connection from 127.0.0.1 on port 1883.
1562243637: New client connected from 127.0.0.1 as aqualinkd_b827ebbb (c1, k5, u'someusername').
1562243741: New connection from ::1 on port 1883.
1562243741: New client connected from ::1 as mosqsub/1532-aqualinkd (c1, k60).
1562243836: Socket error on client mosqsub/1532-aqualinkd, disconnecting.
1562243839: New connection from ::1 on port 1883.
1562243839: New client connected from ::1 as mosqsub/1554-aqualinkd (c1, k60).
1562243923: Socket error on client mqttjs_0d9f8cd1, disconnecting.
1562243927: Socket error on client mosqsub/1554-aqualinkd, disconnecting.
1562243930: New connection from 10.10.30.2 on port 1883.

sfeakes, 2019-07-04:
You have MQTT user and password set incorrectly.
Either comment out the lines, or change them to your actual MQTT user and password. Default mosquitto Install has no MQTT user or password, so you probably just put a # in front of the user and password lines.

johnluber, 2019-07-04:
I previously had it commented and had just uncommitted to see if it made a difference.  Recommended them and still no change. Output from the command is:

mosquitto_sub -t aqualinkd/# -v
aqualinkd/Alive 1
aqualinkd/Temperature/Pool 31.67
aqualinkd/Temperature/Air 25.00
aqualinkd/Freeze_Protect 0
aqualinkd/Freeze_Protect/setpoint 2.78
aqualinkd/Freeze_Protect/enabled 1
aqualinkd/Battery 1
aqualinkd/SWG 0
aqualinkd/SWG/fullstatus 255
aqualinkd/SWG/enabled 0
aqualinkd/Filter_Pump 1
aqualinkd/Filter_Pump/delay 0
aqualinkd/Spa_Mode 0
aqualinkd/Spa_Mode/delay 0
aqualinkd/Aux_1 0
aqualinkd/Aux_1/delay 0
aqualinkd/Aux_2 0
aqualinkd/Aux_2/delay 0
aqualinkd/Aux_3 0
aqualinkd/Aux_3/delay 0
aqualinkd/Aux_4 0
aqualinkd/Aux_4/delay 0
aqualinkd/Aux_5 0
aqualinkd/Aux_5/delay 0
aqualinkd/Aux_6 0
aqualinkd/Aux_6/delay 0
aqualinkd/Aux_7 0
aqualinkd/Aux_7/delay 0
aqualinkd/Pool_Heater 0
aqualinkd/Pool_Heater/enabled 0
aqualinkd/Pool_Heater/setpoint 4.44
aqualinkd/Spa_Heater 0
aqualinkd/Spa_Heater/enabled 0
aqualinkd/Spa_Heater/setpoint 37.22
aqualinkd/Solar_Heater 0
aqualinkd/Solar_Heater/delay 0
aqualinkd/Service_Mode 0
aqualinkd/Temperature/Air 31.67

johnluber, 2019-07-06:
Starting to believe there's a bug here somewhere...

I nuked my aqualinkd build completely.  Reimaged the card, installed git and aqualinkd, mosquitto and configured.  Web UI looks fine.  Nuked my homebridge config as well - a real pain with 53 devices.  It showed the proper value for PPM as well; until this morning.  Now, homebridge still shows the value from last night -- 3700 -- whereas the web UI shows 3600.

Seems like maybe there's an issue in homebridge-aqualinkd updating the PPM values?

sfeakes, 2019-07-06:
If you turn a device on/off from the web ui does the state reflect correctly in HomeKit?

If you restart AqualinkD does the ppm reflect correctly in HomeKit?

What is ‘read_all_devices’ set to in the AqualinkD config?  If no try setting to yes.

If not, Can you set AqualinkD to debug and post output (obviously make sure SWG is running).

johnluber, 2019-07-06:
Yes, I can turn devices on/off from the web UI and the status is reflected correctly in HomeKit.  I do have a couple of label mismatches but that’s on me to sort out.  I usually don’t use the web UI so not too worried about that.

I restarted aqualinkd and ppm is still incorrect in HomeKit.  Aqualinkd says 3600 and HomeKit says 3700.

Read_all_devices was already at ‘yes’

Here are the logs:

Jul-06-19 09:38:11 AM Debug:  RS received packet of type Lng Message length 24
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Lng Message length 24
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Message length 24
Jul-06-19 09:38:12 AM Info:   RS Message :- 'Maintain Temp is off'
Jul-06-19 09:38:12 AM Debug:  Kicking thread 0,0xd36100
Jul-06-19 09:38:12 AM Info:   RS Message :- '9:37 AM'
Jul-06-19 09:38:12 AM Debug:  Programming mode: loop 4 of 5 looking for 'MAINTAIN TEMP IS' received message '9:37 AM'
Jul-06-19 09:38:12 AM Debug:  RS time is accurate '9:37 AM '
Jul-06-19 09:38:12 AM Debug:  Kicking thread 0,0xd36100
Jul-06-19 09:38:12 AM Debug:  Programming mode: loop 5 of 5 looking for 'MAINTAIN TEMP IS' received message '9:37 AM'
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:12 AM Debug:  RS Received STATUS length 12.
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:12 AM Debug:  Thread 1,0xd38390 sleeping, waiting for thread 0,0xd36100 to finish
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:12 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Message length 24
Jul-06-19 09:38:13 AM Info:   RS Message :- 'Air Temp 81�F'
Jul-06-19 09:38:13 AM Debug:  Kicking thread 0,0xd36100
Jul-06-19 09:38:13 AM Debug:  Programming mode: did not find 'MAINTAIN TEMP IS'
Jul-06-19 09:38:13 AM Info:   MQTT: Published id=36: aqualinkd/Temperature/Air 27.22
Jul-06-19 09:38:13 AM Debug:  Thread 0,0x76d7c460 finished
Jul-06-19 09:38:13 AM Info:   MQTT: Message publishing acknowledged (msg_id: 36)
Jul-06-19 09:38:13 AM Info:   MQTT: received (msg_id: 36), looks like my own message, ignoring
Jul-06-19 09:38:13 AM Debug:  MQTT: Ignore aqualinkd/Temperature/Air 27.22
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:13 AM Debug:  RS Received STATUS length 12.
Jul-06-19 09:38:13 AM Debug:  Thread 1,0xd38390 sleeping, waiting for thread 0,0xd36100 to finish
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:13 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:14 AM Info:   URI request: '/ '
Jul-06-19 09:38:14 AM Debug:  Served WEB request
Jul-06-19 09:38:14 AM Debug:  -- Websocket left
Jul-06-19 09:38:14 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:14 AM Debug:  ++ Websocket joined
Jul-06-19 09:38:14 AM Info:   WS: Message - Key 'command' Value 'GET_DEVICES' | Key2 '(null)' Value2 '(null)'
Jul-06-19 09:38:14 AM Debug:  WEB: homebridge used 2090 of 3000
Jul-06-19 09:38:14 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:14 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:14 AM Debug:  Thread 1,0xd38390 is active
Jul-06-19 09:38:14 AM Notice: Getting freeze protection setpoints
Jul-06-19 09:38:14 AM Info:   Queue send '0x09' to controller (programming)
Jul-06-19 09:38:14 AM Debug:  waitForMessage PRESS ENTER* TO SELECT 5
Jul-06-19 09:38:14 AM Debug:  Programming mode: loop 1 of 5 looking for 'PRESS ENTER* TO SELECT' received message 'Air Temp 81�F'
Jul-06-19 09:38:14 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:38:14 AM Debug:  RS received packet of type Message length 24
Jul-06-19 09:38:14 AM Debug:  Kicking thread 1,0xd38390
Jul-06-19 09:38:14 AM Debug:  Programming mode: loop 2 of 5 looking for 'PRESS ENTER* TO SELECT' received message 'Air Temp 81�F'
Jul-06-19 09:38:15 AM Debug:  RS received packet of type Lng Message length 24
Jul-06-19 09:38:15 AM Debug:  RS received packet of type Lng Message length 24
Jul-06-19 09:38:16 AM Debug:  RS received packet of type Lng Message length 24
Jul-06-19 09:38:16 AM Info:   RS Message :- 'Press ~ for more options... Press Enter* to select'
Jul-06-19 09:38:16 AM Debug:  Kicking thread 1,0xd38390
Jul-06-19 09:38:16 AM Debug:  Programming mode: loop 3 of 5 looking for 'PRESS ENTER* TO SELECT' received message 'Press ~ for more options... Press Enter* to select'
Jul-06-19 09:38:16 AM Debug:  Programming mode: String MATCH
Jul-06-19 09:38:16 AM Debug:  Programming mode: found message 'PRESS ENTER* TO SELECT' in 'Press ~ for more options... Press Enter* to select'
Jul-06-19 09:38:16 AM Info:   Queue send '0x18' to controller (programming)
Jul-06-19 09:38:16 AM Debug:  waitForMessage (null) 1
Jul-06-19 09:38:16 AM Debug:  Programming mode: loop 1 of 1 waiting for next message, received 'Press ~ for more options... Press Enter* to select'
Jul-06-19 09:38:16 AM Debug:  RS received packet of type Lng Message length 24
[... 781 lines cut ...]
Jul-06-19 09:39:14 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:14 AM Debug:  RS received packet of type Status length 12
[... cut ...]
Jul-06-19 09:39:16 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:16 AM Debug:  RS Received STATUS length 12.
Jul-06-19 09:39:16 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:17 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:17 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:17 AM Debug:  RS received packet of type Status length 12
Jul-06-19 09:39:17 AM Debug:  RS received packet of type Status length 12
Jul-06-19

sfeakes, 2019-07-06:
Can you post the complete first 2~5 mins of the debug log file.  Also if you monitor the mqtt messages which aqualinkd is starting, you can see what it's posting to HomeKit.  Just start the following command before aqualinkd, and leave it running while you start aqualinkd. 
```mosquitto_sub -t aqualinkd/# -v```

johnluber, 2019-07-06:
Absolutely, here's 5+ minutes of the debug log:

*pi@aqualinkd*:*/var/log $* cat aqualinkd.log

Jul-06-19 16:42:06 PM Notice: Aqualink Daemon v1.3.3d

Jul-06-19 16:42:06 PM Notice: Config log_level         = 7

Jul-06-19 16:42:06 PM Notice: Config socket_port       = 80

Jul-06-19 16:42:06 PM Notice: Config serial_port       = /dev/ttyUSB0

Jul-06-19 16:42:06 PM Notice: Config web_directory     = /var/www/aqualinkd/

Jul-06-19 16:42:06 PM Notice: Config device_id         = 0x00

Jul-06-19 16:42:06 PM Notice: Config read_all_devices  = YES

Jul-06-19 16:42:06 PM Notice: Config use_aux_labels    = NO

Jul-06-19 16:42:06 PM Notice: Config override frz prot = NO

Jul-06-19 16:42:06 PM Notice: Config mqtt_server       = localhost:1883

Jul-06-19 16:42:06 PM Notice: Config mqtt_dz_sub_topic = (null)

Jul-06-19 16:42:06 PM Notice: Config mqtt_dz_pub_topic = (null)

Jul-06-19 16:42:06 PM Notice: Config mqtt_aq_topic     = aqualinkd

Jul-06-19 16:42:06 PM Notice: Config mqtt_user         = (null)

Jul-06-19 16:42:06 PM Notice: Config mqtt_passwd       = (null)

Jul-06-19 16:42:06 PM Notice: Config mqtt_ID           = aqualinkd_b827ebbb

Jul-06-19 16:42:06 PM Notice: Config idx water temp    = 0

Jul-06-19 16:42:06 PM Notice: Config idx pool temp     = 0

Jul-06-19 16:42:06 PM Notice: Config idx spa temp      = 0

Jul-06-19 16:42:06 PM Notice: Config idx SWG Percent   = 0

Jul-06-19 16:42:06 PM Notice: Config idx SWG PPM       = 0

Jul-06-19 16:42:06 PM Notice: Config PDA Mode          = NO

Jul-06-19 16:42:06 PM Notice: Config PDA Sleep Mode    = NO

Jul-06-19 16:42:06 PM Notice: Config force SWG         = YES

Jul-06-19 16:42:06 PM Notice: Config deamonize         = YES

Jul-06-19 16:42:06 PM Notice: Config log_file          =
/var/log/aqualinkd.log

Jul-06-19 16:42:06 PM Notice: Config light_pgm_mode    = 0.00

Jul-06-19 16:42:06 PM Notice: Debug RS485 protocol     = NO

Jul-06-19 16:42:06 PM Notice: Read Pentair Packets     = NO

Jul-06-19 16:42:06 PM Notice: Config BTN Filter_Pump   = label Filter Pump
    | VSP ID None  | PDAlabel FILTER PUMP     | dzidx 0

Jul-06-19 16:42:06 PM Notice: Config BTN Spa_Mode      = label Spa Mode
    | VSP ID None  | PDAlabel SPA             | dzidx 0

Jul-06-19 16:42:06 PM Notice: Config BTN Aux_1         = label Cleaner
[... 1848 lines cut ...]
Jul-06-19 16:43:31 PM Debug:  RS Received STATUS length 12.

Jul-06-19 16:43:31 PM Debug:  RS received packet of type Message length 24

Jul-06-19 16:43:31 PM Info:   RS Message :- 'Hour 4 AM    ~*'

Jul-06-19 16:43:31 PM Debug:  Kicking thread 2,0xf49100

Jul-06-19 16:43:31 PM Debug:  Find item in Menu: loop 12 of 25 looking for
'HOUR 4 PM' received message 'Hour 4 AM    ~*'

Jul-06-19 16:43:31 PM Info:   Queue send '0x18' to controller (programming)

Jul-06-19 16:43:31 PM Debug:  waitForMessage (null) 1

Jul-06-19 16:43:31 PM Debug:  Programming mode: loop 1 of 1 waiting for
next message, received 'Hour 4 AM    ~*'

Jul-06-19 16:43:31 PM Debug:  RS received packet of type Status length 12

Jul-06-19 16:43:31 PM Debug:  RS Received STATUS length 12.

Jul-06-19 16:43:31 PM Debug:  RS received packet of type Message length 24

Jul-06-19

sfeakes, 2019-07-07:
Thanks.  Sorry should have said, can you do it with the SWG running, that way I can see the messages.

johnluber, 2019-07-07:
I was under the impression that my SWG is running if the pump is on but maybe I misunderstood.  In any case, I've re-run the test you suggested with the SWG in boost mode.  Here are the results:

Jul-07-19 14:26:11 PM Notice: Aqualink Daemon v1.3.3d
Jul-07-19 14:26:11 PM Notice: Config log_level         = 7
Jul-07-19 14:26:11 PM Notice: Config socket_port       = 80
Jul-07-19 14:26:11 PM Notice: Config serial_port       = /dev/ttyUSB0
Jul-07-19 14:26:11 PM Notice: Config web_directory     = /var/www/aqualinkd/
Jul-07-19 14:26:11 PM Notice: Config device_id         = 0x00
Jul-07-19 14:26:11 PM Notice: Config read_all_devices  = YES
Jul-07-19 14:26:11 PM Notice: Config use_aux_labels    = NO
Jul-07-19 14:26:11 PM Notice: Config override frz prot = NO
Jul-07-19 14:26:11 PM Notice: Config mqtt_server       = localhost:1883
Jul-07-19 14:26:11 PM Notice: Config mqtt_dz_sub_topic = (null)
Jul-07-19 14:26:11 PM Notice: Config mqtt_dz_pub_topic = (null)
Jul-07-19 14:26:11 PM Notice: Config mqtt_aq_topic     = aqualinkd
Jul-07-19 14:26:11 PM Notice: Config mqtt_user         = (null)
Jul-07-19 14:26:11 PM Notice: Config mqtt_passwd       = (null)
Jul-07-19 14:26:11 PM Notice: Config mqtt_ID           = aqualinkd_b827ebbb
Jul-07-19 14:26:11 PM Notice: Config idx water temp    = 0
Jul-07-19 14:26:11 PM Notice: Config idx pool temp     = 0
Jul-07-19 14:26:11 PM Notice: Config idx spa temp      = 0
Jul-07-19 14:26:11 PM Notice: Config idx SWG Percent   = 0
Jul-07-19 14:26:11 PM Notice: Config idx SWG PPM       = 0
Jul-07-19 14:26:11 PM Notice: Config PDA Mode          = NO
Jul-07-19 14:26:11 PM Notice: Config PDA Sleep Mode    = NO
Jul-07-19 14:26:11 PM Notice: Config force SWG         = YES
Jul-07-19 14:26:11 PM Notice: Config deamonize         = YES
Jul-07-19 14:26:11 PM Notice: Config log_file          = /var/log/aqualinkd.log
Jul-07-19 14:26:11 PM Notice: Config light_pgm_mode    = 0.00
Jul-07-19 14:26:11 PM Notice: Debug RS485 protocol     = NO
Jul-07-19 14:26:11 PM Notice: Read Pentair Packets     = NO
Jul-07-19 14:26:11 PM Notice: Config BTN Filter_Pump   = label Filter Pump     | VSP ID None  | PDAlabel FILTER PUMP     | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Spa_Mode      = label Spa Mode        | VSP ID None  | PDAlabel SPA             | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_1         = label Cleaner         | VSP ID None  | PDAlabel AUX1            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_2         = label Air Blower      | VSP ID None  | PDAlabel AUX2            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_3         = label Pool Lights     | VSP ID None  | PDAlabel AUX3            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_4         = label Pool Light      | VSP ID None  | PDAlabel AUX4            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_5         = label Spa Light       | VSP ID None  | PDAlabel AUX5            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_6         = label NONE            | VSP ID None  | PDAlabel AUX6            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Aux_7         = label NONE            | VSP ID None  | PDAlabel AUX7            | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Pool_Heater   = label Pool Heater     | VSP ID None  | PDAlabel POOL HEAT       | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Spa_Heater    = label Spa Heater      | VSP ID None  | PDAlabel SPA HEAT        | dzidx 0
Jul-07-19 14:26:11 PM Notice: Config BTN Solar_Heater  = label Solar Heater    | VSP ID None  | PDAlabel EXTRA AUX       | dzidx 0
Jul-07-19 14:26:11 PM Debug:  process_id of child process 9953
Jul-07-19 14:26:11 PM Notice: Starting web server on port 80
Jul-07-19 14:26:11 PM Notice: Starting MQTT client to localhost:1883
Jul-07-19 14:26:11 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Jul-07-19 14:26:11 PM Notice: Searching for valid ID, please configure one for faster startup
Jul-07-19 14:26:11 PM Info:   MQTT: Subscribing mqtt with id of: aqualinkd_b827ebbb
Jul-07-19 14:26:11 PM Info:   MQTT: Connection acknowledged
Jul-07-19 14:26:11 PM Info:   MQTT: Subscribing to 'aqualinkd/#'
Jul-07-19 14:26:11 PM Info:   MQTT: Subscription(s) acknowledged
Jul-07-19 14:26:11 PM Info:   MQTT: Published id=1: aqualinkd/Alive 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Alive 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Service_Mode 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Freeze_Protect 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Freeze_Protect/setpoint 2.78
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Freeze_Protect/enabled 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Battery 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/SWG 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/SWG/fullstatus 255
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/SWG/enabled 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Filter_Pump 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Filter_Pump/delay 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Spa_Mode 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Spa_Mode/delay 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Aux_1 1
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Aux_1/delay 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Aux_2 0
Jul-07-19 14:26:11 PM Debug:  MQTT: Ignore aqualinkd/Aux_2/delay 0
[... 2574 lines cut ...]
[... cut ...]
Jul-07-19 14:32:00 PM Debug:  RS received packet of type Status length 12
Jul-07-19 14:32:00 PM Debug:  RS received packet of type Status length 12
Jul-07-19 14:32:00 PM Debug:  RS received packet of type Message length 24
Jul-07-19 14:32:01 PM Debug:  RS received packet of type Lng Message length 24
Jul-07-19 14:32:01 PM Debug:  RS received packet of type Lng Message length 24
Jul-07-19 14:32:01 PM Debug:  RS received packet of type Message length 24
Jul-07-19 14:32:01 PM Info:   RS Message :- 'Boost Pool 01:49 remaining'
J

sfeakes, 2019-07-14:
Think I've found the problem, just not sure why it would happen on your setup.  It looks like AqualinkD is not reading the SWG information directly from your SWG generator anymore and relying on the status messages from the control panel.
Can you run the serial_logger when the pump is on, and post the output, I'd like to see what device ID your SWG in at. 

I've posted an update to AqualinkD that I hope will fix the issue for you.  
Also you may want to change the device ID in aqualinkd config, that will enable a faster startup time.

Change from 
device_id=0x00
to
device_id=0x09

johnluber, 2019-07-15:
xThanks for looking at this.  I’ve installed the latest and have it running.  Unfortunately the current reading fro salt is 3700 which happens to be the value that HomeKit has been “stuck” at so I’ll have to monitor it for a bit to see if the HomeKit display changes.

I set the AqualinkD to x09 as you suggested and ran the serial logger.  Here’s the output:

Notice: Jandy ID's found
Notice: ID 0x60 is not used  <-- can use for Aqualinkd (PDA mode only)
Notice: ID 0x50 is in use  <-- Salt Water Generator (Aquarite mode)
Notice: ID 0x33 is in use  <-- Aqualink (iAqualink?)
Notice: ID 0xa3 is in use  <-- Unknown Device
Notice: ID 0x78 is in use  <-- Unknown Device
Notice: ID 0x08 is not used  <-- can use for Aqualinkd
Notice: ID 0x09 is not used  <-- can use for Aqualinkd
Notice: ID 0x0a is not used  <-- can use for Aqualinkd
Notice: ID 0x0b is not used  <-- can use for Aqualinkd

Thanks again,

John

sfeakes, 2019-07-15:
Not related to SWG, but one thing I noticed from your config and serial information is you have a Pantair VSP that's not configured.  You may also want to add the following to your config, then you will see RPM / WATT information in MQTT and WebUI.
```
read_pentair_packets = yes
button_01_pumpID=0x78
```

Place the button_01_pumpID near the end of the config file where all the other buttons are listed.

johnluber, 2019-07-15:
Ah, ok I see you deleted the message.  I made the changes but didn't see nay difference in the output. 

I do have a Jandy VSP but hadn't really focused on getting it configured with Aqualinkd.  I do think I saw at one point you were asking for testers and at the time I wasn't in a position to do that.  That said, I am bale to now and would be happy to assist. 

BTW, I have seen t least once now where HomeKit displayed something other 3700 so it seems like your fix is working.

sfeakes, 2019-07-16:
Sorry about that, I made a mistake and deleted the post.

I would love to see your VSP information, and also need to get to the bottom of why AqualinkD isn't reading the status directly from your SWG anymore.  If you could create a serial log file and post that, it would help me greatly with both.

If you follow the below, it should give me the info needed.
1) Turn on pump, (SWG should also be on), make note of RPM and WATTS
2) stop aqualinkd
 ```
      sudo systemctl stop aqualinkd
```
3) check the below lines are present / set in /etc/aqualinkd.conf
```
      read_all_devices = yes
      read_pentair_packets = yes
```
3) start aqualinkd in serial log mode
```
      sudo ./release/aqualinkd -d -c  /etc/aqualinkd.conf -rsd
```
4) let it run for ~30 seconds,  change the pump speed a few times wait ~10 or so seconds between each change (please make note of speed).  This will write to a file `/tmp/RS485.log`
5) stop aqualinkd with "ctrl-C"

Please zip & post file in `/tmp/RS485.log` and post the RPM / WATT / GPH (if shown) information.

I'm hoping that I should be able to see all the information I need in that log file.

johnluber, 2019-07-17:
It definitely looks as though your change is working.  I see the status of the salt fluctuating +/- 100 throughout the day and it is being reflected consistently in the web UI and HomeKit.  So, good there.  Thanks for this!

I'll get you the VSP info but it may be the weekend before I can do it.  Work / life has been a bit hectic the last week or so.

closed 2019-10-24 by sfeakes