aqualinkd/AqualinkD #78: HASSIO.Implementation.txt - updates required?

hollapm, 2019-08-07:
I believe there are a few challenges with the Hassio Implementation file:

Honeywell information should be removed?  
- platform: honeywell
    username: brian_sheedy@symantec.com
    password: !secret honeywell_password
    scan_interval: 2400
    region: us

modes under climate, - error in hassio
ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 0
    modes:
      - 0
      - 1

I believe this may need to be
    modes:
      - "off"
      - "heat"

Note I am still trying to figure out something, but MQTT in hassio does not appear to be working.  It's likely something I am doing, but I am having challenges with MQTT working in home assistant calling aqualinkd topics.

I can test via MQTT and put in the topic and it works, but the items listed under pool don't do anything when I try to change values.

Note Temp displays show, and when a device is on, the lightning bolt is lit, just unable to change status on/off

hollapm, 2019-08-07:
Also Light: needs to be light:

bsheedy, 2019-08-07:
Hello, 

I am the original contributor of these files, and I submitted them the day I implemented AqualinkD - a lot has changed since then.  Here are the corrected configuration.yaml entries and an illustration of what it looks like in HASS.IO.

[image]
[HASSIO.Implementation.txt](https://github.com/sfeakes/AqualinkD/files/3476977/HASSIO.Implementation.txt)

Thanks!

bsheedy, 2019-08-07:
Shaun, 

Would you mind replacing the example file and illustration in your project with the ones above?

Much appreciated.

hollapm, 2019-08-07:
That appears to have fixed it. 

My Temp control pictures don't look like yours but the commands are working!

[image]
I'll play with it late on.

bsheedy, 2019-08-07:
Hello,

The reason your Temp Control doesn't look the same is that I have switched my Home Assistant to the Lovelace interface.  You can change it from the states view by simply going to Developer options and switching.

[image]

bsheedy, 2019-08-07:
Glad to see you are up and running!  Now... spread the word to as many Aqualink RS owners so they don't go down the path of accidentally spending $2000 for a proprietary iAqualink solution that doesn't integrate with 3rd party Smart Home apps!

hollapm, 2019-08-08:
LOL, I have iaqualink but as you said its proprietary and I am not able to control with google home.  Im getting closer to that ;-)
trying to set up lets_encrypt now - running into challenges but being so new I'm learning as I go, then work on the google home integration ;-)
starting to see some light

sfeakes, 2019-08-09:
@bsheedy Thanks for the file update and replying to this post.  I'll add the new config in the next aqualinkd update.

closed 2019-08-10 by hollapm

hollapm, 2019-08-11:
bsheedy,I wanted to see if the light control works for you in hassio?
For some reason my lights are the only items I am unable to control (I have a pool and spa light).  When I connect to hassio andI try and turn on any light, it goes off.  If I go into the web interface and make a change (the standard aqualinkd interface, the controls work and hassio will show an update.
I tried updating the light command payload_on 1 to "ON" and payload_off from 1 to "OFF" and that didn't work.
light:
  - platform: mqtt    name: "Spa Light"    state_topic: "aqualinkd/Aux_4"    command_topic: "aqualinkd/Aux_4/set"    qos: 1    payload_on: "1"    payload_off: "0"    retain: false  - platform: mqtt    name: "Pool Light"    state_topic: "aqualinkd/Aux_5"    command_topic: "aqualinkd/Aux_5/set"    qos: 1    payload_on: "1"    payload_off: "0"    retain: false
Wondering what I may be doing wrong?

scroundy, 2019-08-12:
Lights work for me through homeassistant.  Here is my config (my pool light
is hooked up to my "Solar_Heater" topic.

First I define the lights:

  - platform: mqtt
    name: "Spa Light"
    state_topic: "aqualinkd/Aux_3"
    command_topic: "aqualinkd/Aux_3/set"
    qos: 1
    payload_on: "1"
    payload_off: "0"
    retain: false
  - platform: mqtt
    name: "Pool Light"
    state_topic: "aqualinkd/Solar_Heater"
    command_topic: "aqualinkd/Solar_Heater/set"
    qos: 1
    payload_on: "1"
    payload_off: "0"
    retain: false

Then I have them in a group. I have these in separate included files in my
config:

  PoolLightGroup:
    name: Pool Lights
    view: no
    entities:
      - light.Pool_Light
      - light.Spa_Light

hollapm, 2019-08-13:
Home assistant works for me, but I cant get home assistant and smartthings to see the aqualinkd items yet.