aqualinkd/AqualinkD #85: Salt Water Cell - changes from set level to 0% and back

hollapm, 2019-08-10:
I'm running into a challenge where the SWG changes from 0% to 75% and back.

I have the system set to 75%.  The system sets to 0% for about 10 seconds, then goes to 75% (which it should be) for a short time and then back to 0%.

I notice this as my aquapure keeps beeping at me letting me know there is a change.

What makes this more interesting is it was going 0 - > 75 -> 0 for a few hours, then it stopped and its staying at 75.  Only think I did was open aqualinkd website which initially showed the % change, then stopped showing changes after a few mins.  I don't know if this is in relation or not.

Let me know what I can send to help out.

scroundy, 2019-08-10:
I have the same problem. I gave up on it but would love to see this fixed.  Let me know any logs you want to look at, or access even.

sfeakes, 2019-08-10:
I've been away for a few weeks, It's on my list of things to look at.   Have either of you tried setting `read_all_devices` to the opposite of the current setting. ie  `read_all_devices = yes` and `read_all_devices=no` to see if it made any differance?

scroundy, 2019-08-11:
Yes. I've tried both and it does make a difference.

hollapm, 2019-08-11:
It does or does not?
I've only had mine set as 
I have mine set as read_all_devices = yes

scroundy, 2019-08-11:
Thank you for asking. It does NOT make a difference.

sfeakes, 2019-08-11:
Posted an update that should fix this.  v1.3.5

scroundy, 2019-08-11:
I updated and so far good news. This seems to be working for me. I'll keep
my eye on this for a few days. In the past the issue has been intermittent.

hollapm, 2019-08-12:
I'm not seeing anything switching either.  Will continue to watch.

hollapm, 2019-08-14:
Still holding strong.

sfeakes, 2019-08-14:
Glad that fixed it

hollapm, 2019-08-14:
It started up again. It's going 0-75-0 this morning.

sfeakes, 2019-08-14:
The old executable didn’t start again did it?   Can you run Aqualinkd in debug mode and post the output as an attachment?   Make sure you see the % fluctuate while in debug.

hollapm, 2019-08-14:
I'm a complete noobie when it comes to debugging and logging.  within aqualinkd.conf I have log_level=DEBUG.  I have not enabled this command debug_RSProtocol_packets = no.  Do I need to enable that?  Also I had to stop AqualinkD (as log file was not enabled) and have restarted AqualinkD.  When I restarted aqualinkd and it was finally up and running as a webpage I noticed it go from 0 -> 75 and back on the web interface and could hear my controller beep which is how I normally notice something happening.  I do not see 0% in the logs thought?

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/3502291/aqualinkd.log)

sfeakes, 2019-08-14:
You did the debug log correctly.   I can see a lot of errors / problems in your log. They all seem to be coming from serial communication problems.  So I think there may be some issues with the wiring of your RS485 bus.
You are getting a lot of checksum errors, some info is here on that
https://github.com/sfeakes/AqualinkD/wiki#Problems

hollapm, 2019-08-14:
I'll recheck wiring and connections at the control panel and the remote control panel that the raspberry is connected to.  All commands that I send work and it updates everything as well via the interface so I haven't noticed anything weird but will re-check.

After that I can run a direct wire from the control panel inside as a test to see if that helps out.

This is the only item giving some fits but will see if we can backtrack.  Anything else you could think of?

sfeakes, 2019-08-14:
Looking in the logs, there are a few of other issues, it's just that AqualinkD is recovering from them. I think you may see other issues if you actually flex a lot of the functionality in AqualinkD.  When the RS485 are temperamental like this it's very hard to get to the source.  
There is some information here although it's not that clear.  
https://www.schneider-electric.us/en/faqs/FA221785/
Common problem it people T the RS484 wires, or mix the "backbone" vs "daisy chain" connections types listed in the above URL.   Just use daisy chain.  No `T` into existing wires

Kind-a Like this diagram but 4 wires for all devices except the RS4852USB where you'll only need 2. (you can ignore the text)
[image]

https://res.cloudinary.com/engineering-com/image/upload/v1541085989/tips/RS485_vqi7tu.png

hollapm, 2019-08-14:
I am not using adapters so am T into them. Will change that up.

hollapm, 2019-08-16:
I'm trying to figure out how to wire mine better.  I have a raspberry pi connect to the controller in-house, this then connects to the control panel.  The control panel then connects to the first connector on the control board.  Also connected to the transmit/receive are two variable speed pumps.
I also have a second connector on the control board.  This has 2 different devices connected here. One is the Salt Cell and the other I am not certain.  I cleaned up the connections (tightened them, made sure they were solid) and re-ran my log.  I'm attaching it again.  I see a number of warnings still.
The cell is turning on/off again.

I can likely rem out domoticz from aqualinkd.conf since I don't need it now for my environment if that may help?

Also note that my second pool pump gets power when the spa mode is enabled, so there is no power going to that pump till spa mode is turned on, so don't know if aqualinkd may be looking for that device as well and cant see it as there is no power?

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/3507696/aqualinkd.log)

sfeakes, 2019-08-16:
On the two RS485 ports on the main control panel, I have read conflicting documentation from Jandy as to how they should be used, and since my control panel only has one RS485 port I can't pull a board to actually look at the circuitry behind it to figure it out.  
Some documentation states you can use them interchangeable, other states only particular devices on the 2nd port.
Looking at the HEX of the checksum errors, you are getting Jandy messages in the middle of Pentair messages and visa-versa. So that would indicate (although not 100% definitive) that there are more errors on your RS485 than just connecting your Pi.

If you don't use domoticz, then comment out all of those ID's from your config, it won't make any difference to your problem, just make it easier to read the log.

Don't worry about aqualinkd looking for devices that are not powered, it uses the RS485 buss messages for all devices (rather than controller configuration),  so it expects devices to come and go as they are powered up and down.

I may be able to code around the SWG going to 0 problem, but I'll need some detailed / raw logs from you first.   This isn't a proper fix, but I do like to be able to overcome some of the common RS485 errors, so it will help mature the codebase.
If you can 

- add `debug_RSProtocol_packets = yes` to your config, 
- stop aqualinkd
- make sure your pump / swg are on
- start aqualinkd
- wait for the SWG issues to show
- stop aqualinkd
- post the file /tmp/RS485.log to this thread
- (make sure to set the config back ie debug_RSProtocol_packets = no)

Going back to your RS485 connections.
How are your pumps & keypad wired to the 1st RS485 terminal block on the control panel?  Does each pump have it's own wire to the control panel along with the keypad?  If so that's a problem as you are T'ing the RS485 buss.  Any terminal can only have 2 sets of wires MAX connected to it, if it has 3 then you are effectively T'ing the RS buss at that connection. 
Think of it like this, any RS device (including the control panel) can only have one upstream device and one downstream device, so 2 sets of wires is the maximum any RS485 terminal block can have.
So. Maybe this would be the way to wire the first port on the control panel.
Pump#1 -> Pump#2 -> Control Panel -> Inside keypad -> Raspberry pi.

Above it the perfect way to wire it, you can sometimes get away with other ways depending on many other factors, but since you are having problems it may be best to stick to the "perfect" setup.
There are lots of other things to take into consideration to help with these problems, like terminating resistors and such. But you're probably best researching that on the web. Pentair also sell a stupidly priced RS485 "port expander" I assume to make things realy easy, but you shouldn't need that with a correctly wired buss.

One other thing you should check is if the SWG is loosing connection with the control panel and that's why it's going to 0.  I may be able to detect that from the logs you send. But when you have this problem check you SWG and see if the "remote control" led is also changing.  That led will have a different name depending on SWG brand, but basically the LED that shows the SWG is being controlled by the control panel and not it-self.

hollapm, 2019-08-16:
I don't see how I can connect the pumps in series.  From the manual - Note: IntelliFlo® pumps cannot be connected in series with other pumps. https://www.thepoolfactory.com/media/wysiwyg/pdfs/pentair_intelliflo_variable_speed_pump_manual.pdf (Page 9)

The Multiplex boards are CRAZY expensive for what they look like...  GRRR

I'll take a look an re-run the log shortly.

hollapm, 2019-08-16:
I moved another device inline with the salt water cell, so that I now have no more than 2 max on the main control panel.

I have made the changes as outlined above and will wait to see if it starts to go 0 ->75 -> 0 again.  Only been running a few minutes.  I also cleared the log file.

hollapm, 2019-08-16:
Attached are logs and RS485
[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/3511348/aqualinkd.log)

[RS485.log.zip](https://github.com/sfeakes/AqualinkD/files/3511346/RS485.log.zip)

Files attached

sfeakes, 2019-08-17:
Great thanks very much for the RS485 log, I have run that through my aqualinkd player and can see exactly what's going on.  I can replicate what exactly you see.  What's really odd, is your control panel is actually telling the SWG to go to 75, then go to 0 then back to 75 in the RS485 messages it sends it.  BUT what it sends the UI (the keypad & iAqualink) is a constant 75.  I've never seen anything quite like that before.  And it doesn't seem to be doing that due to "loosing a connection to the SWG", which would normally be the case.
I'll need to do a lot more work to see what why it would be doing that, but if you set "read_all_devices = no" that should clear up the SWG going to 0 problem.  But, then you won't be able to see the RPM & WATT info for the Pentair pumps.

BTW, the ID of your second Pentair pump is 0x61, so you should assign that to the SPA_MODE in the config.
`button_02_pumpID=0x61`

hollapm, 2019-08-17:
Thanks for the assistance.  I'd love to get the RPM and Watt information but then don't want it going from 0 -> 75 -> 0 over and over so will make that change.  I'll turn that off for now and see what happens.
I did have the Pentair pump info for the second pump connected to Aux_5 but will add it to #2 as well seeing thats the spa mode (#5 is to power the pump).

One other question - is it possible to have aqualinkd send a message if there is an error detected?  For example my SWG was sending a message and I could see it on the aqualinkd main page - would love to get a text if an error is sent (in addition to the low battery).  Given that I would be turning off the read_all_devices this may not be pertinent anyway?

hollapm, 2019-08-17:
One more suggestion.  Since you can click on a light and select a color as well as change temp, it would be great to click on a pump and change the pump speed (or select pre-configured speeds).  I keep trying to change the speed from the window ;-)

hollapm, 2019-08-17:
I decided to go back out to the controller and check everything.  I had one loose screw which I tightened - don't know if that will make a difference.  I'm sending just in case, but looks like it was still causing a 0 ->75 -> 0 so am changing "read_all_devices = no" 

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/3511533/aqualinkd.log)
[RS485.log](https://github.com/sfeakes/AqualinkD/files/3511534/RS485.log)

sfeakes, 2019-08-17:
I’ve coded round the 75-0-75 problem you were seeing, I’m running it through some full tests and hopefully post the update tomorrow. So don’t worry too much about changing the config.

AqualinkD does post full SWG full status information to MQTT, so if you have any home automation hub integration, that should be able to txt/email you when the SWG says “clean cell” or some such.

As for changing pump speeds that’s on the to do list, but I’m not sure how people actually use that feature (since I do have any VSP pump), so I’ll probably be asking a few questions about that soon.

hollapm, 2019-08-17:
Fantastic! To have it send a message what's the best way to code that in?  I'd want it to communicate any SWG message versus just clean SWG. I'm using home assistant and Google home.

For the VSP I'll help you out. The majority of speed settings are around running low speed for X hours to conserve energy but the pool cleaner may not operate, then bump speed up for the pool cleaner to work for a few hours. Also a great way to adjust the pressure for the Spa. Just to name a few.
Each of my Pentair VSP has 8 speed settings.

sfeakes, 2019-08-17:
Just posted 1.3.5a. This should fix 90% of the 75-0-75 you were seeing.  Still trying to figure out the logic of the Jandy control panel actually setting it to 0, but it looks like after missing 5 acknowledgment from the SWG, the control panel will reset the SWG.  BUT, it looks like they are not resetting the counter after receiving a good acknowledgment.  You'd expect it to be missing 5 in a row, so maybe there is a bug in the control panel firmware.
I'll work on it some more, but this should get you by for the moment.

hollapm, 2019-08-17:
Fantastic, will download later when I get back and post comments.

hollapm, 2019-08-17:
Just upgraded, would you like for me to create any logs?

sfeakes, 2019-08-18:
Not yet.  I just made some more changes (not uploaded yet), I may need the logs from that.  But if you notice your SWG going to 0 is less than it was before I'd like to know.

hollapm, 2019-08-18:
It's definitely been better. Much...

hollapm, 2019-08-18:
It started changing again.

sfeakes, 2019-08-18:
I thought that might be the case.  You should really look into the packet errors on the RS485 buss.  But I'm just about to upload another update that may help, Look for 1.3.5b that should be posted in a hour or so.

You'll need to add a configuration parameter to your /etc/aqualinkd.conf of the below
```
swg_zero_ignore_count = 20
```

This parameter will basically ignore X number of consecutive messages that the control panel sends the SWG, that tell it to set SWG to zero.
From the logs I've seen so far, you're average "set swg to zero" message count is 20, and the highest I've seen in your logs is 50.  So setting that to 50 or 60 should overcome your problems.

Few things to note, when the SWG is actually supposed to turn off, you get a few hundred messages so don't set that higher than 160~200 otherwise you'll never get the off message cleanly. Your average "set swg to zero" message count increases in the logs I've seen of yours, ie start of the log is 5 or so, end of the log is 50 or so. If this simply keeps increasing then after a day or so, nothing will help this problem except fixing the bad packets.

sfeakes, 2019-08-18:
> Fantastic! To have it send a message what's the best way to code that in? I'd want it to communicate any SWG message versus just clean SWG. I'm using home assistant and Google home.
> 

This will depend on the home automation hub, since I don't use HASSIO or Googlehome, I'm going to be generic here.
AqualinkD posts full SWG information to a MQTT topic of `aqualinkd/SWG/fullstatus`, see the wiki for the meanings of all the values. https://github.com/sfeakes/AqualinkD/wiki#MQTT

You'll simply need to read that value and send txt / email depending on what you decide.  Looking at the HASSIO config posted here, it doesn't use that topic, so you'll need to add something like a "selector switch" that has the following MQTT value assigned to the value
`(0=on, 1=no flow, 2=low salt, 4=high salt, 8=clean cell, 9=turning off, 16=high current, 32=low volts, 64=low temp, 128=check PCB, 255=off)`
Then look for documentation on how to send events. This is usually called a "trigger", you configure the trigger to TXT/email you when is sees 'selector switch' set to 128 (check PCB).  The trigger is usually point n click configuration in any decent home automation hub.

hollapm, 2019-08-21:
I think I'm able to have it send messages with Hassio.  I need to do some playing to see if this works - Configuration.yaml
notify:

  - platform: smtp
    name: aqualinkdswg
    server: smtp.gmail.com
    port: 587
    timeout: 15
    sender: YOUR_EMAIL@gmail.com
    encryption: starttls
    username: YOUR_EMAIL@gmail.com
    password: GOOGLE_2Factor_Password
    recipient:
      - YOUR_EMAIL@gmail.com
      - Verizon_Phone_number@vtext.com
    sender_name: AqualinkD

Automations.yaml
- alias: 'Saltwater Cell Message'
  trigger:
    platform: mqtt
    topic: aqualinkd/SWG/fullstatus
    payload: "8"
  action:
    - service: notify.aqualinkdswg
      data:
        title: 'AqualinkD SWG'
        message: 'Salt Water Cell needs cleaning'

Now I just need to have it only send me one message a day versus on a regular basis ;-) . I need to figure that out next.

hollapm, 2019-08-21:
BTW I updated the ignore_count to 0 and so far its working for me - no resets.  

I have notices my pool heater and spa heater turn on, but think it may be when I open the app on the phone and try to scroll I may be hitting those icons, so they are very sensitive.  Will continue to see if that was it.

hollapm, 2019-08-29:
I've updated the software as well as wiring.  Not noticing the 0->75->0 nearly as often.  Need to monitor more closely.  Wiring may definitely have been a challenge.

Will check out lighting as well with the update wiring.

hollapm, 2019-09-03:
So far its doing much better.  Been traveling - will look at lighting this week.

hollapm, 2019-09-04:
Tried out the lighting again.  For some reason the colors are still not working - I'm using the GUI interface.  
I then went to the simple interface and tried again.  Still acting wonky.

Finally I went to the iaqualink app and tried just for giggles.  and it worked.  Not certain what I can capture to help resolve?

I do believe the replacing of wiring helped with 0% and back has been resolved though!

hollapm, 2019-09-05:
Humm,
It started the 0 -> 75 -> 0 again.  My cell needed cleaning so turned off the equipment cleaned and turned back on.  While the cell was being cleaned I also rebooted the raspberry pi.  Once I turned the gear on it started acting odd again.  I don't believe it had time to build any errors up while the equipment was in maintenance mode?

hollapm, 2019-09-14:
I think that I can call the fluxuation fixed. The lights are still acting weird though.

closed 2019-09-14 by hollapm