aqualinkd/AqualinkD #91: iaqualink 5.0 app causing log errors

niharmehta, 2019-09-19:
Jandy recently upgraded their iaqualink android app to version 5.0. When using the 5.0 app interface, Serial Too Large / Bad Packet errors are logged in the aqualinkD .  When using the previous version of the app (4.X)  no errors are logged.  I can run both concurrently and recreate between versions.    
Web interface also shows NO errors, even when launched from the affected app version. 

` Sep-18-19 20:54:16 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|

Sep-18-19 20:54:16 PM Warning:Serial packet too large
Sep-18-19 20:54:17 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|

Sep-18-19 20:54:17 PM Warning:Serial packet too large
Sep-18-19 20:54:17 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x09|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x01|0x01|0x00|0x00|0x05|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x69|0x6c|0x6c|0x6f|0x76|0x65|0x72|0x00|0x01|0x00|0x00|0x05|0x53|0x6c|0x69|0x64|0x65|0x00|0x01|0x00|

Sep-18-19 20:54:17 PM Warning:Serial packet too large
^[[CSep-18-19 21:01:12 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|

Sep-18-19 21:01:12 PM Warning:Serial packet too large
Sep-18-19 21:01:13 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x09|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x01|0x01|0x00|0x00|0x05|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x69|0x6c|0x6c|0x6f|0x76|0x65|0x72|0x00|0x01|0x00|0x00|0x05|0x53|0x6c|0x69|0x64|0x65|0x00|0x01|0x00|

Sep-18-19 21:01:13 PM Warning:Serial packet too large
Sep-18-19 21:01:18 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|

Sep-18-19 21:01:18 PM Warning:Serial packet too large
Sep-18-19 21:01:19 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x09|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x01|0x01|0x00|0x00|0x05|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x69|0x6c|0x6c|0x6f|0x76|0x65|0x72|0x00|0x01|0x00|0x00|0x05|0x53|0x6c|0x69|0x64|0x65|0x00|0x01|0x00|

Sep-18-19 21:01:19 PM Warning:Serial packet too large
Sep-18-19 21:01:31 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|

Sep-18-19 21:01:31 PM Warning:Serial packet too large
Sep-18-19 21:01:32 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x72|0x08|0x01|0x02|0x03|0x04|0x05|0x06|0x07|0x20|0x00|0x01|0x00|0x00|0x09|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x01|0x01|0x00|0x00|0x05|0x4c|0x69|0x67|0x68|0x74|0x00|0x01|0x00|0x00|0x09|0x53|0x70|0x69|0x6c|0x6c|0x6f|0x76|0x65|0x72|0x00|0x01|0x00|0x00|0x05|0x53|0x6c|0x69|0x64|0x65|0x00|0x01|0x00|

Sep-18-19 21:01:32 PM Warning:Serial packet too large
Sep-18-19 21:01:32 PM Warning:BAD PACKET    Jandy Packet | HEX: 0x10|0x02|0x33|0x71|0x06|0x01|0x02|0x03|0x04|0x05|0x06|0x00|0x07|0x41|0x6c|0x6c|0x20|0x4f|0x46|0x46|0x00|0x0a|0x55|0x70|0x70|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x4c|0x6f|0x77|0x65|0x72|0x20|0x50|0x6f|0x6f|0x6c|0x00|0x0a|0x57|0x61|0x74|0x65|0x72|0x66|0x61|0x6c|0x6c|0x73|0x00|0x05|0x50|0x61|0x72|0x74|0x79|0x00|
`

sfeakes, 2019-09-19:
Apart from filling up you log, there shouldn't be any issues. AqualinkD doesn't use the iAqualink device packets for anything, so AqualinkD will simply ignore these packets.

Thanks for the information, I'll code around this in the next release.  BTW, before iAqualink would send one menu item per messages packet, now it looks like it's sending multiple menu items per messages per packet, and that's why it's larger.  Below is basically what that 2 of those messages decodes to.  Before that would have been 9 messages.

Waterfall
Light	
Spillover
Slide

All OFF
Upper Pool
Lower Pool
Waterfalls
Party

niharmehta, 2019-09-27:
Makes sense.. Interesting they actually made a substantial update to their iAqualink. First time in years.  Slightly surprised the app actually defines the packet creation  vs. the iAqualink device itself.

closed 2019-09-27 by niharmehta