aqualinkd/AqualinkD #96: Could not select SET AQUAPURE menu

niharmehta, 2019-10-21:
Playing around with MQTT control and was finding that settings changes were reverting back from the  RS CP.  Changing via the aquaD UI also was not changing the SWG settiings .   Noticed logs had the 'Could not select SET AQUAPURE menu' at each attempt . Restarting the service fixed the problem but also noticed that there is a similar message for the HELP menu on restart. I do not have Debug logs as I have not been able to recreate reliably. 

Oct-21-19 13:39:24 PM Warning:Could not select SET AQUAPURE menu
Oct-21-19 13:45:01 PM Notice: Setting setpoint of Salt Water Generator to 25
Oct-21-19 13:45:01 PM Notice: Setting setpoint of Salt Water Generator to 25
Oct-21-19 13:45:01 PM Notice: Setting SWG % to 25
Oct-21-19 13:45:01 PM Notice: Setting setpoint of Salt Water Generator to 25
Oct-21-19 13:45:02 PM Notice: Setting setpoint of Salt Water Generator to 20
Oct-21-19 13:45:02 PM Notice: Setting setpoint of Salt Water Generator to 20
Oct-21-19 13:45:02 PM Notice: Setting SWG % to 20
Oct-21-19 13:45:28 PM Notice: Setting setpoint of Salt Water Generator to 30
Oct-21-19 13:45:28 PM Notice: Setting setpoint of Salt Water Generator to 30
Oct-21-19 13:45:28 PM Notice: SWG  30s already 30, not changing
Oct-21-19 13:45:49 PM Notice: Setting setpoint of Salt Water Generator to 20
Oct-21-19 13:45:49 PM Notice: Setting setpoint of Salt Water Generator to 20
Oct-21-19 13:45:49 PM Notice: Setting setpoint of Salt Water Generator to 20
Oct-21-19 13:45:49 PM Notice: SWG  20s already 20, not changing
Oct-21-19 13:45:55 PM Notice: Setting setpoint of Salt Water Generator to 30
Oct-21-19 13:45:55 PM Notice: Setting setpoint of Salt Water Generator to 30
Oct-21-19 13:45:55 PM Notice: SWG  30s already 30, not changing
Oct-21-19 13:46:10 PM Warning:Could not select SET AQUAPURE menu
Oct-21-19 13:46:11 PM Notice: Setting setpoint of Salt Water Generator to 20

---- RESTART-----
Oct-21-19 13:48:05 PM Notice: Config BTN Solar_Heater  = label NONE            | VSP ID None  | dzidx 0 | 
Oct-21-19 13:48:05 PM Notice: Starting web server on port 80
Oct-21-19 13:48:05 PM Notice: Starting MQTT client to 192.168.37.39:1883
Oct-21-19 13:48:05 PM Notice: Listening to Aqualink RS8 on serial port: /dev/ttyUSB0
Oct-21-19 13:48:13 PM Notice: Getting control panel information
Oct-21-19 13:49:19 PM Warning:Could not select HELP menu

niharmehta, 2019-10-21:
Is the cycling through menu's a new feature in the last couple of releases?  I seems different than what I saw before in logs. (or may not have been shown).  I notice it sometimes takes over 10s for a SWG setting to be reflected on the status of the IAqualink.    I can make a change externally, it updates AqualinkD, then takes a while to update CP.  At which time, AqualinkD and MQTT goes back to the old setting, and a few moments later, updates with the expected new setting after it finally programs the CP.  Does not happen all the time.

sfeakes, 2019-10-21:
IT's not a new feature, it just means that the setting of SWG didn't take effect.  When setting SWG through AqualinkD, it's always reported the new setting right away, before it actually programs the control panel, (That way people don't get confused), then once programmed all is matched. If the programming fails, it reverts back to the old setting.  So, you probably never noticed it before since AqualinkD never failed at programming.  I did make some substantial changes to the programing logic over the last two releases, to speed it up and overcome some minor errors.  
I'll go back and take a look and figure it out. If it's a problem for you, then going back to 1.3.6 should fix it as that's the last release before the programming changes.

niharmehta, 2019-10-22:
Thanks for the explanation. I opened bug because it seemed like we got in to an odd state where it was in the wrong menu repeatedly even when set directly from AqualinkD.

niharmehta, 2019-10-22:
I captured a debug log of SWG programming failing again.   I think it starts at:
Oct-22-19 13:15:40 PM Debug:  MQTT: topic aqualinkd/SWG/Percent/set 30

[aqualinkd.log.221019-1.zip](https://github.com/sfeakes/AqualinkD/files/3758372/aqualinkd.log.221019-1.zip)

sfeakes, 2019-10-23:
Thanks.  I've posted 1.3.9a, see if that helps.

niharmehta, 2019-10-23:
Loaded 1.3.9a. It is possible there is improvement to one condition, , but captured another miss. Have not checked to see if same as before. 
Starts at: 
Oct-23-19 09:41:42 AM Debug:  MQTT: topic aqualinkd/SWG/Percent/set 25
[aqualinkd.log.231019-1.zip](https://github.com/sfeakes/AqualinkD/files/3763490/aqualinkd.log.231019-1.zip)

sfeakes, 2019-10-23:
Thanks for trying.  Please try 1.3.9b

niharmehta, 2019-10-23:
Seems to be better. Did not get a miss in the 15 minutes I was testing. I will keep trying later as well

sfeakes, 2019-10-23:
Great, Glad to hear it's better.

closed 2019-10-24 by sfeakes