aqualinkd/AqualinkD #97: AqualinkD not communicating with control panel after board change and PureLink Power Center install

john1158way, 2019-10-22:
First thanks for a great project.  I have had AqualinkD running for over a year with no issues.  This summer a neighborhood wide power outage took out my RS Control board and I had it repaired by Robert Pawley at bpengineering.   While it was out I also installed a Jandy Chlorine generator and moved the RS control board to a PureLink power center to provide automation and power to the chlorine generator.    After this work was completed the AqualinkD would not communicate with the control panel.  The RS keypad works fine and the serial logger and the daemon logs show the 485 communication is being received successfully.

What I start up the keypad I see version information for keypad and control panel they are 6700 Revision C and 6521 Revision F

When I run the default web app it shows the buttons and my configuration. The air and water temp are blank and the initial state of the buttons is all off even if some are on.    The buttons will turn on if I click them but no reaction from the control panel.   

When I run the simulator there are no scrolling messages and the buttons have no effect and the on off indicators remain off no matter the current state or if I click buttons

I have completed the following steps
I ran the serial logger and the IDs are not in conflict. 
For whatever reason my keypad jumpers were set to controller 2 / ID 0x09 I tried 0x08 and 0x0a  and it made no difference

I changed the controller jumpers to controller 1 / ID 0x08 and tried 0x09 and 0x0a no change

I set the ID of AqualinkD to match the RS Keypad and using the simulator I would now see the messages sent to the keypad and the buttons would show status but as expected the  app would not control anything

I ran aqua.sh clean and did a fresh install of 1.3.9 no change to any of the above scenarios

I used 0x00 for the AqualinkD ID it chose 0x0a but still no change to any of the above scenarios

I see In the log the control panel is sending a probe to the AqualinkD ID but AqualinkD is not responding.  

I have pasted  a clip from the log showing the 0x0a probe and a fuller view of the start up log below
Oct-22-19 09:31:25 AM Debug:  Serial read 7 bytes
Oct-22-19 09:31:25 AM Debug:  To 0x0a of type    Probe | HEX: 0x10|0x02|0x0a|0x00|0x1c|0x10|0x03|
Oct-22-19 09:31:25 AM Debug:  To 0x0a of type    Probe | HEX: 0x10|0x02|0x0a|0x00|0x1c|0x10|0x03|
Oct-22-19 09:31:25 AM Debug:  RS received packet of type Probe length 7
Oct-22-19 09:31:25 AM Debug:  RS Received PROBE length 7.

Oct-22-19 09:31:24 AM Notice: Aqualink Daemon v1.3.9
Oct-22-19 09:31:24 AM Notice: Config log_level         = 8
Oct-22-19 09:31:24 AM Notice: Config socket_port       = 80
Oct-22-19 09:31:24 AM Notice: Config serial_port       = /dev/ttyUSB0
Oct-22-19 09:31:24 AM Notice: Config web_directory     = /var/www/aqualinkd/
Oct-22-19 09:31:24 AM Notice: Config device_id         = 0x0a
Oct-22-19 09:31:24 AM Notice: Config read_all_devices  = YES
Oct-22-19 09:31:24 AM Notice: Config use_aux_labels    = NO
Oct-22-19 09:31:24 AM Notice: Config override frz prot = NO
Oct-22-19 09:31:24 AM Notice: Config mqtt_server       = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_dz_sub_topic = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_dz_pub_topic = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_aq_topic     = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_user         = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_passwd       = (null)
Oct-22-19 09:31:24 AM Notice: Config mqtt_ID           = aqualinkd_b827eba1
Oct-22-19 09:31:24 AM Notice: Config idx water temp    = -999
Oct-22-19 09:31:24 AM Notice: Config idx pool temp     = -999
Oct-22-19 09:31:24 AM Notice: Config idx spa temp      = -999
Oct-22-19 09:31:24 AM Notice: Config idx SWG Percent   = 0
Oct-22-19 09:31:24 AM Notice: Config idx SWG PPM       = 0
Oct-22-19 09:31:24 AM Notice: Config PDA Mode          = NO
Oct-22-19 09:31:24 AM Notice: Config PDA Sleep Mode    = NO
Oct-22-19 09:31:24 AM Notice: Config force SWG         = NO
Oct-22-19 09:31:24 AM Notice: Config deamonize         = YES
Oct-22-19 09:31:24 AM Notice: Config log_file          = /var/log/aqualinkd.log
Oct-22-19 09:31:24 AM Notice: Config light_pgm_mode    = 0.00
Oct-22-19 09:31:24 AM Notice: Debug RS485 protocol     = NO
Oct-22-19 09:31:24 AM Notice: Read Pentair Packets     = NO
Oct-22-19 09:31:24 AM Notice: Display warnings in web  = NO
Oct-22-19 09:31:24 AM Notice: Config BTN Filter_Pump   = label Filter Pump     | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Spa_Mode      = label Spa Mode        | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_1         = label Cleaner         | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_2         = label Waterfall       | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_3         = label Spa Blower      | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_4         = label Pool Light      | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_5         = label Spa Light       | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_6         = label NONE            | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Aux_7         = label NONE            | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Pool_Heater   = label Pool Heater     | VSP ID None  | dzidx 0 | 
Oct-22-19 09:31:24 AM Notice: Config BTN Spa_Heater    = label Spa Heater      | VSP ID None  | dzidx 0 | 
[... 810 lines cut ...]
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 9 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 7 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x0b of type    Probe | HEX: 0x10|0x02|0x0b|0x00|0x1d|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x0b of type    Probe | HEX: 0x10|0x02|0x0b|0x00|0x1d|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 12 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 9 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 7 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x10 of type    Probe | HEX: 0x10|0x02|0x10|0x00|0x22|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x10 of type    Probe | HEX: 0x10|0x02|0x10|0x00|0x22|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 12 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x08 of type   Status | HEX: 0x10|0x02|0x08|0x02|0x50|0x00|0x00|0x00|0x00|0x6c|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  Serial read 9 bytes
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|
Oct-22-19 09:31:33 AM Debug:  To 0x00 of type      Ack | HEX: 0x10|0x02|0x00|0x01|0x00|0x00|0x13|0x10|0x03|

Any advice you can give me will be appreciated, thanks again

sfeakes, 2019-10-23:
This is an interesting one.  I just realized The debug messages for send data are messed up in 1.3.9, so that’s what’s making you think it’s not sending a reply to the probe.  If you go back to 1.3.6, that may help debugging, but not shure that’ll fix the problem.  I’m inclined to think there is a keypad register check I’ve never seen before with this PureLink panel, I’ve seen that a few times with newer Jandy equipment, and you’re the first I know of to try a PureLink panel.
If you can do the following, before going to 1.3.6, I’ll see if I can work it out.
Make sure AqualinkD is not running, Pull the power wire from the RS8 keypad, start serial logger in debug and no timeout mode, plug the power wire back in the RS8 keypad, wait for it to register, do a few commands on the keypad, then post the complete output from the serial logger.  Below are the arguments to pass to the serial logger.

‘/dev/ttyUSB0 -d -p 0’

sfeakes, 2019-10-23:
I've posted 1.3.9a, it may help with debugging this problem.

john1158way, 2019-10-23:
I am running 1.3.9a  I will perform serial logging with that version

john1158way, 2019-10-23:
Thank you for this,  I have run the serial logger as you requested and attached the file.   I started the pump and several aux devices and then shut them down.   I realize now the logger is a seperate program and so if you want me to do anything with 9b let me know.   
[serial log output with control panel plugged in after start.txt](https://github.com/sfeakes/AqualinkD/files/3763836/serial.log.output.with.control.panel.plugged.in.after.start.txt)

john1158way, 2019-10-23:
since I have installed 1.3.9a I thought I would repeat the exercise of starting the keypad after starting AqualinkD.   I have turned on the debugging log and pasted both the normal log set to serial_debugging and the 485 log.  
[RS485 log plugging in keypad after start.log](https://github.com/sfeakes/AqualinkD/files/3763911/RS485.log.plugging.in.keypad.after.start.log)
[aqualinkd 1.3.9a log plugging in keypad after start.log](https://github.com/sfeakes/AqualinkD/files/3763913/aqualinkd.1.3.9a.log.plugging.in.keypad.after.start.log)

sfeakes, 2019-10-23:
This is exactly what I needed.  It's showing that the RS8 keypad is using an identical initialization sequence to AqualinkD.  So, the next idea I have, is that Jandy has stripped down the PureLink panel like they did with the PDA panel.  Ie, limited the number of devices and ID's that you can actually use.
Can you unplug the keypad, set AqualinkD to id 0x08, and start AqualinkD (with kaypad not installed), see if it comes to life that way.

john1158way, 2019-10-23:
I set the id to 0x08 and disconnected the keypad.  No change from previous scenarios unresponsive.  I plugged in the keypad while AqualinkD was running and cycled some equipment from the keypad I then checked the simulator messages started scrolling up the simulator but there was no response to any button clicks.   I then started up the simple interface and the temperature was showing and the sliders would move from off to on but return to off after a second or so.    I have attached the log from this run
[id0x08keypaddisconnectedatstart.log](https://github.com/sfeakes/AqualinkD/files/3764852/id0x08keypaddisconnectedatstart.log)

john1158way, 2019-10-24:
After thinking about it I worried that connecting the keypad may not be obvious in the log and you might prefer a log that just showed what happened with no keypad connected.  it is below.   No response at all in AqualinkD
[id0x08keypaddisconnected.log](https://github.com/sfeakes/AqualinkD/files/3765066/id0x08keypaddisconnected.log)

sfeakes, 2019-10-24:
Perfect, thanks for that, I was just going to ask for it when you posted it.
This is really odd, and also the information from when keypad was running is confusing.  I need to give this some thought, and probably have to write some test software, to get much further.  One thing as a long shot would be to test the RS cable. How easy would it be for you to plug the pi into the exact wires that the RS8 keypad is using and test with the 0x08 ID?

In the mean time, I’ll go over all your information and see if I can see/think of anything.

john1158way, 2019-10-24:
I will plug the PI directly into the same wires as keypad and report back

john1158way, 2019-10-24:
I connected the wires directly to the cable from the PureLink system.  The one thing that is different from Aqualink is that PureLink comes with a 485 multiplex card that has 4 485 connection points so my RS Control card connects to this and the Chlorine system connects to this and my keypad connects to this
[id0x08directconnect.log](https://github.com/sfeakes/AqualinkD/files/3765281/id0x08directconnect.log)

sfeakes, 2019-10-24:
Thanks.   It’s not the cable.  Give me some time to think about this and spent some more time on your logs.  I’ve never encountered anything quite like this before.

john1158way, 2019-10-24:
your confirmation that AqualinkD was doing the same initialization sequence as the keypad gave me an ID to test.  What if AqualinkD was receiving all of the bus traffic but in the new hardware setup with the purelink multiplex board the adapter on the Pi was not transmitting a signal the RS Control Panel could read?   To test this I took the Pi out to the pool and disconnected the 485 multiplex board and connected the pi directly to the RS control board.   Success!  so it looks to be a 485 bus issue.  I will try to troubleshoot further and share my findings  I have attached a log of the simulator successfully controlling the RS
[piatcontrol.log](https://github.com/sfeakes/AqualinkD/files/3768117/piatcontrol.log)

sfeakes, 2019-10-24:
This is really interesting. The reason I asked you to plug the Pi directly into the keypad cables was to test the bi-direction.  

So the multiplex board looks to be the problem.  I’ve always wondered about exactly what that box does as you actually don’t need it in a correctly wired RS485 network. Problem is most pool installers are not experienced in network cable cabling. So I always thought the multiplex board simply covercame that by requiring one device per port.  

So few options.  Bypass the multiplex board for the Pi connection (and anything else on that wire run). Or remove the multiplex board altogether and wire to RS485 spec (probably with termination). There are a lot of resources in the net for this.    But not knowing the details of the PureLink that might be a lot of trial and error.

Still need to get to the bottom of what the multiplex board does to stop the pi from working through.

sfeakes, 2019-10-24:
On other thing, if you do pull the multiplex board, I'd love to see a quality pic of the back of it. Looking online I can't see any pic's of the back, only the front.  I'm simply trying to determine if there are terminating resistors built in, and how it's actually wired between the ports and source connections.

john1158way, 2019-10-24:
upon closer inspection the multiplex board I have is actually just a multiple bus connection card.  I have attached pictures of front and back.  I also have attached pictures of the AquaPure board.   Currently I can connect through this multiple connection board and everything still work as long as the AquaPure board is not connected.   I am going to do some logging of starting the system without the AquaPure board connected  and connecting it to see if it captures any messages  I also ordered a new adapter in case the AquaPure connection makes the bus communication more difficult but a better adapter could cope.  Finally I am going to look into 485 bus troubleshooting to see if anything like termination could help.
[image]
[image]
[image]

sfeakes, 2019-10-24:
So the way that board is wired, you're limited to 4 devices in total (unless you add another board). You should not have any more than once device per port (it may work, spec's are loose, fault tolerance is good, but to the RS485 design spec as I understand it, you shouldn't). And cable connected to port 2 & 3 need's to be as short as possible (I mean really short, like 1 foot).

If you list out RS485 devices / cable lengths I'll see if I can help.   You may also want to put a multimeter set to Ohms and measure the resistance between D+ and D- on every RS device you have (with power off), so you know which are terminated, and what they are terminated with.

john1158way, 2019-10-24:
Additional test results  ran test with everything but AquaPure board on 485 bus and cycled equipment with AqualinkD i  added Aquapure without shutting down and it stopped responding.      I have attached logs from Aqualink and the RS485 debugging log. 
[aquapureonbus after start.log](https://github.com/sfeakes/AqualinkD/files/3769470/aquapureonbus.after.start.log)
[RS485 Aquapure on bus after start.log](https://github.com/sfeakes/AqualinkD/files/3769471/RS485.Aquapure.on.bus.after.start.log)

john1158way, 2019-10-24:
thank you for the RS485 advice.   I will do the checks with the multimeter.  I may eliminate the board and just daisy chain all of the devices as that seems to be the preferred wiring pattern the board seems to be a convenient but  totally parallel connection

john1158way, 2019-10-25:
the resistance readings for my equipment are as follows:
Jandy RS Control Board   500 ohms regardless of polarity of meter
Jandy Keypad  1k ohms regardless of polarity
Jandy AquaPure 1k ohms regardless of polarity 
DSD Tech USB to 485 adapter 20k ohms if measured positive of meter  to positive data 40k if polarity reversed

sfeakes, 2019-10-25:
The USB adapter is just the resistance of the circuitry, ie not terminated.
The fact that Jandy terminated everything and at different values makes it hard to wire it a true 458 network bus fashion. All my equipment is older and a miss-match of manufactures, so different values.

Do you have 2 RS wire sets running inside, or does the USB adapter and RS Keypad need to be on the same run?   If you have a separate RS wire set for each device, then terminating the USB adapter with 1k and plugging it into the multiplex board on it's own wire might be the solution.
Problem you have is you can't daisy chain / bus link anything off that multiplex the way it's wired, each device needs it's own port and need to be terminated.

From your description, does the PureLink control panel hold both a Aquapure PCB and Control Board PCB and they are connected over RS458 (with that multiplex board)?  Does the Control Board have multiple RS485 sockets or just the one?

john1158way, 2019-10-27:
To answer your last question first.   The PureLink box has both a Aquapure PCB and an Aqualinke RS Control panel.  Note the PureLink does not come with the Aqualink board as standard you have to purchase separately, but it comes with a plug to power the board  and the box is made to mount it top front just as in AquaLink.   In my case  I took the one I had and mounted in the PureLink.    It does come with 4 relays. 

The Aquapure PCB connects to the RS485 bus as it's only communication / control connection.  There is a local control panel that connects directly tot the Aquapure and mounts to the left of the Aqualink control panel

I only have one set of wires running inside.   After reading your advice and from a 485 best practices guide I took the multiplex board out of the wiring and went with daisy chain.   still works without AquaPure but not with Aquapure.   I have ordered a new 485 adapter for my pi to see if that will work.   For now I have the control board connected to a long wire inside.   The first device on the long wire is the PI running AqualinkD and the end device is the keypad.   Since Jandy builds in termination I have termination at both ends albeit 500 ohms on one end and 1k on one end instead of 120 at both.  According to TI 120 at each end is the preferred method and terminating every device is not needed.   I suppose I could put a 150 ohm resistor across both ends and be much closer to 120 on each end (circa 130, 115)

Once I get the different Adapter I will install it and try the connecting the Aquapure again.   While writing this I just realized that in my most recent test the end device was the Aquapure then the Aqualink control panel then AqualinkD, and the keypad on the other end.   When I get the new adapter I will also try connecting the long run from the house to the AquaPure then on to the AquaLink so the end points are the Aqualink and the control pad.   I will also try with termination at the end

If all this fails I will try serial logging from a daisy chain connection between the Aquapure and the Aqualink control panel to monitor the AqualinkD start up using a second device to validate if the initialization commands are readable on the 485 bus at the control panel

I really appreciate your support

john1158way, 2019-10-27:
i have posted a  couple of pictures of the PureLink 
[image]
[image]