aqualinkd/AqualinkD #102: Heater and freeze protection settings only intermittently show up

troywehrle2, 2020-01-31:
I have version 1.3.9c successfully installed and mostly working well. My control panel version is B0029221 REV T.2.

I am only seeing the freeze protection tile show up intermittently on the default page. As well, sometimes the pool and spa heater tiles show up as only switches, whereas other times they show up as thermostats showing enabled or on with the selected target temp. The intermittence behavior changes only as I restart AqualinkD. See attached for debug log.

My control panel version is above. I also have a OneTouch controller and AquaPalm controller installed on this system. It is a new system, just installed two weeks ago.

Would very much appreciate some input on how to make more stable. I intend to create drivers and apps for the Hubitat HA ecosystem and depend on this AqualinkD for integration.

Thank You

[aqualinkd.log](https://github.com/sfeakes/AqualinkD/files/4141337/aqualinkd.log)

sfeakes, 2020-02-07:
Every time AqualinkD starts is will detect what's available, and configure accordingly. So if on startup it can't find Heater Setpoints, it configures them as switches so you can turn on/off but not adjust temp. If it's finds the set points, it will configure them as thermostats so you can turn on/off & set temp.  
So AqualinkD is not finding all devices accurately on startup every time, that's what you seeing.

Just checked logs, and above is confirmed.  I've got a new version that may I hope will fix this, but I haven't found time to fully test it, (it's a huge update). so you might want to go back one version / release and see if that helps until I can post the next version.

troywehrle2, 2020-02-07:
Since posting this, I have found that if at startup it fails to find these set points, I shut bring up your All Button interface and navigate through the Review menu for them manually. Immediately after doing this, I see the freeze protection, pool and heater tiles all get displayed correctly.

Given I can manually workaround this concern, I will stay with the current version. Especially since I have a Pentair VSP and it appears from other postings , that 1.3.9c works best with it. I am indeed not having any other issues at this time.

Thank You

sfeakes, 2020-02-07:
Well done, I'd never thought about doing that. But yes that would work, as soon as AqualinkD realizes it can change set points it will re-configure itself. (up to 5 mins delay getting new config to homebridge though, if you use aquilinkd-homebridge).

Next release has a ton of updates for Pentair VSP.  (full control).  But it's proving a real pain to test / get working with all the different control panel versions, and work has been killing my spare time recently.

troywehrle2, 2020-02-07:
I saw in the code that it essentially runs the All Button commands, so figured give it a try. I was thinking of patching the code to try and correct, but once I saw how easy it was to workaround, I put that on the back burner for other integration priorities.

I use your MQTT integration for Hubitat integation. So immediate update.

Since I am now vested in my pool automation solution using your software, I can volunteer to help test. I am a software architect, so know the concerns/concepts. Of course, within my own "spare time".

sfeakes, 2025-10-21:
Check your config for rssa_device_id and make sure that’s set accurately