aqualinkd/AqualinkD #103: Can two RS485 USB interfaces be used with AqualinkD?

ncpe20012003, 2020-02-06:
Seems the app is looking for ttyUSB0, can a ttyUSB1 be added as well as a second RS485 interface? One for pump and one for heater.

sfeakes, 2020-02-07:
No, only one USB is supported.  aqualinkd does not talk directly to pump or heater, only to control panel.

ncpe20012003, 2020-02-07:
The newer Jandy equipment have RS485 onboard. Seems like AqualinkD should
be able to utilize it instead of panel.

sfeakes, 2020-02-07:
That's not the way the RS485 protocol works.  There can only be one master on a RS485 network, and that's the control panel.  There are many slaves, and Aqualinkd is simply a slave, all your other devices (pumps, heaters, SWG etc) should also be slaves.  AqualinkD will read what the control panel sends to all devices and also what they send back, but can only control devices through the control panel.  If AqualinkD was to control the pump directly, then you'd have to disconnect the control panel.  But even if that was the case, you'd still only need one RS485 interface for AqualinkD, as you daisy-chain devices together.

john1158way, 2020-02-08:
ncpe,  I am curious, do you have a specific reason you wish to avoid using the same RS485 bus? Sean has created a great program that gives you great control over an existing Aqualink system as more or less a very advanced control station on the system, but the Aqualink card in your pool automation center remains in place to drive the physical devices and give you redundancy and local control.   Since the Aqualink card is designed to control pumps and heaters it is a good architecture.    That said I can realize there are specifics that might make a second bus a good choice but it would not be typical.

sfeakes, 2020-02-08:
Im not sure I follow you.  (BTW, I'm probably the Sean/Shaun you refer to, author of AqualinkD and AquapreD). 
I'm not avoiding using the same bus, I'm saying use the same bus, that's why you only need one USB2RS485 adapter on AqualinkD.  You can use /dev/ttyUSB? (? = 0, 1, 2, 3 etc), but only ONE USB2RS485 interface is supported to force you to use the same bus.  Once you're physically connected to the RS485 bus, you can control / communicate with everything else on that bus, that's why you only need one physical connection. The only reason I can see to have two physical connections would be if you need to connect to two separate RS485 bus's. Or you want two separate pieces of software running at the same time communicating with the RS485 bus. But one piece of software should never need to two physical connections to the same RS485 bus.

Can you link to this architecture you refer to, that may help me understand what you mean.

ncpe20012003, 2020-02-08:
New pool just installed. I have Jandy JXi Heater and a Jandy VersaFlo
Variable Speed pump. On the pump, the contractor installed an IQPUMP01 Jandy
Pro Series VS Mobile Pump Interface via the RS485 interface on the pump.
I'm using the iAqualink App on my phone to control. Nothing is connected to
my JXi heater. My thoughts were to connect the new RS485 USB interface from
my RPi, running AqualinkD, directly to the RS485 interface on the heater
for remote control. Eventually I may want to add the pump onto the control
of AqualinkD on the RPi. Sounds like I need to create a RS485 bus for all
three to be connected. In the meantime, can I not connect directly to my
heater and run slave to slave via RS485?

sfeakes, 2020-02-08:
Now I see where you are coming from.  ALL devices should be connected to one RS485 bus, then on that bus you have one master (control panel), and all devices are slaves (pump, heater, SWG etc).  Since you don't have a control panel, your installer simply installed IQPUMP01, I believe this will just control the pump and nothing else (well one other device via a relay).  So no, AqualinkD will not control the heater directly, in your setup it can't even control the pump.  You're kind-a stuck for upgrading as well as far as home automation or remote control goes as well.
A Jandy control panel is designed to communicate with all RS485 devices, control valves, lights, manual pumps, cleaners etc etc.
AqualinkD needs a control panel to communicate with, so you'd have to throw away the IQPUMP01 and add a Jandy control panel, then you could control everything remotely, either with iAqualink and/or AqualinkD.

Other option you have is find a Jandy home automation project that's designed to replace a Jandy control panel completely.  But still have to throw away IQPUMP01.  I don't know of any projects out there that can do that. AquapureD is a start of that, but it's not even close to being a finished project or able to control everything. And the way my spare time is at the moment, that will be a few years off, unless my control panel breaks and I'm forced to spend time on that project.

ncpe20012003, 2020-02-23:
Okay, I broke down and purchased a Jandy RS 8 P&S on ebay with Rev Q PCB.
I'm able to communicate to it via AqualinkD/Raspberry Pi/RS485. I can
enable/disable buttons via Web UI. I have connected my new Jandy JXi gas
heater to it via RS485. Running the seriallogger with AqualinkD shutdown,
the heater ID on the RS485 bus is 0x68, unknown. RS control panel is not
recognizing it. Any ideas?

sfeakes, 2020-02-24:
When installed correctly, your iAqualink interface should be able to talk directly to the RS8, so you should be able to tell if everything is ok without aqualinkd at all.
If you can give me the output of serial logger in debug mode, I can tell what’s active on the bus.  But my guess is the RS8 hasn’t taken control / active on the bus.

ncpe20012003, 2020-02-24:
See attached, 0x68 is the heater.

choyle@AquaLink:~/software/AqualinkD $ sudo ./release/serial_logger /dev/ttyUSB0 -d
AqualinkD serial_logger V1.1
Notice: Logging serial information!

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x32 of type    Probe | HEX: 0x10|0x02|0x32|0x00|0x44|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x33 of type    Probe | HEX: 0x10|0x02|0x33|0x00|0x45|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x38 of type    Probe | HEX: 0x10|0x02|0x38|0x00|0x4a|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x39 of type    Probe | HEX: 0x10|0x02|0x39|0x00|0x4b|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x3a of type    Probe | HEX: 0x10|0x02|0x3a|0x00|0x4c|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x3b of type    Probe | HEX: 0x10|0x02|0x3b|0x00|0x4d|0x10|0x03|

Jandy     To 0x68 of type Unknown  | HEX: 0x10|0x02|0x68|0x0c|0x00|0x64|0x5f|0xff|0x48|0x10|0x03|
Jandy   From 0x68 of type Unknown  | HEX: 0x10|0x02|0x00|0x0d|0x00|0x00|0x00|0x1f|0x10|0x03|

Jandy     To 0x60 of type    Probe | HEX: 0x10|0x02|0x60|0x00|0x72|0x10|0x03|
[... 952 lines cut ...]
Notice: ID 0x70 is not used 
Notice: ID 0x71 is not used 
Notice: ID 0x72 is not used 
Notice: ID 0x73 is not used 
Notice: ID 0x08 is not used  <-- can use for Aqualinkd
Notice: ID 0x09 is not used  <-- can use for Aqualinkd
Notice: ID 0x0a is not used  <-- can use for Aqualinkd
Notice: ID 0x0b is not used  <-- can use for Aqualinkd
Notice: ID 0x40 is not used 
Notice: ID 0x41 is not used 
Notice: ID 0x42 is not used 
Notice: ID 0x43 is not used 
Notice: ID 0x10 is not used 
Notice: ID 0x18 is not used 
Notice: ID 0x20 is not used 
Notice: ID 0x21 is not used 
Notice: ID 0x22 is not used 
Notice: ID 0x23 is not used 
Notice: ID 0x28 is not used 
Notice: ID 0x29 is not used 
Notice: ID 0x2a is not used 
Notice: ID 0x2b is not used 
Notice: ID 0x30 is not used 
Notice: ID 0x31 is not used 
Notice:

sfeakes, 2020-02-24:
From the logs, the control panel is sending message to heater, and heater is replying.  So all looks normal.  So if the control panel is not recognizing the heater, it’s probably due to a version issue.  The RS8 version you have is not compatible with the JXi version. See if there are any JXi settings you can change, and look at the manual for the RS8 rev you have, see what it supports.
RS8 rev info is here
https://github.com/sfeakes/AqualinkD/wiki/Jandy-AqualinkRS-Revisions

The type unknown is simply that I haven’t decoded that protocol yet, so you can ignore that.

ncpe20012003, 2020-02-24:
Page 28 of the JXi manual indicates that Rev N and above should work, I'm
at Q.

https://www.jandy.com/-/media/zodiac/global/downloads/h/h0574300.pdf

Reading the last bullet for Rev N in the pdf you sent it mentions adding
RS485 support for LXi heaters, may be the same family.

ncpe20012003, 2020-03-01:
After additional testing yesterday, I found that rev Q does control the JXi
heater. What i was missing was water temp input on the RS panel. I put a
22k ohm resistor across the water temp across the input on the panel which
equates to 45° F. I could then press the heater button on the panel and
activate the JXi vis rs485.

However, AqualinkD cannot activate the heater through the panel because it
doesn't recognize the heater ID on thr rs485 bus, 0x68. Can something be
tweaked to make this work?

Also, there is a built in water temp sensor in the JXi. I'm wondering if
the water temp is returned via rs485 response when queried.

ncpe20012003, 2020-03-03:
Any advice on this?

timelery, 2020-03-06:
@ncpe20012003 I am trying to figure out how to control my Jxi heater directly without an Aqualink unit. I have RS485 connected between Jxi and my System with 12v power running to POS and Ground oon the Rs485 connection into the Jxi.

I get the following when I probe it from command line

[image]

Do you happen to have the messaging figured out. I want to test whether or not I can turn Spa or Heater on in this way as well as ready these responses coming back from Jxi with a focus on reading the current temperature from Jxi

sfeakes, 2020-03-06:
@timelery The Jxi protocol hasn't been fully worked out yet.  As soon as I get some logs from the unit when it's being controlled from a control panel, I should be able to figure it out.

sfeakes, 2020-03-06:
@ncpe20012003 
https://github.com/sfeakes/AqualinkD/issues/104

timelery, 2020-03-06:
> @timelery The Jxi protocol hasn't been fully worked out yet. As soon as I get some logs from the unit when it's being controlled from a control panel, I should be able to figure it out.

Do you have any recommendations on trying to figure some of the messaging out in the meantime. Without a controller. Any possible probe messages I can send over tot he Jxi unit directly from command line perhaps or a method of trying to figure out the return message from the screenshot I provided.

timelery, 2020-03-06:
@sfeakes 

Testing again I get a lot more back from the Jxi unit. Thoughts?

[image]

timelery, 2020-03-06:
I think a really good first step would be figuring out how to send a message to the Jxi heater to set it to "JANDY REMOTE ONLINE"
[image]

ncpe20012003, 2020-03-07:
Once the JXi is connected to the RS8 panel via 485 is automatically goes
into Jandy Remote mode.

sfeakes, 2024-04-28:
@timelery Not sure if you are still trying to figure out the protocol (this thread is a few years old).
But I have just spent a few hours figuring it out.  Let me know if you want it, it will also be posted in V2.3.5 look in the file `devices_jandy.c` in the repo.